ukBaz/python-bluezero #225: Implement UART for microbit module

ukBaz, 2019-05-05:
this way is ok?

_Originally posted by @xiaolongchengithub in https://github.com/ukBaz/python-bluezero/issues/186#issuecomment-489382741_

ukBaz, 2019-05-05:
Hello @xiaolongchengithub

I am not sure I fully understand your question. It looks like you have also posted a question on @KacperKubara gist:
https://gist.github.com/KacperKubara/c9fbd7aa92d31ca7d81249b2d1a92556#gistcomment-2901780

My assumption is that you are trying to use the microbit UART service which is currently not implemented in Bluezero.

There have been a few requests for the UART service to be supported by Bluezero. Most of the use cases I've had described to me would be better served with the micro:bit event service. More details of this are available at:
https://lancaster-university.github.io/microbit-docs/ble/event-service/
https://lancaster-university.github.io/microbit-docs/resources/bluetooth/bluetooth_profile.html

I will look at implementing the Event or UART service in Bluezero however due to other commitments this is unlikely to be soon.

robertocarvajal, 2019-05-05:
I added support last night by hacking something quickly, I needed to read from another sensor that I pipe the output trough the bt uart, I was able to read from it without much trouble, I haven't tested transmission yet but I will soon, I guess I can send a PR with support for UART service, I don't know how to implement it via the event service approach yet, I will try it today to see if I can subscribe and trigger a callback whenever there is a write on the bluetooth uart.. it is also possible to read the value directly of whatever is saved in the uart buffer and that was my approach and it worked.

ukBaz, 2019-05-06:
I have started to do experiments with the UART service and done an initial commit:
8fc1e759613cf79b3084f89df738d8a3c1f81061

This was the code on the micro:bit
[image]

This was the code on the RPi:
```python
from time import sleep
from bluezero import microbit

ubit = microbit.Microbit(adapter_addr='02:00:AA:48:25:29',
                         device_addr='F1:55:90:65:29:DC',
                         accelerometer_service=False,
                         button_service=False,
                         led_service=False,
                         magnetometer_service=False,
                         pin_service=False,
                         temperature_service=False,
                         uart_service=True)

ubit.connect()

for i in range(5):
    ubit.uart = 'ping#'
    sleep(10)

ubit.disconnect()
```

ukBaz, 2019-05-06:
The TX characteristic works with async mode only so this commit was required
40012cfe6d1b631f52cfdea04a608efb0efee4a2

and was tested with this code on the microbit:
[image]

And was run with this Bluezero example
```python
from bluezero import microbit
from bluezero import async_tools

ubit = microbit.Microbit(adapter_addr='02:00:AA:48:25:29',
                         device_addr='F1:55:90:65:29:DC',
                         accelerometer_service=False,
                         button_service=False,
                         led_service=False,
                         magnetometer_service=False,
                         pin_service=False,
                         temperature_service=False,
                         uart_service=True)
eloop = async_tools.EventLoop()
ubit.connect()

def ping():
    ubit.uart = 'ping#'
    return True

def goodbye():
    ubit.quit_async()
    ubit.disconnect()
    return False

ubit.subscribe_uart(print)
eloop.add_timer(10000, ping)
eloop.add_timer(30000, goodbye)

ubit.run_async()
```

ukBaz, 2019-05-06:
Fixed with b31509d5c12c6e51d655681418fb8beecf06bfbb

closed 2019-05-06 by ukBaz