ukBaz/python-bluezero #227: send data to the micro:bit from raspberry via UART BLE service

xiaolongchengithub, 2019-05-06:
from time import sleep
from bluezero import central
from bluezero import tools

UART_SRV = ''''''''''''''''''''''
UART_tX = '''''''''

ubit = central.Central(adapter_addr='.....',device_addr='......')
sendData = ubit.add_characteristic(UART_SRV,UART_TX)

ubit.connect()
while not ubit.services_resolved:
     sleep(0.5)
     ubit.load_gatt()
while True:
    sendData.value  = "hello".encode("gkb")
    time.sleep(5)
ubit.disconnect()
################################################
hello,I do not know why i can not send data to the micro:bit . Are there some problem?Can you help me. Thanks

xiaolongchengithub, 2019-05-06:
and the micro:bit  only start the bluetooth uart service

xiaolongchengithub, 2019-05-06:
[image]

ukBaz, 2019-05-06:
According to the micro:bit [Bluetooth profile](https://lancaster-university.github.io/microbit-docs/resources/bluetooth/bluetooth_profile.html), if you want to send information from the RPi to the micro:bit you need to use the RX Characteristic. This is `6E400003-B5A3-F393-E0A9-E50E24DCCA9E`.

ukBaz, 2019-05-06:
I have started to do experiments with the UART service and done an initial commit:
8fc1e759613cf79b3084f89df738d8a3c1f81061

This was the code on the micro:bit
[image]

This was the code on the RPi:
```python
from time import sleep
from bluezero import microbit

ubit = microbit.Microbit(adapter_addr='02:00:AA:48:25:29',
                         device_addr='F1:55:90:65:29:DC',
                         accelerometer_service=False,
                         button_service=False,
                         led_service=False,
                         magnetometer_service=False,
                         pin_service=False,
                         temperature_service=False,
                         uart_service=True)

ubit.connect()

for i in range(5):
    ubit.uart = 'ping#'
    sleep(10)

ubit.disconnect()
```

ukBaz, 2019-05-06:
I believe this to be fixed with #225 
If you have issues please open a new issue. Thanks.

closed 2019-05-06 by ukBaz

xiaolongchengithub, 2019-05-07:
It is ok.  Thank you very much.