ukBaz/python-bluezero #231: Is bluezero used of window 10

xiaolongchengithub, 2019-05-07:
Hello,Thank your help.I want to use your library for windows.I need modfy ?

ukBaz, 2019-05-07:
Bluezero library is for Linux only. Primarily for Single Board Computers like the Raspberry Pi.

There are some libraries out there that are targeted at Windows and some that are targeted to work cross platform such as:
https://github.com/TheCellule/python-bleson

As this request is outside the scope of this library I am going to close.

closed 2019-05-07 by ukBaz

ukBaz, 2019-05-07:
If you don't mind it not being Python then Web Bluetooth has the best cross platform support
https://github.com/WebBluetoothCG/web-bluetooth/blob/master/implementation-status.md

xiaolongchengithub, 2019-05-07:
oh,I seee. thank you