ukBaz/python-bluezero #232: Is there a plan for bluetooth mesh support?

david-foo, 2019-05-23:
Is there a plan for bluetooth mesh support in this project?

ukBaz, 2019-05-23:
Mesh is something that I would like to support with the library. The timing of that support is largely driven by the availability of hardware and what is available in BlueZ on Linux.

Is there something specific you are looking for?

The BlueZ API is available at:
https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/mesh-api.txt

Do you know what (commonly available) hardware support Bluetooth Mesh?

david-foo, 2019-07-23:
@ukBaz   I have do someting in zephyr and mynewt, both of which support mesh, and the commonly hardware would be micro:bit(NRF51822 16k RAM size) and nRF52840.  The bluetooth SIG group use micro:bit to demostrate the mesh technology. But micro:bit is short in RAM size, 16K is not enough if you want to use it in real product.  Also I have look into bluez with its meshctl tool. I fell the bluez api is hard for beginner. If you are going to develop a python version it must be a great thing.

ukBaz, 2019-07-23:
Hi @david-foo,

What I would like before I start looking at developing the Python for BlueZ, is a hex file that I can drop on the micro:biti to turn it into a mesh node. Ideally it would be something very simple, maybe representing a light bulb so it would turn the LEDs on and off. This would allow development on the Raspberry Pi to control those light bulbs.

Do you know of the existence of a micro:bit hex file like that?

https://www.bluetooth.com/blog/the-fundamental-concepts-of-bluetooth-mesh-networking-part-1/
https://www.bluetooth.com/wp-content/uploads/2019/04/1903_Mesh-Models-Overview_FINAL.pdf

david-foo, 2019-07-23:
Hi @ukBaz  So you neet hex file that could turn micro:bit into mesh nodes, is that right? If so, I can show you these code and hex file. I will create a git repo and let you know. Just wait  a moment.

ukBaz, 2019-07-23:
>  So you need hex file that could turn micro:bit into mesh nodes, is that right? 

Yes, that is what  am looking for.

david-foo, 2019-07-23:
@ukBaz    Hi, see this repo https://github.com/david-foo/mesh_sw_lt

ukBaz, 2019-07-23:
Thanks. I was hoping to not have to install the Zephyr tool chain. If I use some of the other micro:bit tools they create a hex file that I can drag-n-drop onto the micro:bit. Is it possible for you to create something like that?

david-foo, 2019-07-23:
@ukBaz   You just want hex file, not want to build it form source code?

ukBaz, 2019-07-23:
I currently don't have the Zephyr tool-chain setup. . I have limited time for this development and so I don't want to waste any time setting that up.
I just want an easy way to create some mesh nodes that I can interact with from the Raspberry Pi.

david-foo, 2019-07-23:
@ukBaz  understand. I will compile the switch and light hex file for you. you can find it in the same place https://github.com/david-foo/mesh_sw_lt. Just rename the file suffix.

ukBaz, 2019-07-23:
Excellent!  Exactly what I have been looking for.

Just to check my understanding, I am provisioning it with the nRF mesh app.  
I am following this tutorial for the app:
https://youtu.be/XthbU9NP0Yg

I am having some issues provisioning your Embest_LT_01. Can you try provisioning a node with this app and let me know the basic steps you followed.

Thanks.

david-foo, 2019-07-24:
@ukBaz  Just follow this video. The process can be dvided into two part. First, provision, it is the common step that every mesh node have to do the same step. Just be noted the OOB code you input nRFmesh app must be 1234. I have fixed this to 1234 for debug. In real scene it should be random OOB number. Second, configuration, you need to bind a APPKEY for both the server and client then add a pub address and let the server to sub this address.  
If every thing goes well then you can press button A and B and see the msg form the lt board's uart console.
ur~ uart console is needed.

ukBaz, 2019-07-24:
Thanks @david-foo. Knowing the OOB code is 1234 has got me further.

[image]

I have not been able to connect to the micro:bit via the serial interface using [putty](https://microbit-micropython.readthedocs.io/en/latest/devguide/repl.html) so it is difficult for me to see if anything is working.

Would it be possible to change things so there is feedback via the LEDs on the micro:bit?

david-foo, 2019-07-25:
@ukBaz   Hi, I have add the feature that you can use button 'A' and 'B' on sw to turn on or off lt's LED array. What's the name of the APP you show me?

ukBaz, 2019-07-25:
Thanks for the update. I'll take a look at it next week.

The app is the nRF mesh app from the video I shared previously.
It is available for Android:
https://play.google.com/store/apps/details?id=no.nordicsemi.android.nrfmeshprovisioner&hl=en_GB
And iOS:
https://apps.apple.com/us/app/nrf-mesh/id1380726771

david-foo, 2019-07-26:
I have install nRFmesh APP but I have not seen the picutre your show me. May be my verstion is too low? I will try the link. Have a noce weekend.

ukBaz, 2019-07-26:
I did have a quick look at the update you provided yesterday and it is working great. Thank you. 

There are a lot of steps on the app and some of them are not obvious. I'll try to document so we both are using the same process. Are you on Android or iOS?

ukBaz, 2019-07-27:
Hello @david-foo,

The micro:bits forget their provisioning data so they have to be provisioned every time they are switched on. I have not discovered how to get the nRF mesh app to forget its previous settings so it might be a little confusing that it shows the sw and lt already on there but they need to be provisioned again.

I have done a video when I was provisioning the micro:bits. It is available at:
https://youtu.be/924j8zJRonM

I hope that makes things a little clearer. If it doesn't then let me know.

If you don't have the the "liviing room" group, you need to press the "+" to create it.

david-foo, 2019-07-29:
@ukBaz  Thank you. your video answered my confuse. I'm on iOS. It seem the Adriod nRFmesh app have more freatures than the iOS one.  micro:bit can not keep the provitioning and configure data for its shortage of RAM. In iOS nRFmesh app there is a 'Reset Mesh State Forget Network' button under Setting tag which can reset the app to initial state.
[image]

ukBaz, 2019-07-29:
Thank you @david-foo. I have now found how to clear the previous stored details in the app.

[image]

I had a quick look at mesh in BlueZ and it appears to have not been in any releases as yet. There was also a number of patches submitted last week to the repository about mesh functionality.
I'll checkout master and see what I can do but we might be a little too early for using BlueZ. It is likely to be unstable and buggy.

david-foo, 2019-07-30:
@ukBaz  I aggre with you. the mesh features of bluez is now under development. The latest released bluz that support mesh is tag v5.50 on the official repository.  In this release a tool named meshctl is implemented under /mesh directory. meshctl is aimed at a tool for provition and configure(just like nRF mesh APP) over GATT proxy. In the master branch meshctl is moved to /tools/mesh and a new /mesh dirrectory is created, by which I think is to implement normal mesh node features and provide APIs for upper layer than just a GATT proxy provitionner.  meshctl is a shell command-line tool for Human-computer interaction just like other bluetooth tools like hcitool. It seems meshctl is mature and stopped developing. So may be I can reuse meshctl's code to provide a c based lib for linux application which just want to be a mesh network manager. What do you think?

ukBaz, 2019-07-30:
I have BlueZ v5.50 on my laptop which is a stock install of Ubuntu 19.04. It doesn't seem to have meshctl installed by default. I haven't tested the latest download of Raspbian which has BlueZ v5.50 by default. Do you know if that has meshctl easily available?

Bluezero for GATT relies on people using bluetoothctl to pair between devices. So with mesh my thought was that people would use meshctl (or the nRF app) to provision devices and Bluezero would focus on enabling people to create mesh clients or servers.
To this end I was going to look at creating a simple on/off client as a first step. Something like:
https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/test/test-mesh#n1022

Does that seem like a sensible way forward?

ukBaz, 2019-07-30:
FYI:
Today on the BlueZ mailing list some information was posted about the plans for meshctl:
https://marc.info/?l=linux-bluetooth&m=156447001828284&w=2

ukBaz, 2019-08-02:
Hello @david-foo,  I've cloned the BlueZ master branch and tried to do a few things on this. It hasn't gone well. I think I'm going to close this ticket as it feels too early for Bluezero to be using such bleeding edge functionality.
I will come back to this in the future when BlueZ functionality has stabilised as I mesh is of interest. 
In related news, I have just ordered a [nRF52 Adafruit Feather](https://docs.zephyrproject.org/latest/boards/arm/nrf52_adafruit_feather/doc/index.html) to experiment a little bit more with mesh networks.
Thanks for your help and sorry I'm not able to help at this time.

closed 2019-08-02 by ukBaz

hirenumradia, 2020-02-15:
Hey @ukBaz, There are no real options to work with python and BlueZ at the moment when it comes to bluetooth mesh. I would love to help out and contribute to helping getting a basic implementation working. 

BlueZ has now got a mesh API, https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/mesh-api.txt

Would you be open to having a chat about this? Im from the UK too so might be easy to co-ordinate.

ukBaz, 2020-02-15:
Hello @hirenumradia,
I am always open to having a conversation about implementing Mesh as it seems like an exciting development in Bluetooth. My concern is the maturity of the BlueZ implementation. My DM's are open on Twitter if you want to arrange a call.

ukBaz, 2020-02-17:
There is a discussion happening on the BlueZ developers mailing list that I thought might be relevant to to this discussion:
[mesh: Send mesh model message to group address  ](https://marc.info/?t=158179139700001&r=1&w=2)

In the thread it says about there being an example of a python implementation of on/off client and server available at https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/test/test-mesh

hirenumradia, 2020-02-17:
Nice! I will investigate and reply back to the thread.

hirenumradia, 2020-02-17:
So the D-bus service that is responsible for working with mesh is called "bluetooth-meshd".

They refer to a test python script that interfaces with the service which can be seen here:
https://github.com/hadess/bluez/blob/6de4bdb957cdc85d89851420ab06ca8e226f8d4e/test/test-mesh

Please note that this is a git repo that is a mirror of the bluez git repo.

Another small "bluetooth-meshd" example can be seen in this repo:

https://github.com/SilvairGit/bluetooth-meshd-example

I am going to have a play around to see if I can send some basic commands on/off commands using a python script.

RAlexeev, 2022-06-10:
@ukBaz @hirenumradia @david-foo Hi everybody!
Could you share a bit your success and thoughts about BLE mesh network, using BlueZ stack and Raspberry PI board? Is it possible to create a fully functional BLE mesh node, using BlueZ stack on Linux? I wonder about BLE mesh gateway architecture and its implementation. Whether it's possible to realize BLE mesh gateway using current BlueZ stack implementation on kinda Raspberry PI board, without connecting any extra BLE chips, merely using, for instance, internal Cypress CYW43455 chip...
Thank you in advance for any thoughts and recommendations.

ukBaz, 2022-06-11:
@RAlexeev, 
Bluetooth Mesh makes heavy use of AEAD-AES_CCM encryption, which is implemented in the Linux kernel on most platforms. I haven't checked lately, but this was not enabled by default on Raspberry Pi so required a rebuild of the Kernel for BlueZ mesh. This level of set-up makes it a show-stopper for this library and its target audience.
While BlueZ seems in a better state now, it still feels too early for Bluezero to be using such functionality.
Bluetooth Mesh is still of interest. 
When Raspberry Pi OS is shipping with a BlueZ version and kernel that has support for mesh out of the box then that will be the right time for this library to look at support.

RAlexeev, 2022-06-11:
Many thanks for your expanded explanation!
Generally speaking, if we build Linux kernel for our SoM, so we can skip this blocker. In fact, we would like to use our own boards. Raspberry Pi was as a well-known example. I thought that if it's workable with Raspberry Pi and BlueZ stack, then we will be able to launch the same on our custom boards.
I worked with Bluetooth mesh 1,5-2 years ago, creating end devices on nRF52840. But that time we swtiched from Bluetooth mesh, since its weak spreading in a market (except propietary solutions with vendors' provisioning) and difficulties with BLE mesh gateways. And right now I would like to understand about ability to support Bluetooth mesh on boards with Linux/AOSP. Did I understand you correctly, that if we rebuild Linux kernel with appropriate flags to support AEAD-AES_CCM encryption and use the last BlueZ stack release, then it's possible to create a BLE mesh node from Raspberry Pi without connection any extra BLE modules? And we will be able to directly communicate from Raspberry Pi with other BLE mesh nodes, won't we? The idea is to use it as a gateway between WiFi and BLE mesh networks without physical connection any extra Bluetooth LE modules and any proxy nodes.