ukBaz/python-bluezero #233: iOS will not disconnect from peripheral

mhazley, 2019-06-06:
This is not an issue with Bluezero (I *think*), I am just quite lost in looking for a solution to my problem so wanted to ask here in case anyone had any experience or can came across a similar issue? 

I use Bluezero with BlueZ 5.49 to run a simple peripheral on an RPi - it is very similar to the `cpu_temperature` example...

```
# Setup bluezero
self.app = localGATT.Application()

# Setup Auth Service
self.auth_srv = localGATT.Service(1, AUTH_SRVC, True)

# Setup Auth Characteristics
self.auth_chrc = GenericChrc(self.auth_srv, AUTH_AUTH_CHRC, "Auth Characteristic", ['read', 'write', 'notify'], 2)
self.auth_chrc.service = self.auth_srv.path

# Add everything to the GATT application
self.app.add_managed_object(self.auth_srv)
self.app.add_managed_object(self.auth_chrc)

# Create GATT manager and register gatt application with it
self.srv_mng = GATT.GattManager(adapter.list_adapters()[0])
self.srv_mng.register_application(self.app, {})

# Set up the BLE adapter
self.dongle = adapter.Adapter(adapter.list_adapters()[0])
self.dongle.pairable = False

# Create the BLE advertisement
advert = advertisement.Advertisement(1, 'peripheral', name)

# Tell advertisement to advertise the Service UUID
advert.service_UUIDs = [AUTH_SRVC]

# Power up and register advert
if not self.dongle.powered:
    self.dongle.powered = True

self.app.start()
```

Full disclosure, I forked about a year ago to add support for the `LocalName` property in Advertisement (I should probably pull request this back - I need to work out how). I notice that development has picked up again in the last few months but I do not think it is in this area? 

So, everything above works _perfectly_ with Android... 

Recently though, I have started using iOS as well and while an iOS device can connect and interact with my characteristics fine, it will not disconnect and it will only drop the connection if I turn bluetooth off on the device. 

The app on the device gets the disconnect callback from iOS, so the app thinks that it has disconnected, but the device hasn't.

Is anyone aware of anything I could be setting incorrectly that is causing iOS to hang onto the connection? 

Or could it be my side that is hanging onto the connection?

Is there a way for me to terminate the connection with the central on my side??

ukBaz, 2019-06-06:
Hi @mhazley,

Not sure I have anything obvious that I know of.  A bit of searching around has turned up a few other people with (what to me) sound like similar issues:
https://devzone.nordicsemi.com/f/nordic-q-a/7090/disconnected-peripheral-reconnects-direct-in-ios
https://stackoverflow.com/questions/30212820/corebluetoothdisconnect-peripheral-connection-from-application

The only thing that comes to my mind is that the RPi often advertises a number of Primary services. Before you connect, what services is it advertising? Just yours?  Or is it advertising audio sink (for example) also?
Based on those links above, could it be that one of the other services is maintaining the connection even after your app requests a disconnect?

mhazley, 2019-06-07:
Hi @ukBaz - thanks for your reply!

The advertisement looks like the following:

https://user-images.githubusercontent.com/6282802/59098629-b6c38c80-8918-11e9-80c6-09dd70c91cb0.png

Looking at this now - this is advertising as Classic & LE which could be causing it to behave differently.

I actually am starting to think that you might be onto something with the services too. 

I managed to get my hands on a BLE Sniffer today and have traced the connection - there seems to be some other services discovered apart from mine:

[image]

Going to keep digging today - do you mind me leaving this open and I can report back - might be good to have a reference for future?

ukBaz, 2019-06-07:
Thanks for the update. Yes, please leave this open for updates. It will be good to learn if there is something we can do in the library.

Re advertisement.
The  Classic can be controlled with /etc/bluetooth/main.conf
Some more detail at:
https://github.com/ukBaz/python-bluezero/blob/master/docs/install_bluez.rst#switch-controller-to-bluetooth-low-energy-only

It will also be interesting if you could open bluetootchctl in another terminal. Then while your program is advertising do `show`. That should also give a list of what services your adapt has available for clients to connect to.

Good luck!

mhazley, 2019-06-07:
Hey - thanks for your input on this - I managed to get it to LE only and iOS disconnected totally fine after that. 

It is very odd though, it seems that if I change the `main.conf` to have `ControllerMode = le` and then run `sudo service bluetooth restart` it works fine, but if I reboot the device, the config persists, but I am back to CLASSIC and LE until I restart the service.

For completion, the list of services from `bluetoothctl` were:

```
        UUID: Headset AG                (00001112-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: Vendor specific           (a5360000-b536-c536-d536-e536d536e536)
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
        UUID: Handsfree Audio Gateway   (0000111f-0000-1000-8000-00805f9b34fb)
```

I hadn't been aware of those until now, so its good knowledge to have. 

So yeah - trying to work out how to get this to persist now, but I will close this - thanks!

closed 2019-06-07 by mhazley

ukBaz, 2019-06-07:
Thanks for the update and glad to hear you've made progress.

You might want to have a look at compiling Bluez without the a2dp and avrcp plugins. I've not tried it but it might be that because those are enabled, it overwrites the settings. More information at:
https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/README#n139

mhazley, 2019-06-07:
Thanks for the tip!

mhazley, 2019-06-11:
@ukBaz - sorry to awaken you on this closed thread but I have been looking into this a bit more to try and understand if theres anything I could do to my advertisement to make it appear as LE Only. 

I've been reading the specification and noticed that there is a Flags data type that contains one bit Boolean flags that can be included in the advertising data. 

Looking up the possible values of those flags ([Supplement to the Bluetooth Core Specification, CSSv8](https://www.bluetooth.org/docman/handlers/downloaddoc.ashx?doc_id=457081), page 12) I can see that there are the following types:

```
Bit 0 LE Limited Discoverable Mode
Bit 1 LE General Discoverable Mode
Bit 2 BR/EDR Not Supported. 
```

I don't see flags being supported on BlueZ API but I do see this:

```
dict Data [Experimental]

			Advertising Type to include in the Advertising
			Data. Key is the advertising type and value is the
			data as byte array.

			Note: Types already handled by other properties shall
			not be used.

			Possible values:
				<type> <byte array>
				...
```

Do you have any experience using the above?

ukBaz, 2019-06-11:
Sorry, no experience as I think that is only in the recent versions of Bluez.  My instinct is that It doesn't sound like the right thing. 
Away from my laptop at the moment so it is difficult to investigate. 
Not sure it is the type of thing they are likely to be in the dbus API.

This kind of thing is more in the mgmt API. 
https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/mgmt-api.txt#n2591
That API is not easily reachable from Python. (Although some projects have managed)

Do you want to open this as a new issue about setting the flags?

mhazley, 2019-06-11:
Let me inspect the packet some more when I get back in front of my own computer and I will see if the flags are in the raw data - if they are and are different in each case, I'll raise a separate ticket!

mhazley, 2019-06-11:
@ukBaz I think you were spot on with the mgmt API.

This article outlines using the btmgmt tool which can be build from bluez source so I will try that.
https://urbanjack.wordpress.com/2014/06/05/how-to-set-bluez-into-ble-or-le-only-mode-ibeacon/

I want to try the flags as well but I think that it will not do what I want and is likely more indicative. 

Thanks once again for the signposting!

ArMouReR, 2020-12-16:
Hi,
Sorry for popping up this old thread.
But, I am facing exactly the same problem as you have described: 
iOS do not disconnect from my peripheral (RPi3) it only drops the connection.
I also see that setting ControllerMode = le doesn't work at all in my case, the Pi always advertising as CLASSIC and LE...
Have you found the solution you problem ?

mhazley, 2020-12-16:
Hi @ArMouReR yes, I found my way around this using the mgmt API. We even got the student in our team to turn the btmgmt tool into a python library so we could do it from within our codebase.

Check out:

https://github.com/BOJIT/btmgmt

There's example code there to demonstrate setting LE Only:

https://github.com/BOJIT/btmgmt/blob/master/tests/set_LE.py

ArMouReR, 2020-12-16:
Hi Matt,

Thanks for prompt reply ! Great library. tried installed and it works.
Now my RPi3 is advertising as LE only, but the main problem still exist:
Disconnect in iOS app do not actually disconnect from RPi3.
Do you remember what helped you with this issue ?

mhazley, 2020-12-16:
Sorry that solved my issue I am afraid!

ukBaz, 2020-12-16:
@ArMouReR I'd suggest your run `sudo btmon` in another terminal window when you do the disconnect and see what it reports.

You might get better help from the BlueZ project directly. Contact details at: http://www.bluez.org/contact/

I'll post the link to your Stack Overflow question here for future reference:
https://stackoverflow.com/q/65325640/7721752

mhazley, 2020-12-16:
One thing to note @ArMouReR , we found iOS is really weird in the Bluetooth world... It seems to cache so so much about a device under the hood that sometimes we'd have to get RPi to change mac address to see a change in iOS. Try turning on privacy via the mgmt API to get a random mac adress and see if that makes iOS behave!!

ArMouReR, 2020-12-16:
@ukBaz
Thanks for stepping in ! I will definitely try to contact developers of BlueZ.
Regrading btmon - it absolutely quiet when I do disconnect in iOS and the same on dbus monitor.
I have another setup where I use pybleno (works via hci and based on bleno) and it works almost perfectly till the moment I start heavy OpenCV processing. So, when processing is running pybleno can't connect/mantain? the link.
So, when pybleno server is disconnect I see the commands in btmon. With BlueZ it absolutely quiet.

@mhazley
Thanks for suggestion. I will try it. I have downloaded the latest rasberian image and was thinking to do some fresh start as last measure to solve the issue, so it will be the last thing to check before fresh start...

In last hour I was playing with adapter settings
<
    adapter_props.Set("org.bluez.Adapter1", "Pairable", dbus.Boolean(0))
    adapter_props.Set("org.bluez.Adapter1", "Discoverable", dbus.Boolean(1))
>
And it changed something, I don't see pairing request anymore, but sometimes after connect I see in btmgmt  that BLE stuck in authorization loop. Very strange...