ukBaz/python-bluezero #234: Issues sending UART commands from RPi 3

ARStark, 2019-06-06:
Hi, I've been using the bluezero library and I've looked at some similar issues but for some reason I'm still having problems sending commands from my Raspberry Pi 3 to my microbit. 

this is the code running on my microbit ( i apologize for the formatting. I was having issues) 

void showConnected()
{
    uBit.display.scroll("C");
}

void showDisconnected()
{
    uBit.display.scroll("D");
}

void onConnected(MicroBitEvent e)
{
    showConnected();
    isConnected=true;

    // my client will send ASCII strings terminated with the colon character
    ManagedString eom(":");

    while (isConnected == true) {
        uBit.display.scroll("here"); //used to to check Rpi was getting inside loop
        ManagedString message = uart->readUntil(eom);
        uBit.display.scroll(message.length()); //used to see if message was being received at all.
        
        if (message.length() == MESSAGE_LENGTH_ON) {
            if (message.substring(MESSAGE_START_INDEX ,MESSAGE_END_INDEX_ON) == "on") 
            {    
                row = charToInt(message.charAt(MESSAGE_ROW_INDEX_ON));
                col = charToInt(message.charAt(MESSAGE_COL_INDEX_ON));
                
                if(isdigit(row) && isdigit(col))
                { 
                    row = intToMatrixRow(row);
                    col = intToMatrixCol(col);
                    turnLedOn(row, col);
                    
                } // end if isdigit(row) & isdigit(col)
                
            } // end if message.substring == "on"
            
        } // end of if message == 6
        
    } // end while isConnected

//other code below here that is running but unimportant. removed for sake of post

} // end onConneted()

void onDisconnected(MicroBitEvent e)
{
    showDisconnected();
    isConnected=false;
}// end onDisconnected

and here is the code running on my Raspberry Pi

#!/usr/bin/env python

from bluezero import microbit

ubit = microbit.Microbit(adapter_addr='B8:27:EB:E8:AD:4D',
                         device_addr='FA:33:DE:18:91:F6',
                         accelerometer_service=False,
                         button_service=False,
                         led_service=False,
                         magnetometer_service=False,
                         pin_service=False,
                         temperature_service=False,
                         uart_service=True)

ubit.connect()

ubit.uart = 'on 2 2:'

ubit.disconnect()

Now the code running on my microbit works as intended if i connect my phone (iphone X running 12.1 using BluetoothLE app) and use a bluetooth terminal app. But messages from my raspberry pi aren't being sent. When I run my python code, i see the C and D come across the microbit as expected but the uart command isn't working. do i need to use the subscribe_uart function? if so, I'm a little confused as to what it exactly does and how to utilize it correctly. Once again I'm sorry for the terrible formatting of my code.

ukBaz, 2019-06-07:
Let's assume the code on the micro:bit is correct as it works with the Bluetooth Terminal App. So let's focus on the RPi code as the possible source of the problem.
I don't have hardware around me at the moment, but a couple of experiments I would run are:
1) Have you tried putting a `sleep(5)` in your code after the `ubit.uart = 'on 2 2:` line? I wonder if the `D` on disconnect is taking over the scrolling of the uart information.
2) My assumption from the shebang line (`#!/usr/bin/env python`) of your python script is that you are using Python 2. Have you tried running with Python 3?   (`#!/usr/bin/env python3`) 

If you only want to write information to the micro:bit uart then you do not need to subscribe. That is only to get notification when the micro:bit writes to the uart.

As a side note, I've corrected the formatting of your code above. For future reference this is how you do it:
https://help.github.com/en/articles/creating-and-highlighting-code-blocks

ARStark, 2019-06-07:
Suggestion 2 is what made the code work. Thank you so much! Once i finish the project I'm working on (nothing too crazy) I'll post what I used your library for somewhere on here for you to see!

ukBaz, 2019-06-07:
Excellent, glad it helped.  I'll close this issue now.

Look forward to hearing about your project. You can either add something here, a new issues, or I have created a wiki page to track these things.
https://github.com/ukBaz/python-bluezero/wiki/Built-with-Bluezero---Projects-using-Bluezero
You should be able to add things to the page.

closed 2019-06-07 by ukBaz

ARStark, 2019-06-09:
Hey! 

 I was going to post the project on the wiki but then I realized, its more of a proof of concept rather than a project. But [here](https://youtu.be/QuNLYXvboVI) is a link to a temporary demo video of my home automation project I was working on. I will make a detailed video once I get a control relay and set it up with a lamp or fan. The video is a little of a minute long so you should check it out! 

Also [here](https://github.com/ARStark/Proof-Of-Concept/tree/master/Home-Automation) is a link to where the code is housed, if you're curious as to what that looks like and how I utilized your library (even though I'm sure you already know lol). 

Thanks again for your help. This won't be the last time I use the library, so I'll probably have more questions for you in the future.

ukBaz, 2019-06-09:
Very interesting!  Thanks for sharing.