ukBaz/python-bluezero #235: Do peripheral characteristic able to notify only string value?

yoko-chance, 2019-06-13:
I implementing perihelial BLE device.

I want send notify event when modify characteristic value.

It characteristic value is number, but I seem to only allow string for `bluezero.peripheral.Characteristic.send_notify_event()` value argument.

Because `bluezero.dbus_tools.str_to_dbusarray()` suppose string it given argument.

So I wrote below code.

```
        self.characteristic.WriteValue(value, options={})
        self.characteristic.PropertiesChanged(constants.GATT_CHRC_IFACE,
                               {'Value': int_to_uint32(self.characteristic.ReadValue(options={}))},
                               [])
        # instead of th below code
        # self.characteristic.send_notify_event(value)
```

I think not enough cover various use cases with current `bluezero.peripheral.Characteristic.send_notify_event()` method.

https://github.com/ukBaz/python-bluezero/blob/v0.2.0/bluezero/peripheral.py#L655

I think the following code is better.

```
        elif isinstance(self.value, str):
            # print('Notify: ', dbus_tools.str_to_dbusarray(self.value))
            self.PropertiesChanged(
                constants.GATT_CHRC_IFACE,
                {'Value': dbus_tools.str_to_dbusarray(self.value)},
                [])
        else:
            self.PropertiesChanged(
                constants.GATT_CHRC_IFACE,
                {'Value': self.value},
                [])
```

Is it restriction for something or I missed something?

Thank you implemented peripheral module, I wanted this!

ukBaz, 2019-06-13:
Thanks for the feedback and a good catch. I do not have access to hardware at the moment so I cannot test what you have sent for a few days. 

The data for PropertiesChanged needs to be a byte array so an "else" is maybe not the best option. My initial thought is that `send_notify_event` should take a string or byte array. It would need to test accordingly. Then your code using the peripheral could be:
`send_notify_event(int_to_uint32(my_value))`

Does that work?

yoko-chance, 2019-06-14:
Oh.. my idea was lazy.
Sure, it better than before one.

I created pull request from youer idea.
https://github.com/ukBaz/python-bluezero/pull/236
It worked on my device.

However, I have concerns.
Tests do not work on my environment.(rase `KeyError: 'org.bluez.Device1' `)
maybe i not doing something.
Are there documents that can be helpful?

ukBaz, 2019-08-02:
The PR has been merged so I'm closing this up

closed 2019-08-02 by ukBaz