ukBaz/python-bluezero #239: Test with Bluez 5.50 because it is the default in Buster

ukBaz, 2019-07-01:
Laterst versions Raspbian and Linaro are now shipping with OS based on Buster which has BlueZ 5.50 as their default version. Ensure Bluezero works with this release

https://itsfoss.com/debian-10-buster/
https://qa.debian.org/madison.php?package=bluez

mdxs, 2020-02-25:
While still experimenting and trying to see if this works for me, first result is that with a RPi 3B using bluezero 0.3.0 from PyPI there is some success using a remote Device via Central; I could get notification working for one Characteristic and write some value to another Characteristic.

ukBaz, 2021-01-01:
Testing is being done against BlueZ 5.50 now so this can be closed

closed 2021-01-01 by ukBaz