ukBaz/python-bluezero #241: unable to access bluezero module using Python 3

MrsHesse, 2019-07-23:


ukBaz, 2019-07-23:
Hello,

Sorry to hear you are having issues with the install.

If you have a recent version of Raspbian then you should not need to do step 1 above. The version of BlueZ that Bluezero is compatible with was chosen to be compatible with it. Kudos for doing the extra work.
I think your issue is because you are using `pip` which as you note is installed for use with `python2`. If you want to use `python3` then using `pip3` is what is required.

Better instructions to follow might be:
https://ukbaz.github.io/howto/ubit_workshop.html

Let me know if using pip3 solves the issue.

Cheers,
Barry

MrsHesse, 2019-07-24:
Thanks for getting back to me.  I realised after I sent the email that the
pizero in question is old - hence having to do all the updates.  I am still
new with all the python package installation process and I missed that you
used pip3 in the workshop instructions.

Using *sudo pip3 install bluezero* removed the *ImportError: No module
named 'bluezero'* error but I then received an *ImportError: No module
named 'dbus'* error.

sudu pip3 install dbus-python prevented that error.

Now the first piece of code from the workshop is throwing errors from  the
call to ubit.connect()...  but that is a problem for another day.

Thank you for your quick response and help.

G.

-- 
*Gillian Hesse*
*Computer Science Teacher and 9 Elm Form Tutor*
Grey Court school
Tel: 020 8948 1173

*I am currently reading... Educated by Tara Westover*

ukBaz, 2019-07-24:
Good to hear that we are making progress.

If possible, I would suggest that updating the RPi Zero to a more recent version of Raspbian has many benefits. If doing an update of the operating system is not practical, then I still have the instructions from back in the day when upgrading BlueZ was required. Those can be found at:
https://github.com/ukBaz/python-bluezero/blob/master/docs/install_bluez.rst

For your latest issue, I wonder if you need to install the system files for Python3 DBus with:
`sudo apt-get install python3-dbus`

I'll leave this open for the time being so you can let me know how it goes.

Good luck!

MrsHesse, 2019-07-26:
Hi
Thanks for the suggestion,  I have upgraded the pi zero to *stretch *and
then followed the workshop instructions and confirmed that Python3 DBus
files are installed but still get an error when I run the first workshop
example (sending message to microbit) I get the following error.

[image: code-output.PNG]

Scanning with bluetoothctl shows that the microbit is being seen
[image: bluetooth scan.PNG]

I have  used *"B8:27:EB:71:EF:95"* as my adaptor address and
*"F4:CA:40:CE:FC:08"* as my device address.

Any suggestions would be appreciated.

Thanks

Gillian

-- 
*Gillian Hesse*
*Computer Science Teacher and 9 Elm Form Tutor*
Grey Court school
Tel: 020 8948 1173

*I am currently reading... Educated by Tara Westover*

ukBaz, 2019-07-26:
The images have not come through on GitHub so I can't see what you have done.

However, let us divide this problem up to see if it is an issue with Python or the Raspberry Pi/micro:bit.

My plan is to use `bluetoothctl` to establish that the Bluetooth communication is working OK between the micro:bit and the Raspberry Pi.

In the `bluetoothctl` window after you have done the pairing. Connect to the micro:bit. (I'll use `E3:AC:D2:F8:EB:B9` but you would use `F4:CA:40:CE:FC:08`)
```
[bluetooth]# connect E3:AC:D2:F8:EB:B9
```
Once it has successfully connected, open another terminal window on the RPi and type this command  (this is to find out where the information about the micro:bit button B is stored in the Bluetooth system):
```
gdbus introspect --system --dest org.bluez --object-path / --recurse | grep -i -B17 DA91 | grep node
```
You should get an answer that looks something like:
```
            node /org/bluez/hci0/dev_E3_AC_D2_F8_EB_B9/service002b/char002f {
```
We will now take this path information back into `bluetoothctl` so we can read it in there. So back in the first terminal window where `bluetoothctl` is running and connected to the micro:bit. We will select the attribute using the path we just found:
```
[BBC micro:bit [tuzep]]# select-attribute /org/bluez/hci0/dev_E3_AC_D2_F8_EB_B9/service002b/char002f
```
Now reading the attribute it should return zero:
```
[BBC micro:bit [tuzep]:/service002b/char002f]# read
Attempting to read /org/bluez/hci0/dev_E3_AC_D2_F8_EB_B9/service002b/char002f
[CHG] Attribute /org/bluez/hci0/dev_E3_AC_D2_F8_EB_B9/service002b/char002f Value: 0x00
  00
```
If you press and hold the Button B on the micro:bit down and at the same time do a read you will get a non-zero value (1 = short press & 2 = long press)
```                                               .
[BBC micro:bit [tuzep]:/service002b/char002f]# read
Attempting to read /org/bluez/hci0/dev_E3_AC_D2_F8_EB_B9/service002b/char002f
[CHG] Attribute /org/bluez/hci0/dev_E3_AC_D2_F8_EB_B9/service002b/char002f Value: 0x02
  02       
```

If this works then, then things are setup correctly and we can focus on the Python. If there are issues with the above then we need to focus on that.

MrsHesse, 2019-07-28:
Thanks for getting back to me.  I will sit down and have a look at this
tomorrow.

-- 
*Gillian Hesse*
*Computer Science Teacher and 9 Elm Form Tutor*
Grey Court school
Tel: 020 8948 1173

*I am currently reading... Educated by Tara Westover*

MrsHesse, 2019-07-29:
Hi

Connecting in bluetoothctl worked
[image]

But the gdbus command failed.  So I looked at each step in the pipe seperately

`gdbus introspect --system --dest org.bluez --object-path / --recurse`

produced the output
[gdbus.log](https://github.com/ukBaz/python-bluezero/files/3441357/gdbus.log)

Adding the next pipe
gdbus introspect --system --dest org.bluez --object-path / --recurse | grep -i -B17 DA91
produced nothing

The gdbus output did contain references to my microbit as 
`gdbus introspect --system --dest org.bluez --object-path / --recurse | grep  -i dev_F$`

produced the following output
[gdbus_dev_F4.log](https://github.com/ukBaz/python-bluezero/files/3441369/gdbus_dev_F4.log)

Thanks

Gill

ukBaz, 2019-07-29:
Excellent work providing the gdbus.log. That was very helpful!

The `DA91` is part of the UUID for the Button B state characteristic. I know this from looking at the Bluetooth profile specified at: https://lancaster-university.github.io/microbit-docs/resources/bluetooth/bluetooth_profile.html
We can see that the this characteristic is part of the Button Service `E95D9882-251D-470A-A062-FA1922DFA9A8`

If you look at the gdbus.log section for what Service UUIDs are available on the micro:bit we see that Button Service UUID is in there:
```
        node /org/bluez/hci0/dev_F4_CA_40_CE_FC_08 {
          interface org.freedesktop.DBus.Introspectable {
            methods:
              Introspect(out s xml);
            signals:
            properties:
          };
          interface org.bluez.Device1 {
            methods:
              Disconnect();
              Connect();
              ConnectProfile(in  s UUID);
              DisconnectProfile(in  s UUID);
              Pair();
              CancelPairing();
            signals:
            properties:
              readonly s Address = 'F4:CA:40:CE:FC:08';
              readonly s Name = 'BBC micro:bit [tigeg]';
              readwrite s Alias = 'BBC micro:bit [tigeg]';
              readonly u Class;
              readonly q Appearance = 512;
              readonly s Icon;
              readonly b Paired = false;
              readwrite b Trusted = true;
              readwrite b Blocked = false;
              readonly b LegacyPairing = false;
              readonly n RSSI = -72;
              readonly b Connected = true;
              readonly as UUIDs = ['00001800-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', 'e95d0753-251d-470a-a062-fa1922dfa9a8', 'e95d6100-251d-470a-a062-fa1922dfa9a8', 'e95d93af-251d-470a-a062-fa1922dfa9a8', 'e95d93b0-251d-470a-a062-fa1922dfa9a8', 'e95d9882-251d-470a-a062-fa1922dfa9a8', 'e95dd91d-251d-470a-a062-fa1922dfa9a8', 'e97dd91d-251d-470a-a062-fa1922dfa9a8'];
              readonly s Modalias;
```

But if I look for characteristics for that service to appear in the log I don't see them:
Using `grep -P -B1 "o.Service.*F4" gdbus.log`
Bluetooth special characteristic (service changed)
```
                  readonly s UUID = '00002a05-0000-1000-8000-00805f9b34fb';
                  readonly o Service = '/org/bluez/hci0/dev_F4_CA_40_CE_FC_08/service0008';
```
Pairing and over the air firmware update characteristic
```
                  readonly s UUID = 'e95d93b1-251d-470a-a062-fa1922dfa9a8';
                  readonly o Service = '/org/bluez/hci0/dev_F4_CA_40_CE_FC_08/service000c';
```
??????
```
                  readonly s UUID = 'e97d3b10-251d-470a-a062-fa1922dfa9a8';
                  readonly o Service = '/org/bluez/hci0/dev_F4_CA_40_CE_FC_08/service000f';
```
Bluetooth characteristic (model number string)
```
                  readonly s UUID = '00002a24-0000-1000-8000-00805f9b34fb';
                  readonly o Service = '/org/bluez/hci0/dev_F4_CA_40_CE_FC_08/service0013';
```
Bluetooth characteristic (serial number string)
```
                  readonly s UUID = '00002a25-0000-1000-8000-00805f9b34fb';
                  readonly o Service = '/org/bluez/hci0/dev_F4_CA_40_CE_FC_08/service0013';
```
Bluetooth characteristic (firmware device string)
```
                  readonly s UUID = '00002a26-0000-1000-8000-00805f9b34fb';
                  readonly o Service = '/org/bluez/hci0/dev_F4_CA_40_CE_FC_08/service0013';
```
Default micro:bit event service characteristic
```
                  readonly s UUID = 'e95d9775-251d-470a-a062-fa1922dfa9a8';
                  readonly o Service = '/org/bluez/hci0/dev_F4_CA_40_CE_FC_08/service001a';
```
Default micro:bit event service characteristic
```
                  readonly s UUID = 'e95d5404-251d-470a-a062-fa1922dfa9a8';
                  readonly o Service = '/org/bluez/hci0/dev_F4_CA_40_CE_FC_08/service001a';
```
Default micro:bit event service characteristic
```
                  readonly s UUID = 'e95d23c4-251d-470a-a062-fa1922dfa9a8';
                  readonly o Service = '/org/bluez/hci0/dev_F4_CA_40_CE_FC_08/service001a';
```
Default micro:bit event service characteristic
```
                  readonly s UUID = 'e95db84c-251d-470a-a062-fa1922dfa9a8';
                  readonly o Service = '/org/bluez/hci0/dev_F4_CA_40_CE_FC_08/service001a';
```

I am a little confused by what I have seen here. So I only have questions for you.

I am making the assumption that you have created the hex file on the micro:bit using the Makecode editor as suggested in the Bluezero workshop material. Is that correct?

What pairing mode did you have set?

Did you do the pairing process first? (press button A & B and reset etc)
After it did pairing, did you power cycle the micro:bit and then connect using bluetoothctl?

Maybe, in bluetoothctl, remove the device information for the micro:bit and start the process of pairing and connecting again. You remove device information in bluetoothctl with:
```remove F4:CA:40:CE:FC:08```

Let me know the answer from those questions and we can see where that gets us.

MrsHesse, 2019-07-30:
> I am making the assumption that you have created the hex file on the micro:bit using the Makecode editor as suggested in the Bluezero workshop material. Is that correct?
I am - I reloaded it again just to make sure I was using the correct code on the microbit.

> What pairing mode did you have set?
Just works pairing

> Did you do the pairing process first? (press button A & B and reset etc)
Yes - so the microbit display goes from  b image to the bluetooth icon then the barchart looking image.

> After it did pairing, did you power cycle the micro:bit and then connect using bluetoothctl?
Sorry - not sure what you mean by power cycle (turn it on then off? no).  
I usually do the pairing while I am doing a scan in bluetoothctl so I can see when the pi's bluetooth is able to see the microbit, then I quit the scan and connected to the microbit.

> Maybe, in bluetoothctl, remove the device information for the micro:bit and start the process of pairing and connecting again. You remove device information in bluetoothctl with:
> `remove F4:CA:40:CE:FC:08`
I used remove is disconnect and then reconnected, but still got the same results.  The new log (for the F4:.... device) was

[gdbus.30.07.19.log](https://github.com/ukBaz/python-bluezero/files/3446585/gdbus.30.07.19.log)

I really appreciate the time you are taking to look at this - and explain what you are looking for,  Working with Bluetooth is completely new to me.

Thanks

G.

ukBaz, 2019-07-30:
Thanks for sticking with this. The intent is for this library to be an easier on-ramp to people interacting with Bluetooth for the first time, for teachers like yourself, so if I can't get it working for you then I haven't done a very good job. 

I have done a quick test and it has given me a theory to try. While the bar chart is showing, do you do a `pair F4:CA:40:CE:FC:08` or a `connect F4:CA:40:CE:FC:08`

If I `connect` when the bar chart is showing I get similar UUIDs as you have been reporting.

_When you load a new hex file onto a micro:bit there is no secure connect between it and the Raspberry Pi (in our case). The exchanging of security keys is what is referred to as the pairing process and only has to happen once.
The keys will be wiped from the micro:bit if it is flashed (new hex file loaded). Power cycling (switching off and on again) the micro:bit will **not** remove these keys._ 

When you have an image similar to the following (bar chart) on the micro:bit you need to type  `pair F4:CA:40:CE:FC:08`.
[image]
You need to do this pairing process before starting to use the Python programs.

Once you have done the pairing it will appear in device list in bluetoothctl. You need to wait for the image below (Bluezero B) to appear on the micro:bit before you connect. When that appears you can  `connect F4:CA:40:CE:FC:08`
[image]

If you now do
```
gdbus introspect --system --dest org.bluez --object-path / --recurse| grep -i -B17 DA91 | grep node
```
you should see Button B State characteristic available
```
            node /org/bluez/hci0/dev_E3_AC_D2_F8_EB_B9/service002b/char002f {
```

If that worked then you should be OK to start with the Python workshop code.

If it didn't work, can I see the text from the terminal where you have been running bluetoothctl.
Running it with the following command will write it to a file and the screen
`bluetoothctl  | tee my.log`

The pairing process is a bit fiddly, but you only have to do it once. You can then connect and disconnect as many times as you like.

Let me know how it goes.

MrsHesse, 2019-07-31:
We seem to be making some progress today

1. I reloaded the hex file (so I was starting from scratch).

2.  I started bluetoothctl (log provided below).
[btctl.log](https://github.com/ukBaz/python-bluezero/files/3451223/btctl.log)

3.  Pressed A+B+reset on the microbit and confirmed using scan on that the microbit was seen.

4.  When the barchart appeared I typed **pair F4:CA:40:CE:FC:08** in bluetoothctl.  Pairing was successful.

5. A tick appeared on the microbit (never had that before...) and waited for the b image on the LEDs.

6. typed **connect F4:CA:40:CE:FC:08** into bluetoothctl and connection was successful.  A C appeared on the microbit LEDs.

7. Without quitting from bluetoothctl I opened a new terminal and entered
`gdbus introspect --system --dest org.bluez --object-path / --recurse| grep -i -B17 DA91 | grep node`
Nothing happened so I just typed in the gdbus part of the command 
`gdbus introspect --system --dest org.bluez --object-path / --recurse`

and the following output was produced.  There seems to be a lot missing now....
[image]

[gdbus (1).log](https://github.com/ukBaz/python-bluezero/files/3451224/gdbus.1.log)

1 step forward - 2 steps back it seems!

ukBaz, 2019-07-31:
I have to confess that I am slightly flummoxed at this point. 

If I look in your `btctl.log` the entry is there
```
$ grep -B2 -i DA91 btctl.log
[NEW] Characteristic
        /org/bluez/hci0/dev_F4_CA_40_CE_FC_08/service002b/char002f
        e95dda91-251d-470a-a062-fa1922dfa9a8
```
At the end of the `btctl.log there are all the UUIDs I would expect to see:
```
[CHG] Device F4:CA:40:CE:FC:08 UUIDs: 00001800-0000-1000-8000-00805f9b34fb 
[CHG] Device F4:CA:40:CE:FC:08 UUIDs: 00001801-0000-1000-8000-00805f9b34fb 
[CHG] Device F4:CA:40:CE:FC:08 UUIDs: 0000180a-0000-1000-8000-00805f9b34fb 
[CHG] Device F4:CA:40:CE:FC:08 UUIDs: e95d0753-251d-470a-a062-fa1922dfa9a8 
[CHG] Device F4:CA:40:CE:FC:08 UUIDs: e95d6100-251d-470a-a062-fa1922dfa9a8 
[CHG] Device F4:CA:40:CE:FC:08 UUIDs: e95d93af-251d-470a-a062-fa1922dfa9a8 
[CHG] Device F4:CA:40:CE:FC:08 UUIDs: e95d93b0-251d-470a-a062-fa1922dfa9a8 
[CHG] Device F4:CA:40:CE:FC:08 UUIDs: e95d9882-251d-470a-a062-fa1922dfa9a8 
[CHG] Device F4:CA:40:CE:FC:08 UUIDs: e95dd91d-251d-470a-a062-fa1922dfa9a8 
[CHG] Device F4:CA:40:CE:FC:08 UUIDs: e97dd91d-251d-470a-a062-fa1922dfa9a8 
[CHG] Device F4:CA:40:CE:FC:08 ServicesResolved: yes
```
The `ServicesResolved: yes` is an important entry. The information will not be available to gdbus until that entry has appeared.

So I don't understand why gdbus is returning so little.  I can't help thinking I am missing something obvious. Apologies that this is taking so long.

The fact that you are seeing the tick now sounds promising.

We can add some steps to ensure that the Bluetooth on the Raspberry Pi is cleaned up before we start pairing. Can you please do the following and report back:
1. Replace hex file on micro:bit
2. start `bluetoothctl` on RPi
3. In bluetoothctl remove previous pairing
3a. type `devices`
3b. If micro:bit listed, type `remove F4:CA:40:CE:FC:08`
4. In another terminal type `sudo service bluetooth restart`
5. Put micro:bit in pairing mode (A+B+reset)
6. Wait for bar chart to appear
7. bluetoothctl; `scan on`
8. When micro:bit is shown in bluetoothctl
9. bluetoothctl; `scan off`
10. bluetoothctl; `pair F4:CA:40:CE:FC:08`
11. Wait for tick and then "Bluezero B" to appear
12. bluetoothctl: `connect F4:CA:40:CE:FC:08`
13. Wait for it to say `ServicesResolved` in bluetoothctl
14. Try selecting the button B state characteristic anyway:
14a. bluetoothctl: `select-attribute  /org/bluez/hci0/dev_F4_CA_40_CE_FC_08/service002b/char002f`
14b. If 14a doesn't work with copy and paste, then type the command. FYI: Bluetoothctl supports command autocomplete. This means you can start typing the command then press the `<tab>` key. It will try to complete the command. So type:  `sel<tab>-a<tab>/<tab>2b/<tab>f<return>` should get you the command from 14a.

If that doesn't work, send me the bluetoothctl log and the gdbus log.

If there isn't anything obvious in those logs then we will look at hardware and OS versions to see if I can reproduce the issues at this end.

Good luck!

WayneKeenan, 2019-07-31:
I have, on numerous occasions, had to resort to purging all the Bluetooth daemons config:

```
sudo su -
service bluetooth stop && rm -fr /var/lib/bluetooth/* && service bluetooth start
```

ukBaz, 2019-07-31:
Thanks @WayneKeenan, I was not aware of that.

@MrsHesse, I think that would be a stronger version of step 4 in my list. Maybe try my suggestion first, if that doesn't work then try Wayne's suggestion.

MrsHesse, 2019-08-04:
Thank you @ukBaz  for all your help.  

I went through the list step by step and it worked!  I was able to run the first example code that sends messages to display on the microbit.  Yippy

[image]

I saved the bluetoothctl log so I have included that for your info.
[my.log](https://github.com/ukBaz/python-bluezero/files/3464643/my.log)

Just to test I reset everything - and without using bluetoothctl

1. Reset the microbit -  "Bluezero B" appeared 
2. I restarted the python code.
3. Connection occurred and "C" appeared on the microbit
4. The send message code worked beautifully.

I will now start working through the other examples, and let you know how that goes.

Thanks again for all your support.

ukBaz, 2019-08-04:
Excellent!  Thank you for sticking with it and working through the issues.

I'm not sure if there are any edits you would suggest to make the workshop documentation clearer?

I am going to mark this issue as closed. If you have any other issues then please open an new issue.

closed 2019-08-04 by ukBaz