ukBaz/python-bluezero #242: Updating Beacon payload  dynamically

dancane38, 2019-08-18:
First off **thank you** for this. Amazingly helpful to get up the learning curve. 
I'm attempting to use a beacon has a "cheep" peripheral to broadcast sensor data (e.g. cpu temp) without needing to pair. (BTW, just tell me if this is all a bad idea). What I want to do is something like this:

beacon = advertisement.Advertisement(i, 'broadcast')   
... set payload data
beacon.start()
... update the beacon data

Currently when I start, the process becomes the thread itself, so i can't control it (stop it, change the data, restart it). 

What's the best way to accomplish this. I don't mind it being a little wasteful (e.g. new up an entirely different advertisement each second).

The data only needs to be "pulsed" out say 1x-2x per sec. Could be even more lenient.

Thanks in advance -- this is really great.

ukBaz, 2019-08-18:
> I'm attempting to use a beacon has a "cheep" peripheral to broadcast sensor data (e.g. cpu temp) without needing to pair. (BTW, just tell me if this is all a bad idea).

Generally speaking this is not a bad idea. Lots of beacons out there doing this kind of thing.

It is something that I have thought about supporting with Bluezero but the underlying BlueZ API uses a static value from when the advertisement is registered with the advertising manager so this makes it a little bit fiddly. As a result, I've never quite had the time to sort it out.

For the advertisement value to be updated, my understanding is that the following steps need to be done:
1) unregister advertisement from the advertising manager
2) change the value in the advertisement data
3) register advertisement with advertising manager

As you want this to happen on a timed interval then I suspect `GLib.timeout_add` is the function that you would use. timeout_add would use a callback with the above steps in.

I realise this is a very abstract implementation but I don't have access to my development environment currently and there is some detail to think about and test before being able to give a more detailed implementation.
If that abstract implementation is helpful, then great! I look forward to the pull request ;-)
If it is not enough detail, then let me know and I can try and get some time at my development workstation.

dancane38, 2019-08-18:
Thanks a ton. Helpful to know the direction is sound, and it's just a matter of making the impl work. I'm in 'tinker' mode, so there isn't a rush on making this work. I'll play around and keep you posted. Really appreciate the quick reply!

dancane38, 2019-08-19:
The callback method works -- sort of. It doesn't deallocate the thread, so they keep stacking up after each callback.  What might be easier then 'editing' a beacon (which in our case was attempting to kill one and replace it over and over again) would be a simple on/off switch for a beacon. 

Right now, when I .start() the advertiser the thread blocks. I'm no python guru, but I have to imagine there is a way for me to start an async thread that i can .start() and .stop() while the rest of my application continues to run. That's actually the most ideal solution -- the presence of the beacon tells me all I need to know. I just need to toggle it on/off based on what the rest of my application says is going on.

ukBaz, 2019-08-19:
It is a little hard for me to follow along without simple code examples. For example, not sure what you are referring to when you are talking about threads so let me try to show some example of the implementation I was thinking about and you can tell me what doesn't work.
There should be a `beacon.stop()` there already. The `start` and `stop` are referring to the event loop. I am thinking the overhead of starting, stopping and restarting the event loop shouldn't be necessary.

As I have no way of running tests currently I have taken a URL example from the library and modified it to randomly change the URL being advertised every 5 seconds.

```python
import random
import string
from gi.repository import GLib
from bluezero import advertisement
beacon = advertisement.Advertisement(1, 'broadcast')
beacon.service_UUIDs = ['FEAA']
beacon.service_data =  {'FEAA': [0x10, 0x08, 0x03, 0x75, 0x6B,
                                 0x42, 0x61, 0x7A, 0x2e, 0x67,
                                 0x69, 0x74, 0x68, 0x75, 0x62,
                                 0x2E, 0x69, 0x6F]}
ad_manager = advertisement.AdvertisingManager()
ad_manager.register_advertisement(beacon, {})

def update_data():
    ad_manager.unregister_advertisement(beacon)
    # Replace last letter of URL with randomly picked letter
    beacon.service_data['FEAA'][-1] = ord(random.choice(string.ascii_letters))
    ad_manager.register_advertisement(beacon, {})
    return True

GLib.timeout_add_seconds(5, update_data)
beacon.start()
```

Does that do what I think it should?

dancane38, 2019-08-20:
It sure does. The process itself becomes the beacon when you .start() (Ok, I'm sure there is a more technical answer than that, but the rest of the processing stops). The callback works perfectly - so I just re-architected the code so that my "work" was done each time the callback triggers, and viola! A few global variables so that I'm not thrashing around unregistering and registering advertisements, and everything is working well. 

I can't thank you enough for your pointers and help!

ukBaz, 2019-08-20:
Great!  Sounds like you have got it working for yourself so I'll go ahead and close this.

For my understanding, are you saying that you don't need to un-register and re-register the advertisements in the callback? i.e. you just need to change the data value for the beacon to change?

closed 2019-08-20 by ukBaz

dancane38, 2019-08-21:
I did need to un-register and re-register for the changes to take effect. So, only 1 .start() and then using the update to the service data to make changes. thanks for everything!

karlp, 2022-05-05:
This was a super useful example, and would be useful in the examples! I has been fiddling with this for a while, trying to figure out what was wrong.

```python
import bluezero.advertisement
import bluezero.async_tools

def main():
    beacon = bluezero.advertisement.Advertisement(1, 'broadcast')
    ad_manager = bluezero.advertisement.AdvertisingManager()
    ad_manager.register_advertisement(beacon, {})

    beacon.manufacturer_data(0x999, b"initial blah")
    # a literal int won't carry to the closure... :(
    data = [1]

    def update_beacon():
        ad_manager.unregister_advertisement(beacon)
        beacon.manufacturer_data(0xffff, bytearray([1, 2, 3, data[0] % 255, data[0] % 255, data[0] % 255]))
        ad_manager.register_advertisement(beacon, {})
        data[0] = data[0] + 1
        return True

    bluezero.async_tools.add_timer_ms(1000, update_beacon)
    try:
        beacon.start()
    except KeyboardInterrupt:
        beacon.stop()

if __name__ == '__main__':
    main()
```

ukBaz, 2022-05-06:
Thanks @karlp for posting the example.

I'm a little uncertain of the point you are making. Are you saying that your example is working? And you requesting that it is added to the Bluezero examples?

My initial response on adding it to the examples is no. There are a few reasons for this:
1) The more examples that are added increases the support burden of this library. This is mainly because there is not an easy way to automate the testing of examples. This means they have to be manually tested before a release which is time consuming.
2) Your example uses manufacturer data but it is not likely that people using this library have spent the time and money to register for a [company id](https://www.bluetooth.com/specifications/assigned-numbers/company-identifiers/). I would prefer to not encourage bad behaviour.
3) Having dynamically updating beacons is a more sophisticated use case than is the primary target for this library. Great that people are doing this, but it feels like people that want to do this are probably sophisticated enough to find this issue and the example code

ukBaz, 2022-05-06:
I accept there are a number of ways that a Company ID can be legitimally used (especially if it is not in a "product") with some knowledge and care. However, there is a history of code just being copied and ending up in product without understanding and I'm trying to avoid Bluezero being a source if I can.

I also agree that I haven't seen too many other Python Linux libraries do the peripheral and broadcaster roles. Doing the dynamic beacons is interesting but it is a bit of a hack with the stopping and starting. My concern is that I think it is possible for a user to specify updates more frequently than the advertising interval. Haven't got it clear in my head how the library code notice that and what it should do if it did. Have a queue of adverts? Silently drop them?

I think if I was going to include the functionality in the library, then the place to do it would be the `Beacon` class from the `broadcaster` module by adding `update_*` methods.