ukBaz/python-bluezero #249: python3 send_notify_event InvalidArgsException()

freedreamer82, 2019-12-09:
using python 2 no problem at all, with python 3 :

bluezero/peripheral.py", line 669, in send_notify_event
    raise InvalidArgsException()
bluezero.peripheral.InvalidArgsException: org.freedesktop.DBus.Error.InvalidArgs: 

any idea?

Marco

ukBaz, 2019-12-09:
The `send_notify_event` looks like it is expecting the data to either be a string or a list. Do you know what the data format is that you are passing to `send_notify_event`.
I wonder if you data is correct but the tests are too restrictive in the function.

freedreamer82, 2019-12-09:
I checked.in python 3 value is type <class 'bytes'> while in python 2 (same code running is a list i guess)

ukBaz, 2019-12-09:
The code I think we are referring to is:
https://github.com/ukBaz/python-bluezero/blob/master/bluezero/peripheral.py#L644-L668

I only run with Python 3 so I'm a little confused at to why I have not seen this issue. Do you have an example of the value that is being passed to `send_notify_event`?

freedreamer82, 2019-12-10:
add PR https://github.com/ukBaz/python-bluezero/pull/250
the fact is related I've been using bytes type.

closed 2020-01-10 by freedreamer82