ukBaz/python-bluezero #252: disconnect - number of function args tested for 1 is incorrect for class methods.

jmfloyd, 2020-01-01:
Problem in adapter._property_changed()

tools.get_fn_parameters test is incorrect for a class method.
This results in do_connect working correctly with a class method (fn_parameters >0)
but do_disconnect does not get called as fn_parameters > 1 in the elif test.

Solution? Make fn_parameters test only for do_disconnect an if ... else ... instead of a if .. elif ...
However this is really only checking the fn_parameters > 0 (which is done for the do_connect)
Better?    Force named arguments in call? 

Is it possible to change get_fn_parameters to adjust for a class method?

Please treat my suggestions as suggestions only as I am a rather crude python programmer.

Cheers John :-)

freedreamer82, 2020-01-09:
Hi,
I don't think I correctly understood : maybe it's better if put some code example.

thanks
Marco

ukBaz, 2021-01-01:
Closing because of inactivity. Please reopen if this is still an issue

closed 2021-01-01 by ukBaz