ukBaz/python-bluezero #257: TypeError: validate_object_path() argument 1 must be str, not None

ywc-hb, 2020-03-01:


ukBaz, 2020-03-02:
Let us divide this problem up to see if it is an issue with Python or the Raspberry Pi/micro:bit.

My plan is to use `bluetoothctl` to establish that the Bluetooth communication is working OK between the micro:bit and the Raspberry Pi.

In the `bluetoothctl` window after you have done the pairing. Connect to the micro:bit. (I'll use `E3:AC:D2:F8:EB:B9` but you would use whatever is the address of your micro:bit)
```
[bluetooth]# connect E3:AC:D2:F8:EB:B9
```
Once it has successfully connected, open another terminal window on the RPi and type this command  (this is to find out where the information about the micro:bit button B is stored in the Bluetooth system):
```
gdbus introspect --system --dest org.bluez --object-path / --recurse | grep -i -B17 DA91 | grep node
```
You should get an answer that looks something like:
```
            node /org/bluez/hci0/dev_E3_AC_D2_F8_EB_B9/service002b/char002f {
```
We will now take this path information back into `bluetoothctl` so we can read it in there. So back in the first terminal window where `bluetoothctl` is running and connected to the micro:bit. We will select the attribute using the path we just found:
```
[BBC micro:bit [tuzep]]# select-attribute /org/bluez/hci0/dev_E3_AC_D2_F8_EB_B9/service002b/char002f
```
Now reading the attribute it should return zero:
```
[BBC micro:bit [tuzep]:/service002b/char002f]# read
Attempting to read /org/bluez/hci0/dev_E3_AC_D2_F8_EB_B9/service002b/char002f
[CHG] Attribute /org/bluez/hci0/dev_E3_AC_D2_F8_EB_B9/service002b/char002f Value: 0x00
  00
```
If you press and hold the Button B on the micro:bit down and at the same time do a read you will get a non-zero value (1 = short press & 2 = long press)
```                                               .
[BBC micro:bit [tuzep]:/service002b/char002f]# read
Attempting to read /org/bluez/hci0/dev_E3_AC_D2_F8_EB_B9/service002b/char002f
[CHG] Attribute /org/bluez/hci0/dev_E3_AC_D2_F8_EB_B9/service002b/char002f Value: 0x02
  02       
```

If this works then, then things are setup correctly and we can focus on the Python. If there are issues with the above then we need to focus on that.

ln-komandur, 2020-07-17:
Thanks @ukBaz, 
I get the same error as @ywc-hb  when executing the "Sending a message to the micro:bit" python example code in https://ukbaz.github.io/howto/ubit_workshop.html 

> $ python3 microbit-example1.py 
Traceback (most recent call last):
  File "microbit-example1.py", line 11, in <module>
    ubit.connect()
  File "/usr/local/lib/python3.7/dist-packages/bluezero/microbit.py", line 148, in connect
    self.ubit.connect()
  File "/usr/local/lib/python3.7/dist-packages/bluezero/central.py", line 84, in connect
    self.load_gatt()
  File "/usr/local/lib/python3.7/dist-packages/bluezero/central.py", line 60, in load_gatt
    chrc.resolve_gatt()
  File "/usr/local/lib/python3.7/dist-packages/bluezero/GATT.py", line 125, in resolve_gatt
    self.chrc_uuid)
  File "/usr/local/lib/python3.7/dist-packages/bluezero/dbus_tools.py", line 293, in get_methods
    return get_dbus_iface(iface, get_dbus_obj(path_obj))
  File "/usr/local/lib/python3.7/dist-packages/bluezero/dbus_tools.py", line 86, in get_dbus_obj
    return bus.get_object(constants.BLUEZ_SERVICE_NAME, dbus_path)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 241, in get_object
    follow_name_owner_changes=follow_name_owner_changes)
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 244, in __init__
    _dbus_bindings.validate_object_path(object_path)
TypeError: validate_object_path() argument 1 must be str, not None
$

I followed steps similar to the approach you have suggested to separate the problem between `bluetoothctl` and `python3` . I am able to successfully read Button B as below after scanning and connecting microbit from my raspberry pi using `bluetoothctl` . But I am stuck on the python3 part.

> [BBC micro:bit [tuzut]]# menu gatt
[BBC micro:bit [tuzut]]# list-attributes
........
........
Characteristic
	/org/bluez/hci0/dev_CC_4F_26_07_68_A8/service002b/char002f
	e95dda91-251d-470a-a062-fa1922dfa9a8
	MicroBit Button B State
Descriptor
	/org/bluez/hci0/dev_CC_4F_26_07_68_A8/service002b/char002f/desc0031
	00002902-0000-1000-8000-00805f9b34fb
	Client Characteristic Configuration
........
........
[BBC micro:bit [tuzut]]# select-attribute /org/bluez/hci0/dev_CC_4F_26_07_68_A8/service002b/char002f
[BBC micro:bit [tuzut]:/service002b/char002f]#
[BBC micro:bit [tuzut]:/service002b/char002f]# read
Attempting to read /org/bluez/hci0/dev_CC_4F_26_07_68_A8/service002b/char002f
[CHG] Attribute /org/bluez/hci0/dev_CC_4F_26_07_68_A8/service002b/char002f Value:
  00                                               .               
  00 
[BBC micro:bit [tuzut]:/service002b/char002f]# read
Attempting to read /org/bluez/hci0/dev_CC_4F_26_07_68_A8/service002b/char002f
[CHG] Attribute /org/bluez/hci0/dev_CC_4F_26_07_68_A8/service002b/char002f Value:
  01                                               .               
  01                                               .               
[BBC micro:bit [tuzut]:/service002b/char002f]# read
Attempting to read /org/bluez/hci0/dev_CC_4F_26_07_68_A8/service002b/char002f
[CHG] Attribute /org/bluez/hci0/dev_CC_4F_26_07_68_A8/service002b/char002f Value:
  02                                               .               
  02                                               .               
[BBC micro:bit [tuzut]:/service002b/char002f]# 

Here are some details on the installations and versions on my Raspberry Pi 3B

> $ uname -a
Linux raspberrypi 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux
$ python3 --version
Python 3.7.3
$ bluetoothctl --version
bluetoothctl: 5.50

I am new to Python. Would appreciate any directions on troubleshooting the python part. Thanks.

ukBaz, 2020-07-17:
I copied the code from from the workshop link and modified the addresses for the adapter and the device.
I created a Pythyon Virtual Environment and installed bluezero with pip.
I created the micro:bit code using the latest MakeCode editor and loaded it on the micro:bit. I had problems pairing (have others seen this?) so I put the micro:bit 'no pairing required mode'
I've also run it in a separate area on disk to make sure I'm not picking up my development code.
Transcript of run:
```
(venv) pi@raspberrypi:~/stack_overflow $ python bluez_workshop.py 
Enter message: Test
Enter message: Long words
Enter message: 
```
This worked at expected.

I checked the versions and they seemed to match with what you reported.
```
(venv) pi@raspberrypi:~/stack_overflow $ uname -a
Linux raspberrypi 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux
(venv) pi@raspberrypi:~/stack_overflow $ python3 --version
Python 3.7.3
(venv) pi@raspberrypi:~/stack_overflow $ bluetoothctl --version
bluetoothctl: 5.50
```

This makes it difficult for me to debug as it works for me. 

@ln-komandur, I noticed in your bluetoothctl log that you interacted with the button Characteristic, while the code will use the led text characteristic.

My suspicion is that there is either an issue with the code you have entered. Or more likely, that the micro:bit Bluetooth services started and the Bluetooth services in your code don't match. Or possibly some dependency missing on you system.
It is now how we prove or rule those out.

Can you confirm your micro:bit has the same services as the one I have:
```
(venv) pi@raspberrypi:~/stack_overflow $ bluetoothctl info DC:DB:16:6B:8C:5F
Device DC:DB:16:6B:8C:5F (random)
        Name: BBC micro:bit [zegig]
        Alias: BBC micro:bit [zegig]
        Appearance: 0x0200
        Paired: no
        Trusted: no
        Blocked: no
        Connected: no
        LegacyPairing: no
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        UUID: MicroBit Accelerometer... (e95d0753-251d-470a-a062-fa1922dfa9a8)
        UUID: MicroBit Temperature Se.. (e95d6100-251d-470a-a062-fa1922dfa9a8)
        UUID: MicroBit Event Service    (e95d93af-251d-470a-a062-fa1922dfa9a8)
        UUID: MicroBit DFU Control Se.. (e95d93b0-251d-470a-a062-fa1922dfa9a8)
        UUID: MicroBit Button Service   (e95d9882-251d-470a-a062-fa1922dfa9a8)
        UUID: MicroBit LED Service      (e95dd91d-251d-470a-a062-fa1922dfa9a8)
        UUID: Vendor specific           (e97dd91d-251d-470a-a062-fa1922dfa9a8)
        AdvertisingFlags:
  06                                               .               
```
And that the micro:bit instantiation is the same:
```python
from bluezero import microbit
ubit = microbit.Microbit(adapter_addr='B8:27:EB:22:57:E0',
                         device_addr='DC:DB:16:6B:8C:5F',
                         accelerometer_service=True,
                         button_service=True,
                         led_service=True,
                         magnetometer_service=False,
                         pin_service=False,
                         temperature_service=True)
my_text = 'Hello, world'
ubit.connect()

while my_text is not '':
    ubit.text = my_text
    my_text = input('Enter message: ')

ubit.disconnect()
```

ukBaz, 2020-07-17:
As a bonus experiment, I have done that exercise in the workshop without using Bluezero.  You will need to `pip3 install pydbus` for this to run.

This uses just the minimum functionality, so if this runs then that might be a useful datapiont...

> You should just need to change `DEVICE_ADDR` to match your micro:bit

```python
from time import sleep
import pydbus

DEVICE_ADDR = 'DC:DB:16:6B:8C:5F'
LED_TEXT = 'E95D93EE-251D-470A-A062-FA1922DFA9A8'

# DBus object paths
BLUEZ_SERVICE = 'org.bluez'
ADAPTER_PATH = '/org/bluez/hci0'
device_path = f"{ADAPTER_PATH}/dev_{DEVICE_ADDR.replace(':', '_')}"

# setup dbus
bus = pydbus.SystemBus()
mngr = bus.get(BLUEZ_SERVICE, '/')
adapter = bus.get(BLUEZ_SERVICE, ADAPTER_PATH) 
device = bus.get(BLUEZ_SERVICE, device_path)

device.Connect()

while not device.ServicesResolved:
    sleep(0.5)

def get_characteristic_path(dev_path, uuid):
    """Look up DBus path for characteristic UUID"""
    mng_objs = mngr.GetManagedObjects()
    for path in mng_objs:
        chr_uuid = mng_objs[path].get('org.bluez.GattCharacteristic1', {}).get('UUID')
        if path.startswith(dev_path) and chr_uuid == uuid.casefold():
           return path
# Characteristic DBus information
led_path = get_characteristic_path(device._path, LED_TEXT)
led_txt = bus.get(BLUEZ_SERVICE, led_path)

my_text = 'Hello, world'

while my_text is not '':
    # Turn text into bytes
    new_value = [ord(c) for c in my_text]
    # Write bytes (and truncate for Bluetooth if needed)
    led_txt.WriteValue(new_value[:19], {})
    my_text = input('Enter message: ')

device.Disconnect()
```

ln-komandur, 2020-07-17:
Thanks for your response @ukBaz 

> I created the micro:bit code using the latest MakeCode editor and loaded it on the micro:bit. I had problems pairing (have others seen this?) so I put the micro:bit 'no pairing required mode'

Yes, I faced the same problem. And I am using my micro:bit only in the 'no pairing required mode' . I also had problems when I chose any other option from MakeCode editor

> @ln-komandur, I noticed in your bluetoothctl log that you interacted with the button Characteristic, while the code will use the led text characteristic.

You are right. I did this only to isolate the problem to python like you had suggested to @ywc-hb , to make sure that I was able to read Button B from `bluetoothctl` . Yes, what I read for Button B from the command prompt is different from the python code
 
> Can you confirm your micro:bit has the same services as the one I have:

They look the same. Please see below from my micro:bit. I am missing the `AdvertisingFlags: ` in yours, and instead have `RSSI: -70` . I don't know if that matters for services

```
[bluetooth]# info CC:4F:26:07:68:A8
Device CC:4F:26:07:68:A8 (random)
	Name: BBC micro:bit [tuzut]
	Alias: BBC micro:bit [tuzut]
	Appearance: 0x0200
	Paired: no
	Trusted: yes
	Blocked: no
	Connected: no
	LegacyPairing: no
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	UUID: MicroBit Accelerometer... (e95d0753-251d-470a-a062-fa1922dfa9a8)
	UUID: MicroBit Temperature Se.. (e95d6100-251d-470a-a062-fa1922dfa9a8)
	UUID: MicroBit Event Service    (e95d93af-251d-470a-a062-fa1922dfa9a8)
	UUID: MicroBit DFU Control Se.. (e95d93b0-251d-470a-a062-fa1922dfa9a8)
	UUID: MicroBit Button Service   (e95d9882-251d-470a-a062-fa1922dfa9a8)
	UUID: MicroBit LED Service      `(e95dd91d-251d-470a-a062-fa1922dfa9a8)`
	UUID: Vendor specific           (e97dd91d-251d-470a-a062-fa1922dfa9a8)
	RSSI: -70
[bluetooth]# 
```
> And that the micro:bit instantiation is the same:

Yes, this is identical to yours too , except for the device_addr. Please see below

```
from bluezero import microbit
ubit = microbit.Microbit(adapter_addr='B8:27:EB:C8:47:3F',
                         device_addr='CC:4F:26:07:68:A8',
                         accelerometer_service=True,
                         button_service=True,
                         led_service=True,
                         magnetometer_service=False,
                         pin_service=False,
                         temperature_service=True)
my_text = 'Hello, world'
ubit.connect()

while my_text is not '':
    ubit.text = my_text
    my_text = input('Enter message: ')

ubit.disconnect()
```

> As a bonus experiment, I have done that exercise in the workshop without using Bluezero. You will need to pip3 install pydbus for this to run.
> This uses just the minimum functionality, so if this runs then that might be a useful datapiont...

I did `pip3 install pydbus` . I edited the `DEVICE_ADDR` in the bonus experiment to my micro:bit's and ran it after I had already connected the micro:bit to the Raspberry Pi through `bluetoothctl` from another terminal window. I got errors as below

```
$ python3 bonus_mb_test.py 
Traceback (most recent call last):
  File "bonus_mb_test.py", line 40, in <module>
    led_txt.WriteValue(new_value[:19], {})
AttributeError: '<CompositeObject>' object has no attribute 'WriteValue'
$ 

```

And then I disconnected the micro:bit from `bluetoothctl` from the other terminal window and ran the bonus experiment again to let go through the connection steps. I got the below errors

```
$ python3 bonus_mb_test.py 
Traceback (most recent call last):
  File "bonus_mb_test.py", line 18, in <module>
    device.Connect()
  File "/home/pi/.local/lib/python3.7/site-packages/pydbus/proxy_method.py", line 75, in __call__
    0, timeout_to_glib(timeout), None).unpack()
gi.repository.GLib.Error: g-io-error-quark: GDBus.Error:org.bluez.Error.Failed: Software caused connection abort (36)
$ 
```

For the error in the last line, https://bitbucket.org/benallard/galileo/issues/328/bluez-error-authentication-failed mentions trying without pairing. But I am already only on that mode.

ukBaz, 2020-07-18:
Changing the UUID of LED_TEXT is wrong. You have changed it to the LED service which will never work. You can test this in `bluetoothctl`.  The source of these values is https://lancaster-university.github.io/microbit-docs/resources/bluetooth/bluetooth_profile.html

I have tested the pairing issue using my phone and the micro:bit and it worked correctly on there. So not a micro:bit issue.

These issues are looking like they aren't a Bluezero library specific problem.

I started a system update following the instructions at:
https://www.raspberrypi.org/documentation/raspbian/updating.md

I noticed that there is a Bluez firmware update in the list which seems like it might be important. I would suggest you update/upgrade your system if you can and try again.

ukBaz, 2020-07-18:
Update and upgrade complete on the Raspberry Pi.  The pairing issue is now fixed as the "just works" pairing is working again.

I have modified the pydbus script to print out the DBus `led_path` for the LED Text characteristic, other than that, the code is the same as above.

I ran the script 6 times, 2 of them failed, 4 of them worked. Transcript:
```
(venv) pi@raspberrypi:~/stack_overflow $ python pydbus_workshop.py 
path /org/bluez/hci0/dev_DC_DB_16_6B_8C_5F/service002c/char002f
Enter message: 
(venv) pi@raspberrypi:~/stack_overflow $ python pydbus_workshop.py 
Traceback (most recent call last):
  File "pydbus_workshop.py", line 18, in <module>
    device.Connect()
  File "/usr/local/lib/python3.7/dist-packages/pydbus/proxy_method.py", line 75, in __call__
    0, timeout_to_glib(timeout), None).unpack()
gi.repository.GLib.Error: g-io-error-quark: GDBus.Error:org.bluez.Error.Failed: Software caused connection abort (36)
(venv) pi@raspberrypi:~/stack_overflow $ python pydbus_workshop.py 
Traceback (most recent call last):
  File "pydbus_workshop.py", line 18, in <module>
    device.Connect()
  File "/usr/local/lib/python3.7/dist-packages/pydbus/proxy_method.py", line 75, in __call__
    0, timeout_to_glib(timeout), None).unpack()
gi.repository.GLib.Error: g-io-error-quark: GDBus.Error:org.bluez.Error.Failed: Software caused connection abort (36)
(venv) pi@raspberrypi:~/stack_overflow $ python pydbus_workshop.py 
path /org/bluez/hci0/dev_DC_DB_16_6B_8C_5F/service002c/char002f
Enter message: 
(venv) pi@raspberrypi:~/stack_overflow $ python pydbus_workshop.py 
path /org/bluez/hci0/dev_DC_DB_16_6B_8C_5F/service002c/char002f
Enter message: 
(venv) pi@raspberrypi:~/stack_overflow $ python pydbus_workshop.py 
path /org/bluez/hci0/dev_DC_DB_16_6B_8C_5F/service002c/char002f
Enter message: 
(venv) pi@raspberrypi:~/stack_overflow $ 
```

I then ran the original Bluezero workshop code six times. 2 times it failed and 4 times in passed.
```
(venv) pi@raspberrypi:~/stack_overflow $ python bluez_workshop.py 
Enter message: 
(venv) pi@raspberrypi:~/stack_overflow $ python bluez_workshop.py 
Enter message: 
(venv) pi@raspberrypi:~/stack_overflow $ python bluez_workshop.py 
Enter message: 
(venv) pi@raspberrypi:~/stack_overflow $ python bluez_workshop.py 
Traceback (most recent call last):
  File "bluez_workshop.py", line 11, in <module>
    ubit.connect()
  File "/home/pi/stack_overflow/venv/lib/python3.7/site-packages/bluezero/microbit.py", line 148, in connect
    self.ubit.connect()
  File "/home/pi/stack_overflow/venv/lib/python3.7/site-packages/bluezero/central.py", line 79, in connect
    self.rmt_device.connect()
  File "/home/pi/stack_overflow/venv/lib/python3.7/site-packages/bluezero/device.py", line 266, in connect
    self.remote_device_methods.Connect()
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.bluez.Error.Failed: Software caused connection abort
(venv) pi@raspberrypi:~/stack_overflow $ python bluez_workshop.py 
Traceback (most recent call last):
  File "bluez_workshop.py", line 11, in <module>
    ubit.connect()
  File "/home/pi/stack_overflow/venv/lib/python3.7/site-packages/bluezero/microbit.py", line 148, in connect
    self.ubit.connect()
  File "/home/pi/stack_overflow/venv/lib/python3.7/site-packages/bluezero/central.py", line 79, in connect
    self.rmt_device.connect()
  File "/home/pi/stack_overflow/venv/lib/python3.7/site-packages/bluezero/device.py", line 266, in connect
    self.remote_device_methods.Connect()
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.bluez.Error.Failed: Software caused connection abort
(venv) pi@raspberrypi:~/stack_overflow $ python bluez_workshop.py 
Enter message: 
(venv) pi@raspberrypi:~/stack_overflow $ 
```

Not sure what to do next with this. It seems it is a problem at a lower level than Bluezero. I'll carry on investigating but this seems like it might be out of my control....

ln-komandur, 2020-07-18:
Thanks @ukBaz 
I checked my current firmware version before going through upgrades / updates. It was as below

```
pi@raspberrypi:~/Desktop $ /opt/vc/bin/vcgencmd version
Feb 12 2020 12:38:08 
Copyright (c) 2012 Broadcom
version 53a54c770c493957d99bf49762dfabc4eee00e45 (clean) (release) (start)
```

> Changing the UUID of LED_TEXT is wrong. You have changed it to the LED service which will never work. 

I changed this  back to `LED_TEXT = 'E95D93EE-251D-470A-A062-FA1922DFA9A8'` . And also did `sudo apt upgrade`, `sudo apt-get update --fix-missing`, `sudo apt update`, `sudo apt full-upgrade` . I have had some unpleasant experiences in the past doing `sudo rpi-update`  and I therefore didn't do it. 

> I noticed that there is a Bluez firmware update in the list which seems like it might be important. I would suggest you update/upgrade your system if you can and try again.

The kernel and firmware are now  
```
pi@raspberrypi:~/Desktop/microbit $ uname -a
Linux raspberrypi 4.19.118-v7+ #1311 SMP Mon Apr 27 14:21:24 BST 2020 armv7l GNU/Linux
pi@raspberrypi:~/Desktop/microbit $ /opt/vc/bin/vcgencmd version
Apr 15 2020 11:43:08 
Copyright (c) 2012 Broadcom
version 82f9bb929ce2186eb1824178c1ae82902ad6275c (clean) (release) (start)
```

Are you referring this firmware version as having Bluez updates?  

I still get the exact same error messages as before for the bonus experiment 
```
pi@raspberrypi:~/Desktop/microbit $ python3 bonus_mb_test.py 
Traceback (most recent call last):
  File "bonus_mb_test.py", line 40, in <module>
    led_txt.WriteValue(new_value[:19], {})
AttributeError: '<CompositeObject>' object has no attribute 'WriteValue'
pi@raspberrypi:~/Desktop/microbit $
```
and also for the "Sending a message to the micro:bit" example on the [workshop page](https://ukbaz.github.io/howto/ubit_workshop.html). 

```
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 244, in __init__
    _dbus_bindings.validate_object_path(object_path)
TypeError: validate_object_path() argument 1 must be str, not None

```

But both programs are connecting to the micro:bit successfully (I got the tick mark on micro:bit LEDs I was expecting for a successful connection)

ln-komandur, 2020-07-18:
I tried the "Sending a message to the micro:bit" example on the [workshop page](https://ukbaz.github.io/howto/ubit_workshop.html) on my ubuntu 18.04 PC. I did no changes  to the micro:bit whatsoever. I worked like a charm and was able to display the text on the micro:bit

```
userXX@ubuntuPC-XX:~$ uname -a
Linux ubuntuPC-XX 5.3.0-62-generic #56~18.04.1-Ubuntu SMP Wed Jun 24 16:17:03 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
userXX@ubuntuPC-XX:~$ bluetoothctl --version
bluetoothctl: 5.48
userXX@ubuntuPC-XX:~$ python3 --version
Python 3.6.9
userXX@ubuntuPC-XX:~$ python3 ./Desktop/microbit-ble/send-message.py 
Enter message: my
Enter message: message 
Enter message: displays
Enter message: on the
Enter message: micro:bit
Enter message: 
userXX@ubuntuPC-XX:~$ 
```

I also executed the "Reading values from the micro:bit" from the  [workshop page](https://ukbaz.github.io/howto/ubit_workshop.html) without any issues

```
userXX@ubuntuPC-XX:~$ python3 ./Desktop/microbit-ble/read-values-microbit.py 
Connected... Press a button to select mode
Face up
Face up
Face up
Face down
Face down
Face down
Face up
Face up
Face up
Exiting
userXX@ubuntuPC-XX:~$ 
```

ukBaz, 2020-07-18:
Good that you have it working on your ubuntu PC. I think this shows that it is an issue with the RPi.

I was referring to the bluez-firmware package was updated when I did an update/upgrade:
https://packages.debian.org/buster/bluez-firmware

It might not be related, but it seems strange there were recent updates on the following git repo: 
https://github.com/RPi-Distro/bluez-firmware/commits/master

As it is working for me, I am not sure what more I can do to debug this issue as it seems hardware specific. If you have more ideas or suggestions then please let me know.

ln-komandur, 2020-07-18:
> As it is working for me, I am not sure what more I can do to debug this issue as it seems hardware specific. If you have more ideas or suggestions then please let me know.

Yes, @ukBaz I understand the challenge. But I got some good news now. It is all working fine now. But I don't know why its working  now but did not before
Here is how it happened
1. I started all over again, saving the hex file from MakeCode editor and adjusting the python code in the raspberry pi only for the services needed for each program. i.e. to send-messages to micro:bit, I enabled only LED service in MakeCode editor set only the `led_service=True` under
[image]

```
ubit = microbit.Microbit(adapter_addr='64:5A:04:7D:5E:F6',
                         device_addr='CC:4F:26:07:68:A8',
                         accelerometer_service=False,
                         button_service=False,
                         led_service=True,
                         magnetometer_service=False,
                         pin_service=False,
                         temperature_service=False)
```
I then got the program worked and gave output as below

```
pi@raspberrypi:~ $ python3 ./Desktop/microbit/send-message.py 
Enter message: this
Enter message: message
Enter message: is
Enter message: getting
Enter message: across
Enter message: 
```
:-)
And then I enabled just the LED and Button services in MakeCode editor and flashed the hexfile to micro:bit
[image]

I aligned the services in the python code as below
```
ubit = microbit.Microbit(adapter_addr='64:5A:04:7D:5E:F6',
                         device_addr='CC:4F:26:07:68:A8',
                         accelerometer_service=False,
                         button_service=True,
                         led_service=True,
                         magnetometer_service=False,
                         pin_service=False,
                         temperature_service=False)
```

When I executed
```
pi@raspberrypi:~ $ python3 ./Desktop/microbit/read-button-press.py 
```
the micro:bit showed arrow marks to the left and right buttons :-)

And then I enabled only the LED, Button and Accelerometer services in MakeCode editor and flashed the hexfile to micro:bit. I commented out the portions for temperature reading in the 3rd example in [workshop page](https://ukbaz.github.io/howto/ubit_workshop.html)

[image]

I aligned the services in the python code as below
```
ubit = microbit.Microbit(adapter_addr='64:5A:04:7D:5E:F6',
                         device_addr='CC:4F:26:07:68:A8',
                         accelerometer_service=True,
                         button_service=True,
                         led_service=True,
                         magnetometer_service=False,
                         pin_service=False,
                         temperature_service=False)
```

The program execution and output are as below
```
pi@raspberrypi:~ $ python3 ./Desktop/microbit/face-up-or-down.py 
Connected... Press a button to select mode
Temperature is commented out
Temperature is commented out
Temperature is commented out
Face up
Face up
Face up
Face down
Face down
Face down
Face up
Exiting
pi@raspberrypi:~ $ 
```
:-)
And finally I enabled the LED, Button, Accelerometer and temperature services in MakeCode editor and flashed the hexfile to micro:bit. I used the 3rd example in [workshop page](https://ukbaz.github.io/howto/ubit_workshop.html) as is

[image]

I aligned the services in the python code as below
```
ubit = microbit.Microbit(adapter_addr='64:5A:04:7D:5E:F6',
                         device_addr='CC:4F:26:07:68:A8',
                         accelerometer_service=True,
                         button_service=True,
                         led_service=True,
                         magnetometer_service=False,
                         pin_service=False,
                         temperature_service=True)
```

The program execution and output are as below
```
pi@raspberrypi:~ $ python3 ./Desktop/microbit/face-up-or-down-temperature.py 
Connected... Press a button to select mode
Face up
Face up
Face down
Face up
Face down
Face down
Face up
Temperature: 27
Temperature: 27
Temperature: 27
Temperature: 27
Exiting
pi@raspberrypi:~ $ 

```
In any of these 4 trials, I did not connect any other device to the  micro:bit 

I am wondering if these present any clues to the initial behaviour as I have  come back a full circle. But now that its working in my Raspberry Pi, I am good. And thank you very much for your help.

ukBaz, 2020-07-18:
I am going to close this as there hasn't been an update from @ywc-hb and it is working for @ln-komandur 

I've made a note to stay aware of this in the future and to see if there is anything I can do to the library to protect against this.

closed 2020-07-18 by ukBaz