ukBaz/python-bluezero #258: Allowing longer Eddystone-URLs

daghemo, 2020-03-05:
Hi Barry,
line https://github.com/ukBaz/python-bluezero/blob/master/bluezero/eddystone_beacon.py#L40 says  "> 17" but this should be "> 20".

As shown on https://github.com/google/eddystone/tree/master/eddystone-url the URL should be encoded using up to 18 characters: the first representing the schema, then 1 to 17 character for the rest of the URL. 
Frame type and TX power are counted apart. Up to a total of 20 bytes.

ukBaz, 2020-03-05:
Well spotted. Do you want to create a pull request?

daghemo, 2020-03-05:
Feel free to make the change on your own.

ukBaz, 2020-03-08:
#262 fixes this.

Doing some experimentation, my hardware would broadcast the URL up to 23 characters but I've gone with 20 as this is what is on the spec and this [summary](https://ukbaz.github.io/asset/beacon_packet_cheatsheet.jpg) say it should be

closed 2020-03-08 by ukBaz