ukBaz/python-bluezero #264: Example of UART use with micro:bit and RPi

ukBaz, 2020-04-14:
Hi @StephenThompson17,

I've moved your question to here as it seems more appropriate than tagged onto a commit/code-review

> I'm currently experimenting with communications between a microbit and a RPi. I'm curious about what this example does exactly. Can I get a description of its functionality, and more specifically how the reading of the UART data via subscribe_uart works? Any help would be appreciated, thank you!

There is more information about that example at:
https://bluezero.readthedocs.io/en/latest/examples.html#peripheral-nordic-uart-service
For the want of a better expression, the intention with that example was to emulate the micro:bit so you can use your RPi to send and receive short bits of information with the Nordic App on your phone.

If you want to communicate between the RPi and the micro:bit, then that is in client mode and that is built right into the Bluezero micro:bit module with these two commands:
https://bluezero.readthedocs.io/en/latest/level1.html#bluezero.microbit.Microbit.subscribe_uart
https://bluezero.readthedocs.io/en/latest/level1.html#bluezero.microbit.Microbit.uart

There is an example of sending data to the micro:bit in the following issue:
https://github.com/ukBaz/python-bluezero/issues/227#issuecomment-489737238

Reading data is a little bit more involved as it requires the Python code to be run in the eventloop. There is a long discussion about how to create good examples of that at:
https://github.com/ukBaz/python-bluezero/issues/189

I hope that points you in the right direction. If you have more more questions then please post them in this issue.

Cheers,
Barry

ukBaz, 2020-04-14:
OK, I picked up on the wrong example. Was it actually this one that you were asking about @stephenThompson17 ?
https://github.com/ukBaz/python-bluezero/blob/master/examples/microbit_uart.py

If that is the case, I am not sure what your question is. Would you care to elaborate?

Are you asking about what needs to happen on the micro:bit?
These read and write examples should help.
https://makecode.microbit.org/reference/bluetooth/uart-read-until
https://makecode.microbit.org/reference/bluetooth/uart-write-string

StephenThompson17, 2020-04-16:
Thanks for the response.
#189 really helped and I think it was a combination of not fully understanding the eventloop as well as other issues with my program on the microbit which I'm writing in C/C++ using the Lancaster University runtime. Both of which I have now solved.

As an aside, I'm trying to investigate the limiting factor involved with the sending of data over BLE. I've struggled to find any information about the timings involved with carrying out different processes. I have tried by testing the maximum rate for accessing the various microbit services (about 10 Hz (100 ms per read), which I would have thought would be limited by the RPi??), and also now by sending data using the UART service (also about 10 Hz, this could be limited by either end although I thought it would be more likely limited by the sender of data - in this case the microbit??). This common rate suggests it is the same thing limiting it though, but I'm not sure what that is.

Feel free to ignore this aside as it is mostly unrelated, I was just curious whether there was an obvious answer that I'm missing.

Thanks again,
Stephen

ukBaz, 2020-04-16:
Glad that my response helped move you forward on your project. 

The asides are the best bit about maintaining this library :-)

This BLE chip on the micro:bit is a Nordic Semiconductor nRF51, if I remember correctly, so that would be a good place to investigate throughput.
https://devzone.nordicsemi.com/f/nordic-q-a/5423/ble-real-time-data-transmission-nrf51-dk-code
That has a links off to
https://devzone.nordicsemi.com/f/nordic-q-a/1105/how-do-i-calculate-throughput-for-a-ble-link#reply-3441

BLE is focused on saving battery life rather than throughput. For many applications, classic Bluetooth is used as that will generally give higher throughput.

As an aside from me:
I keep looking to doing some micro:bit development in C/C++ but give up at installing the tool chain. I have looked at the online Mbed development environment previously, but there were some issues. I haven't looked recently.
Any tips for a development environment?

Thanks,
Barry

ukBaz, 2020-04-25:
There has been no activity on this for a few days so I am closing. Please reopen if there is an issue with the library.

StephenThompson17, 2020-04-27:
This has all been very helpful to me and my project. I am now looking to be able to unsubscribe from the microbit UART service, I believe this should be possible. I can see it is not included as a microbit function unlike subscribe_uart, however the ability to stop notifying is there in the general GATT classes. Is this possible and is there any help you could offer for this?

I don't know a whole lot about tool chains, but was keen to be able to work on this project offline. I mostly followed the instructions by Lancaster University here - https://lancaster-university.github.io/microbit-docs/offline-toolchains/
I'm using Visual Studio Code as my environment because I had used it before, and then using yotta to build the project from the command line. I have no idea if there are better options than this. It has really been a lot of first attempts and experimenting for me on this project.

Thanks,
Stephen

ukBaz, 2020-04-27:
Glad this is helping.

These is probably a quick and dirty method for testing this, which is delving into the `Microbit()` object you created and using the private `_uart_tx` to call the `stop_notify` method.
Something like: 

```python
<your_ubit_object>._uart_tx.stop_notify()
```

Longer term, the `Microbit` class should be updated with something like:
``` Python
    def unsubscribe_uart(self):
        """
        Stop notifications from the UART service
        """
        self.uart_tx_cb = None
        self._uart_tx.add_characteristic_cb(self._uart_read)
        self._uart_tx.stop_notify()
```

All the above is untested to enable a quick reply. If you do test then let me know either by adding to this ticket or a pull request.

StephenThompson17, 2020-04-27:
I'm finding that even though user callback (uart_tx_cb) is now None, there still exists the characteristic callback, which was added at the point of subscribing, which attempts to run whenever the UART TX characteristic changes but returns an error after unsubscribing because _uart_read is now not achievable with a NoneType user callback. 

``
**ERROR:dbus.connection:Exception in handler for D-Bus signal:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 230, in maybe_handle_message
    self._handler(*args, **kwargs)
``
``
  File "/usr/local/lib/python3.5/dist-packages/bluezero/microbit.py", line 572, in _uart_read
    self.uart_tx_cb(''.join([str(v) for v in changed_props['Value']]))
``
``
TypeError: 'NoneType' object is not callable
**
``

Is there any obvious way to remove the characteristic callback when unsubscribing and then add it again when subscribing?

I appreciate all the help.

ukBaz, 2020-04-27:
Hmm?  This is clearly trickier than I first thought (which is maybe why I didn't implement it previously)

There are some remove methods documented here:
https://dbus.freedesktop.org/doc/dbus-python/dbus.connection.html
`remove_signal_receiver` and `remove` look the most likely candidates, just not sure if we have the right object to call them from.
Currently we are setting the callback with `connect_to_signal` from https://dbus.freedesktop.org/doc/dbus-python/dbus.proxies.html

Will need to dig around a bit more so no quick answer from me.  :-(

ukBaz, 2020-04-27:
Update after some experiments this evening.

Not sure how readable this is but this is my full test script.

```python
from bluezero import microbit
from bluezero import async_tools

def handle_value(value):
    u_key, u_value = value.split(':')
    i_value = int(float(u_value))
    print(f'{u_key} = {i_value}')

def goodbye():
    print('Disconnecting')
    ubit.quit_async()
    ubit.disconnect()
    return False

def no_updates():
    print('stop notify')
    print(dir(ubit.ubit.dongle.bus._connection))
    print(ubit.ubit.dongle.bus._signal_sender_matches)
    ubit.ubit.dongle.bus._connection.remove_signal_receiver(
        ubit.ubit.dongle.bus._signal_sender_matches
    )
    ubit._uart_tx.stop_notify()
    return False

ubit = microbit.Microbit(device_addr='DE:82:35:E7:43:BE',
                         accelerometer_service=False,
                         button_service=False,
                         led_service=False,
                         magnetometer_service=False,
                         pin_service=False,
                         temperature_service=False,
                         uart_service=True)

eloop = async_tools.EventLoop()
ubit.connect()

ubit.subscribe_uart(handle_value)

# Stop notifications after 20 seconds
eloop.add_timer(20000, no_updates)

# Disconnect at 40 seconds
eloop.add_timer(40000, goodbye)

ubit.run_async()
```

These are the key couple of lines.
```python
    ubit.ubit.dongle.bus._connection.remove_signal_receiver(
        ubit.ubit.dongle.bus._signal_sender_matches
    )
    ubit._uart_tx.stop_notify()
```
This isn't what I would want to put in the library, but it is something that is working in my test and can be built on.

ukBaz, 2020-04-28:
Another test, this time with subscribing to two notifications. This was to button A and the uart service. The code on the micro:bit was:
[image]

The code on the Linux box was:
```python
from bluezero import microbit
from bluezero import async_tools

def handle_value(value):
    u_key, u_value = value.split(':')
    i_value = int(float(u_value))
    print(f'{u_key} = {i_value}')

def goodbye():
    print('Disconnecting')
    ubit.quit_async()
    ubit.disconnect()
    return False

def no_updates():
    print('stop notify')
    ubit._uart_tx.stop_notify()
    # ubit._btn_a_state.stop_notify()
    return False

ubit = microbit.Microbit(device_addr='DE:82:35:E7:43:BE',
                         accelerometer_service=False,
                         button_service=True,
                         led_service=False,
                         magnetometer_service=False,
                         pin_service=False,
                         temperature_service=False,
                         uart_service=True)

eloop = async_tools.EventLoop()
ubit.connect()

ubit.subscribe_uart(handle_value)
ubit.subscribe_button_a(print)

# Stop notifications after 20 seconds
eloop.add_timer(20000, no_updates)

# Disconnect at 40 seconds
eloop.add_timer(40000, goodbye)

ubit.run_async()
```

There was no need for the `remove_signal_receiver` functionality. Just the `stop_notify` was enough to stop the notifications.

This needs to be incorporated into the microbit class so that users don't need to use the private variable to stop the notifications.

@StephenThompson17 can you test this to prove it works for someone other than me?

StephenThompson17, 2020-04-28:
What I am finding is that the `stop_notify` functionality is enough to stop the notifications, however if the microbit is asked to subscribe again later it will begin by spurting out all the data that may have been transmitted in the meantime.

This may be fine (and even desired in some scenarios) however it does not constitute unsubscribing from the service in my opinion. It's more a case of just hiding the fact that data is being received.

ukBaz, 2020-04-28:
Thanks for the update. Let me update the testcase to reproduce that scenario. I can then work on resolving it.

ukBaz, 2020-04-28:
The testcase I did was to send a count every second from the micro:bit. On the Python end, I turned notifications off for 15 seconds and then restarted. This should allow me to see if values are being backed up.

## The micro:bit code:

[image]

## The Python

```python
from bluezero import microbit
from bluezero import async_tools

def handle_value(value):
    u_key, u_value = value.split(':')
    i_value = int(float(u_value))
    print(f'{u_key} = {i_value}')

def goodbye():
    print('Disconnecting')
    ubit.quit_async()
    ubit.disconnect()
    return False

def no_updates():
    print('stop notify')
    ubit._uart_tx.stop_notify()
    return False

def new_notification():
    print('starting notify')
    ubit._uart_tx.start_notify()

ubit = microbit.Microbit(device_addr='DE:82:35:E7:43:BE',
                         accelerometer_service=False,
                         button_service=False,
                         led_service=False,
                         magnetometer_service=False,
                         pin_service=False,
                         temperature_service=False,
                         uart_service=True)

eloop = async_tools.EventLoop()
ubit.connect()

ubit.subscribe_uart(handle_value)

# Stop notifications after 10 seconds
eloop.add_timer(10000, no_updates)

# Start notifications again after 15 seconds
eloop.add_timer(25000, new_notification)

# Disconnect at 40 seconds
eloop.add_timer(40000, goodbye)

ubit.run_async()

```

## Output:

```
count = 1
count = 2
count = 3
count = 4
count = 5
count = 6
count = 7
count = 8
count = 9
stop notify
starting notify
count = 25
count = 26
count = 27
count = 28
count = 29
count = 30
count = 31
count = 32
count = 33
count = 34
count = 35
count = 36
count = 37
Disconnecting
```

This looks like it is giving the expected output. 
@StephenThompson17, would you like to comment? 
What am I doing differently?

ukBaz, 2021-01-01:
Closing this because of inactivity. Please reopen if there is still an issue I can help with.

closed 2021-01-01 by ukBaz