ukBaz/python-bluezero #266: ValueError: Cannot find a device when running cpu_temperature.py example

hubert3, 2020-05-25:
Running the cpu_temperature.py example, as soon as a client connects to the peripheral being advertised I get this exception:

```# python3 cpu_temperature.py
CPU temperature is 42.0C
b'h\x10'
Advertisement registered
ERROR:dbus.connection:Exception in handler for D-Bus signal:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 230, in maybe_handle_message
    self._handler(*args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/bluezero-0.3.0-py3.7.egg/bluezero/adapter.py", line 267, in _properties_changed
    device_addr=macaddr)
  File "/usr/local/lib/python3.7/dist-packages/bluezero-0.3.0-py3.7.egg/bluezero/device.py", line 56, in __init__
    " using adapter: " + adapter_addr)
ValueError: Cannot find a device: 5E:FA:19:03:E1:60 using adapter: DC:A6:32:69:5C:FF
```

Platform is Raspbian GNU/Linux 10, kernel Linux raspberrypi 4.19.97-v7l+, bluez package version 5.50-1.2~deb10u1+rpt1

ukBaz, 2020-05-25:
Hmmm?

I don't have a RPi 4 but I have a fairly similar setup to you:
```
(venv) pi@raspberrypi:~ $ python -V
Python 3.7.3
(venv) pi@raspberrypi:~ $ bluetoothctl -v
bluetoothctl: 5.50
(venv) pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux
```

I have just run this on my system and have not been able to reproduce your issues. 

I have used the nRF Connect app on an Android phone. What are you using as a client?

If there is nothing obvious with your client (or I can't reproduce it with a similar client) then we will need to look to see if we can get more debug information form one of these:
https://bluezero.readthedocs.io/en/latest/system_setup.html?highlight=debug#notes-for-getting-debug-information

ukBaz, 2020-05-25:
There is a BLE client built into recent versions of the Chrome brower (chrome://bluetooth-internals) and this has worked for me also:
[image]

ukBaz, 2020-05-30:
As there is not enough information to reproduce this I am close. If there is more information then please let me know and I will reopen.

arslan437, 2020-11-07:
Device advertise correctly but I not able to connect to it I have tested it with uart and temperature example but non of them work. 
bluetoothclt -v = 5.50
raspberry pi 4 

```
setup ad_manager
setup service_manager
Advertising service  6E400001-B5A3-F393-E0A9-E50E24DCCA9E
Register Advert /ukBaz/bluezero/advertisement3038813648
Register Application  /ukBaz/bluezero/application3038813392
GetManagedObjects
GATT application registered
Advertisement registered
ERROR:dbus.connection:Exception in handler for D-Bus signal:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 230, in maybe_h                                                                                                             andle_message
    self._handler(*args, **kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/bluezero/adapter.py", line 2                                                                                                             67, in _properties_changed
    device_addr=macaddr)
  File "/home/pi/.local/lib/python3.7/site-packages/bluezero/device.py", line 56                                                                                                             , in __init__
    " using adapter: " + adapter_addr)
ValueError: Cannot find a device: 45:10:69:42:BA:C0 using adapter: DC:A6:32:4B:9                                                                                                             F:57
^CClosing Mainloop
```

Dbus data

```
signal time=1604738844.264717 sender=:1.2 -> destination=(null destination) serial=813 pat                                                                                                   h=/org/freedesktop/systemd1/unit/sys_2ddevices_2dplatform_2dsoc_2dfe201000_2eserial_2dtty_                                                                                                   2dttyAMA0_2dhci0_2dhci0_3a64_2edevice; interface=org.freedesktop.DBus.Properties; member=P                                                                                                   ropertiesChanged
   string "org.freedesktop.systemd1.Unit"
   array [
      dict entry(
         string "ActiveState"
         variant             string "inactive"
      )
      dict entry(
         string "SubState"
         variant             string "dead"
      )
      dict entry(
         string "StateChangeTimestamp"
         variant             uint64 1604738844262414
      )
      dict entry(
         string "StateChangeTimestampMonotonic"
         variant             uint64 2436603947
      )
      dict entry(
         string "InactiveExitTimestamp"
         variant             uint64 1604738838553032
      )
      dict entry(
         string "InactiveExitTimestampMonotonic"
         variant             uint64 2430894564
      )
      dict entry(
         string "ActiveEnterTimestamp"
         variant             uint64 1604738838553032
      )
      dict entry(
         string "ActiveEnterTimestampMonotonic"
         variant             uint64 2430894564
      )
      dict entry(
         string "ActiveExitTimestamp"
         variant             uint64 1604738844262414
      )
      dict entry(
         string "ActiveExitTimestampMonotonic"
         variant             uint64 2436603947
[... 46 lines cut ...]
   ]
   array [
      string "Conditions"
      string "Asserts"
   ]
signal time=1604738844.265571 sender=:1.2 -> destination=(null destination) serial=814 pat                                                                                                   h=/org/freedesktop/systemd1; interface=org.freedesktop.systemd1.Manager; member=UnitRemove                                                                                                   d
   string "sys-devices-platform-soc-fe201000.serial-tty-ttyAMA0-hci0-hci0:64.device"
   object path "/org/freedesktop/systemd1/unit/sys_2ddevices_2dplatform_2dsoc_2dfe201000_2                                                                                                   eserial_2dtty_2dttyAMA0_2dhci0_2dhci0_3a64_2edevice"
```

Bluetooth -nEd results 
```
bluetoothd[1455]: Bluetooth daemon 5.50
bluetoothd[1455]: src/main.c:parse_config() parsing /etc/bluetooth/main.conf
bluetoothd[1455]: src/main.c:parse_config() Key file does not have key “DiscoverableTimeou                                                                                                   t” in group “General”
bluetoothd[1455]: src/main.c:parse_config() Key file does not have key “PairableTimeout” i                                                                                                   n group “General”
bluetoothd[1455]: src/main.c:parse_config() Key file does not have key “Privacy” in group                                                                                                    “General”
bluetoothd[1455]: src/main.c:parse_config() Key file does not have key “Name” in group “Ge                                                                                                   neral”
bluetoothd[1455]: src/main.c:parse_config() Key file does not have key “Class” in group “G                                                                                                   eneral”
bluetoothd[1455]: src/main.c:parse_config() Key file does not have key “DeviceID” in group                                                                                                    “General”
bluetoothd[1455]: src/main.c:parse_config() Key file does not have key “ReverseServiceDisc                                                                                                   overy” in group “General”
bluetoothd[1455]: src/main.c:parse_config() Key file does not have key “MinEncKeySize” in                                                                                                    group “GATT”
D-Bus setup failed: Name already in use
bluetoothd[1455]: Unable to get on D-Bus
```

ukBaz, 2020-11-07:
Thank you for posting more debug information. The only piece of information that looks interesting is:

> D-Bus setup failed: Name already in use
> bluetoothd[1455]: Unable to get on D-Bus

Did you do `sudo service bluetooth stop` before you did `bluetoothd -nEd`?

I have just re-ran the CPU example and have been able to connect and read data.

There have been some issues with versions of the bluez-firmware on Raspberry Pi recently. Can you check you are up-to-date with the various packages. This is what I currently have on my system:

```none
pi@raspberrypi:~ $ dpkg -l | grep bluez
ii  bluez                                5.50-1.2~deb10u1+rpt2               armhf        Bluetooth tools and daemons
ii  bluez-firmware                       1.2-4+rpt6                          all          Firmware for Bluetooth devices
```

arslan437, 2020-11-07:
Yes I did stop the bluetooth basically I followed this link 
https://bluezero.readthedocs.io/en/stable/system_setup.html#notes-for-getting-debug-information

```
pi@raspberrypi:~ $ sudo service bluetooth start
pi@raspberrypi:~ $ bluetoothd -nEd
bluetoothd[2181]: Bluetooth daemon 5.50
bluetoothd[2181]: src/main.c:parse_config() parsing /etc/bluetooth/main.conf
bluetoothd[2181]: src/main.c:parse_config() Key file does not have key “DiscoverableTimeout” in group “General”
bluetoothd[2181]: src/main.c:parse_config() Key file does not have key “PairableTimeout” in group “General”
bluetoothd[2181]: src/main.c:parse_config() Key file does not have key “Privacy” in group “General”
bluetoothd[2181]: src/main.c:parse_config() Key file does not have key “Name” in group “General”
bluetoothd[2181]: src/main.c:parse_config() Key file does not have key “Class” in group “General”
bluetoothd[2181]: src/main.c:parse_config() Key file does not have key “DeviceID” in group “General”
bluetoothd[2181]: src/main.c:parse_config() Key file does not have key “ReverseServiceDiscovery” in group “General”
bluetoothd[2181]: src/main.c:parse_config() Key file does not have key “MinEncKeySize” in group “GATT”
D-Bus setup failed: Connection ":1.174" is not allowed to own the service "org.bluez" due to security policies in the configuration file
bluetoothd[2181]: Unable to get on D-Bus

```
Packages that I have on my pi 

```
pi@raspberrypi:~ $ dpkg -l | grep bluez
ii  bluez                                 5.50-1.2~deb10u1+rpt2                  armhf        Bluetooth tools and daemons
ii  bluez-firmware                        1.2-4+rpt6                             all          Firmware for Bluetooth devices
```

arslan437, 2020-11-07:
here is the some of the data from the bluetoothclt with uart example running 

```

[CHG] Controller DC:A6:32:4B:9F:57 SupportedInstances: 0x03
[CHG] Controller DC:A6:32:4B:9F:57 ActiveInstances: 0x02
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 0000ffff-0000-1000-8000-00805f9b34fb
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 0000112d-0000-1000-8000-00805f9b34fb
[CHG] Controller DC:A6:32:4B:9F:57 SupportedInstances: 0x04
[CHG] Controller DC:A6:32:4B:9F:57 ActiveInstances: 0x01
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 0000ffff-0000-1000-8000-00805f9b34fb
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 0000112d-0000-1000-8000-00805f9b34fb
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 0000112d-0000-1000-8000-00805f9b34fb
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 6e400001-b5a3-f393-e0a9-e50e24dcca9e
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 0000ffff-0000-1000-8000-00805f9b34fb
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 0000112d-0000-1000-8000-00805f9b34fb
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 6e400001-b5a3-f393-e0a9-e50e24dcca9e
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 0000ffff-0000-1000-8000-00805f9b34fb
[CHG] Controller DC:A6:32:4B:9F:57 SupportedInstances: 0x03
[CHG] Controller DC:A6:32:4B:9F:57 ActiveInstances: 0x02
[NEW] Device 48:93:56:CE:08:F2 48-93-56-CE-08-F2
[CHG] Device 48:93:56:CE:08:F2 Connected: no
[DEL] Device 48:93:56:CE:08:F2 48-93-56-CE-08-F2
```

ukBaz, 2020-11-07:
The transcript above shows you using *start* not *stop*. There cannot be two `bluetoothd` daemons running so that is why it gives the error with that command which is probably unrelated to your connection issue.

Your packages look up to date (or at least the same as what I have). We need to find out what is different between our two environments. 

I am confused by your `bluetoothctl` transcript. Where are those UUID's coming from? For example: `0000ffff-0000-1000-8000-00805f9b34fb`. Maybe we need to step back a bit. My assumption was that you were trying to create a BLE peripheral on the RPi and connect to it from another device. e.g. your phone. Is that correct?

I have just ran both the [ble_uart.py ](https://github.com/ukBaz/python-bluezero/blob/master/examples/ble_uart.py) and [cpu_temperature.py](https://github.com/ukBaz/python-bluezero/blob/master/examples/cpu_temperature.py) example successfull.  Are you running these examples unmodified?

Have you changed the [D-Bus permissions](https://bluezero.readthedocs.io/en/stable/system_setup.html#change-dbus-permissions-for-bluezero)?

We need is to identify what is different on your system than mine so that I can reproduce the error or correct your system.  Can you do a `bluetoothctl show` and share.

Also, try having `sudo btmon` running in a different terminal to see if anything useful is reported in there when the failure happens.

arslan437, 2020-11-07:
`0000ffff-0000-1000-8000-00805f9b34fb` ignore this I added this uuid to adervertisement using bluetoothclt.  

yes, I am running these examples unmodified.

yes, I have copied the file to system.d as documentation (I have waste couple of hour on that already).

Now I got the uart example working perfectly but here what I did so it started working 
`bluetoothctl -> menu advertise -> uuid 6E400001-B5A3-F393-E0A9-E50E24DCCA9E 0x45 0x45 `

```
pi@raspberrypi:~/tcodes $ bluetoothctl
Agent registered
[bluetooth]# menu advertise
Menu advertise:
Available commands:
-------------------
uuids [uuid1 uuid2 ...]                           Set/Get advertise uuids
service [uuid] [data=xx xx ...]                   Set/Get advertise service data
manufacturer [id] [data=xx xx ...]                Set/Get advertise manufacturer data
data [type] [data=xx xx ...]                      Set/Get advertise data
discoverable [on/off]                             Set/Get advertise discoverable
discoverable-timeout [seconds]                    Set/Get advertise discoverable timeout
tx-power [on/off]                                 Show/Enable/Disable TX power to be advertised
name [on/off/name]                                Configure local name to be advertised
appearance [on/off/value]                         Configure custom appearance to be advertised
duration [seconds]                                Set/Get advertise duration
timeout [seconds]                                 Set/Get advertise timeout
clear [uuids/service/manufacturer/config-name...] Clear advertise config
back                                              Return to main menu
version                                           Display version
quit                                              Quit program
exit                                              Quit program
help                                              Display help about this program
export                                            Print evironment variables
[bluetooth]# service 6E400001-B5A3-F393-E0A9-E50E24DCCA9E 0x45 0x45
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 6e400001-b5a3-f393-e0a9-e50e24dcca9e
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 0000112d-0000-1000-8000-00805f9b34fb
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 6e400001-b5a3-f393-e0a9-e50e24dcca9e
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Controller DC:A6:32:4B:9F:57 UUIDs: 0000112d-0000-1000-8000-00805f9b34fb
[CHG] Controller DC:A6:32:4B:9F:57 SupportedInstances: 0x04
[CHG] Controller DC:A6:32:4B:9F:57 ActiveInstances: 0x01
[NEW] Device 6C:AF:4B:99:95:EB 6C-AF-4B-99-95-EB
[NEW] Primary Service
        /org/bluez/hci0/dev_6C_AF_4B_99_95_EB/service0001
        00001801-0000-1000-8000-00805f9b34fb
        Generic Attribute Profile
[NEW] Characteristic
        /org/bluez/hci0/dev_6C_AF_4B_99_95_EB/service0001/char0002
        00002a05-0000-1000-8000-00805f9b34fb
        Service Changed
[NEW] Primary Service
        /org/bluez/hci0/dev_6C_AF_4B_99_95_EB/service0028
        0000fe35-0000-1000-8000-00805f9b34fb
        HUAWEI Technologies Co., Ltd
[NEW] Characteristic
        /org/bluez/hci0/dev_6C_AF_4B_99_95_EB/service0028/char0029
        00002a00-0000-1000-8000-00805f9b34fb
        Device Name
[NEW] Characteristic
        /org/bluez/hci0/dev_6C_AF_4B_99_95_EB/service0028/char002b
[... 263 lines cut ...]
        Filter policy: Allow Scan Request from Any, Allow Connect Request from Any (0x00)
> HCI Event: Command Complete (0x0e) plen 4                             #22 [hci0] 32.528877
      LE Set Advertising Parameters (0x08|0x0006) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Advertise Enable (0x08|0x000a) plen 1             #23 [hci0] 32.528910
        Advertising: Enabled (0x01)
> HCI Event: Command Complete (0x0e) plen 4                             #24 [hci0] 32.529300
      LE Set Advertise Enable (0x08|0x000a) ncmd 1
        Status: Success (0x00)
> HCI Event: Number of Completed Packets (0x13) plen 5                  #25 [hci0] 36.359869
        Num handles: 1
        Handle: 64
        Count: 1
> HCI Event: Disconnect Complete (0x05) plen 4                          #26 [hci0] 36.359962
        Status: Success (0x00)
        Handle: 64
        Reason: Connection Timeout (0x08)
@ MGMT Event: Device Disconnected (0x000c) plen 8                  {0x0002} [hci0] 36.359978
        LE Address: 4E:E4:C0:F5:D2:DB (Resolvable)
        Reason: Connection timeout (0x01)
@ MGMT Event: Device Disconnected (0x000c) plen 8                  {0x0001} [hci0] 36.359978
        LE Address: 4E:E4:C0:F5:D2:DB (Resolvable)
        Reason: Connection timeout (0x01)

```

arslan437, 2020-11-07:
yes i am trying to make pi as peripheral and trying to connect the pi with phone.

my final goal is to make ble based midi peripheral. when button is pressed(which are connect to pi) then send the midi data using ble.

here is the working code using esp32 (arduino)

```
#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>

uint8_t note = 38;
int SNARE[6] = {150, 4000, 38, 3, 0, 0}; //{threshold, sensitivity, note(no use), flag, velocity, last peakValue}
boolean snareFlag = false;

BLECharacteristic *pCharacteristic;
bool deviceConnected = false;

#define MIDI_SERVICE_UUID        "03b80e5a-ede8-4b33-a751-6ce34ec4c700"
#define MIDI_CHARACTERISTIC_UUID "7772e5db-3868-4112-a1a9-f2669d106bf3"

uint8_t midiPacket[] = {
  0x80,  // header
  0x80,  // timestamp, not implemented
  0x00,  // status
  0x3c,  // 0x3c == 60 == middle c
  0x00   // velocity
};

class MyServerCallbacks: public BLEServerCallbacks {
    void onConnect(BLEServer* pServer) {
      deviceConnected = true;
    };

    void onDisconnect(BLEServer* pServer) {
      deviceConnected = false;
    }
};

void setup() {
  Serial.begin(115200);

  BLEDevice::init("MIDI"); //Device Name

  // Create the BLE Server
  BLEServer *pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());
  BLEDevice::setEncryptionLevel((esp_ble_sec_act_t)ESP_LE_AUTH_REQ_SC_BOND);

  // Create the BLE Service
  BLEService *pService = pServer->createService(BLEUUID(MIDI_SERVICE_UUID));

  // Create a BLE Characteristic
  pCharacteristic = pService->createCharacteristic(
                      BLEUUID(MIDI_CHARACTERISTIC_UUID),
                      BLECharacteristic::PROPERTY_READ   |
                      BLECharacteristic::PROPERTY_NOTIFY |
                      BLECharacteristic::PROPERTY_WRITE_NR
                    );
  pCharacteristic->setAccessPermissions(ESP_GATT_PERM_READ_ENCRYPTED | ESP_GATT_PERM_WRITE_ENCRYPTED);

  // https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.descriptor.gatt.client_characteristic_configuration.xml
  // Create a BLE Descriptor
  pCharacteristic->addDescriptor(new BLE2902());

  // Start the service
  pService->start();

  // Start advertising

  BLESecurity *pSecurity = new BLESecurity();
  pSecurity->setAuthenticationMode(ESP_LE_AUTH_REQ_SC_BOND);
  pSecurity->setCapability(ESP_IO_CAP_NONE);
  pSecurity->setInitEncryptionKey(ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK);

  pServer->getAdvertising()->addServiceUUID(MIDI_SERVICE_UUID);
  pServer->getAdvertising()->start();

}

void loop() {
  if (deviceConnected) {
     
      // noteOn
      midiPacket[2] = 0x90; // note down, channel 0
      midiPacket[3] = 51; //snare note is 38
      midiPacket[4] = 100;  // velocity
      pCharacteristic->setValue(midiPacket, 5); // packet, length in bytes
      pCharacteristic->notify();
      // noteOff
      midiPacket[2] = 0x80; // note up, channel 0
      midiPacket[3] = 51; //snare note is 38
      midiPacket[4] = 0;    // velocity
      pCharacteristic->setValue(midiPacket, 5); // packet, length in bytes)
      pCharacteristic->notify();

      snareFlag = true;
  }
  delay(100);
}
```

I am thinking to modify the uart example for this. any comments on this

ukBaz, 2020-11-07:
What you have done with `bluetoothctl` is the same as what is done with the [Bluezero code](https://github.com/ukBaz/python-bluezero/blob/0df375b14f766ab4c2c7b0247d1297aed3887edf/bluezero/peripheral.py#L241-L265) is doing. 

I am still unable to reproduce the fault here. A thought is that you have the system in some unexpected state. If you have had failures earlier in the adding the advertisements, I sometimes find it necessary to restart the Bluetooth service `sudo service bluetooth restart` to get it back into a known condition. 

Your MIDI question is a separate issue so I've started a new ticket for that.

arslan437, 2020-11-08:
I did restart the service and rebooted the pi multiple time but the situation is still the same.

so I did upgraded the bluez to 5.54 but no luck.
here are the steps 
```
wget http://www.kernel.org/pub/linux/bluetooth/bluez-5.54.tar.xz
sudo tar xf bluez-5.54.tar.xz
cd bluez-5.54
./configure --prefix=/usr \
            --mandir=/usr/share/man \
            --sysconfdir=/etc \
            --localstatedir=/var \
            --enable-experimental \
            --enable-maintainer-mode\
            --enable-midi
make -j 4 && sudo make install
sudo sed -i '/^ExecStart.*bluetoothd\s*$/ s/$/ --experimental/' /lib/systemd/system/bluetooth.service
sudo systemctl daemon-reload
sudo service bluetooth restart

sudo cp /usr/lib/bluetooth/bluetoothd /usr/lib/bluetooth/bluetoothd-550.orig
sudo ln -sf /usr/libexec/bluetooth/bluetoothd /usr/lib/bluetooth/bluetoothd
sudo systemctl daemon-reload
```

arslan437, 2020-11-08:
here is the error from the nrf connect app. By the way which app are you using for testing?

[image]

ukBaz, 2020-11-08:
I have been using a combination of chrome://bluetooth-internals/#device, nRF Connect app, and nRF UART 2.0 app and it has generally been without issue.

Then finally today I had one connection that gave me the error:
```none
(venv) pi@raspberrypi:~/development/python-bluezero $ python examples/ble_uart.py 

setup ad_manager
setup service_manager
Advertising service  6E400001-B5A3-F393-E0A9-E50E24DCCA9E
Register Advert /ukBaz/bluezero/advertisement1970724976
Register Application  /ukBaz/bluezero/application1974338128
GetManagedObjects
GATT application registered
Advertisement registered
ERROR:dbus.connection:Exception in handler for D-Bus signal:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 230, in maybe_handle_message
    self._handler(*args, **kwargs)
  File "/home/pi/development/python-bluezero/bluezero/adapter.py", line 267, in _properties_changed
    device_addr=macaddr)
  File "/home/pi/development/python-bluezero/bluezero/device.py", line 56, in __init__
    " using adapter: " + adapter_addr)
ValueError: Cannot find a device: 75:8F:AE:E6:48:4E using adapter: 00:02:5B:03:44:07
my test data
^C^CClosing Mainloop
(venv) pi@raspberrypi:~/development/python-bluezero $ python examples/ble_uart.py 

setup ad_manager
setup service_manager
Advertising service  6E400001-B5A3-F393-E0A9-E50E24DCCA9E
Register Advert /ukBaz/bluezero/advertisement1971355824
Register Application  /ukBaz/bluezero/application1979622352
GetManagedObjects
GATT application registered
Advertisement registered
my test data
```
However, when I ran it again it worked without error. In the session that got the ERROR, it still allowed me to send `my test data`.

I need to test this a little more to find out why it only is an issue intermittently. I need to do some logging of what the contents are that cause the good and bad triggering of the `_properties_changed` call back in the adapter:
https://github.com/ukBaz/python-bluezero/blob/master/bluezero/adapter.py#L258

arslan437, 2020-11-08:
Its totally the opposite case, just worked few time I don't how.  But will be waiting for your response.

ukBaz, 2020-11-08:
Can you take a look at the `ble_peripheral` branch I have just created.
https://github.com/ukBaz/python-bluezero/tree/ble_peripheral

I've done a new [commit](https://github.com/ukBaz/python-bluezero/commit/ae3be0972be9778bf0fbcfa63aed2d11c7f86848)which I hope solves this problem.
It seems the issue was when there was a properties changed event that was for a disconnect. The adapter was still trying to create a new device.
I haven't been able to create a failure with this new commit. Would be good if someone else could try it also.

arslan437, 2020-11-09:
I have tested it and it worked perfectly. It solved the issue.  
I have quick question can we use this module as asynchronously (non blocking code). 
Currently after uart.start() no code is executed after this line. 
have good experience c and c++.
Is it possible something like this? 
```
def read_callback(value):
     #do some  stuff like print value 
     print(val)

#register above callback in service or when some thing is received then the above function is called 

while True:
     if condition == True:
          uart.notify(value)

```

some thing like this.

Thanks.

ukBaz, 2020-11-09:
Thanks for testing. I'll do some more testing using the adapter code when it is used as a central device to check it hasn't broken that and then I'll merge.

ukBaz, 2022-06-17:
This fix has been merged so I'm going to close this issue

closed 2022-06-17 by ukBaz