ukBaz/python-bluezero #270: GATT Notifying not working

lekoOwO, 2020-10-25:
I'm working with the cpu temp example as well, but I found that the notification is not working.
My bluez version is 5.50, is that the main issue? I failed to downgrade it :(

lekoOwO, 2020-10-25:
Yes it's the bluez version.
I have successfully downgraded to 5.43 and it works fine.

closed 2020-10-25 by lekoOwO

ukBaz, 2020-10-25:
The intention of Bluezero is that it will work with the latest BlueZ being distributed with Debian/Raspberry Pi OS. 

This is currently 5.50 so let me take a look and see if I can reproduce your error. If I can then I'll either reopen this issue or link it to a new one.

It might be difficult if you have already downgraded, but do you have the details of the packages you have installed? e.g.
```
(venv) pi@raspberrypi:~ $ dpkg -l | grep bluez
```

lekoOwO, 2020-10-25:
> The intention of Bluezero is that it will work with the latest BlueZ being distributed with Debian/Raspberry Pi OS.
>
> This is currently 5.50 so let me take a look and see if I can reproduce your [...]

Well I believe that running `sudo apt install pi-bluetooth` matches with my current software version except that I have covered my bluez w/ 5.43 version.

lekoOwO, 2020-10-25:
Well I've found something strange is that after I updated my bluez to 5.50, the notify still works too.
I'm using RPi3, I'm wondering that is doing
```
sudo apt-get purge bluez -y
sudo apt install pi-bluetooth
```
works

lekoOwO, 2020-10-25:
BTW, w/ this working bluez 5.50, I have:
```
ii  bluez                                5.50-1.2~deb10u1+rpt2               armhf        Bluetooth tools and daemons
ii  bluez-firmware                       1.2-4+rpt6                          all          Firmware for Bluetooth devices
ii  bluez-obexd                          5.50-1.2~deb10u1+rpt2               armhf        bluez obex daemon
ii  bluez-tools                          2.0~20170911.0.7cb788c-2            armhf        Set of tools to manage Bluetooth devices for linux
```

ukBaz, 2020-10-25:
I have tested with:
```
pi@raspberrypi:~ $ dpkg -l | grep bluez
ii  bluez                                5.50-1.2~deb10u1+rpt2               armhf        Bluetooth tools and daemons
ii  bluez-firmware                       1.2-4+rpt6                          all          Firmware for Bluetooth devices
```
And it is working for me:
```
(venv) pi@raspberrypi:~/development/python-bluezero $ python examples/cpu_temperature.py 
CPU temperature is 56.4C
b'\xcc\x15'
Advertisement registered
Notifying on
Starting timer event
examples/cpu_temperature.py:69: PyGIDeprecationWarning: GObject.timeout_add is deprecated; use GLib.timeout_add instead
  GObject.timeout_add(500, self.temperature_cb)
Getting new temperature [53.7] True
Array value:  [dbus.Byte(250), dbus.Byte(20)]
Notifying off
Getting new temperature [53.7] False
Array value:  [dbus.Byte(250), dbus.Byte(20)]
```

Raspberry Pi did release a version of the the firmware that was a problem. Maybe you had that version installed? More details at: https://github.com/RPi-Distro/firmware-nonfree/ issue number 8

lekoOwO, 2020-10-25:
I believe that i was on `5.50-1.2~deb10u1+rpt1`, hmm