ukBaz/python-bluezero #271: Failing to register advertisement when broadcasting

martinfalang, 2020-11-03:
Hi

I'm currently working on an application on a Raspberry Pi that will broadcast some basic information about itself to nearby devices (something as simple as just the device name for example), without allowing anyone to connect to the Raspberry Pi.

I'm running into problems registering the advertisements when advertising as a broadcaster instead of a peripheral. When running as a peripheral all is fine and I get "Registered advertisement" from the callback function. When calling as a broadcaster it fails to register the advertisement and prints "Failed to register advertisement: org.bluez.Error.Failed: Failed to register advertisement". 

This happens if I run any bluezero code (e.g. eddystone-url-beacon.py), but also with the example-advertisement file in the BlueZ repo, so I suspect it's not a bug with bluezero, but I can't seem to find anyone else who has had the same problem so I'm posting it here.

I'm running BlueZ version 5.53 in experimental mode, and have run all the commands specified in the [setup](https://bluezero.readthedocs.io/en/stable/system_setup.html) page of the bluezero documentation.

Any insight would be greatly appreciated!

Best regards

ukBaz, 2020-11-03:
I have run a quick test with `python examples/eddystone-url-beacon.py` and I can reproduce your error with the following output with `btmon`:

``` None
pi@raspberrypi:~ $ sudo btmon 
Bluetooth monitor ver 5.50
= Note: Linux version 5.4.72-v7+ (armv7l)                                                               0.424398
= Note: Bluetooth subsystem version 2.22                                                                0.424409
= New Index: 00:02:5B:03:44:07 (Primary,USB,hci0)                                                [hci0] 0.424412
= Open Index: 00:02:5B:03:44:07                                                                  [hci0] 0.424414
= Index Info: 00:02:5B:03:44:07 (Cambridge Silicon Radio)                                        [hci0] 0.424418
@ MGMT Open: bluetoothd (privileged) version 1.14                                              {0x0001} 0.424421
@ MGMT Open: btmon (privileged) version 1.14                                                   {0x0002} 0.424483
@ MGMT Command: Add Advertising (0x003e) plen 34                                        {0x0001} [hci0] 5.727211
        Instance: 1
        Flags: 0x00000000
        Duration: 0
        Timeout: 0
        Advertising data length: 23
        16-bit Service UUIDs (complete): 1 entry
          Google (0xfeaa)
        Service Data (UUID 0xfeaa): 10080367697468756200756b42617a
        Scan response length: 0
@ MGMT Event: Advertising Added (0x0023) plen 1                                         {0x0002} [hci0] 5.727237
        Instance: 1
< HCI Command: LE Set Random Address (0x08|0x0005) plen 6                                     #1 [hci0] 5.727330
        Address: 1C:1C:A2:69:2E:2E (Non-Resolvable)
> HCI Event: Command Complete (0x0e) plen 4                                                   #2 [hci0] 5.728319
      LE Set Random Address (0x08|0x0005) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Advertising Parameters (0x08|0x0006) plen 15                            #3 [hci0] 5.728379
        Min advertising interval: 62.500 msec (0x0064)
        Max advertising interval: 93.750 msec (0x0096)
        Type: Non connectable undirected - ADV_NONCONN_IND (0x03)
        Own address type: Random (0x01)
        Direct address type: Public (0x00)
        Direct address: 00:00:00:00:00:00 (OUI 00-00-00)
        Channel map: 37, 38, 39 (0x07)
        Filter policy: Allow Scan Request from Any, Allow Connect Request from Any (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                                   #4 [hci0] 5.730319
      LE Set Advertising Parameters (0x08|0x0006) ncmd 1
        Status: Invalid HCI Command Parameters (0x12)
@ MGMT Event: Advertising Removed (0x0024) plen 1                                       {0x0002} [hci0] 5.730395
        Instance: 1
@ MGMT Event: Command Status (0x0002) plen 3                                            {0x0001} [hci0] 5.730407
      Add Advertising (0x003e)
        Status: Invalid Parameters (0x0d)
= bluetoothd: Failed to add advertisement: Invalid Parameters (0x0d)                                    5.730716
```
I could also reproduce this with the `btmgmt` tool on the command line:
```None
pi@raspberrypi:~ $ sudo btmgmt add-adv -u feaa -d 1016aafe100001626c7565746f6f746800 -g 1
Add Advertising failed with status 0x0d (Invalid Parameters)
```

This seemed suspicious as this was working a few days ago. So after a bit of tinkering I identified that this was when the adapter wasn't discoverable. If I did the following alteration to `examples/eddystone-url-beacon.py` it worked for me.
```python
from bluezero import adapter
from bluezero import eddystone_beacon

dongle = adapter.Adapter()

if not dongle.discoverable:
    print('Making discoverable', dongle.address)
    dongle.discoverable = True
else:
    print('Already discoverable')
eddystone_beacon.EddystoneURL('https://github.com/ukBaz')
```

Can you run a test and let me know if this works for you also?

martinfalang, 2020-11-03:
I tested it and that fixed it for me too. Thank you for the fast response!

closed 2020-11-03 by martinfalang

martinfalang, 2020-11-03:
I found another curious thing when I tried to run btmon like you did, running `python3 examples/eddystone-url-beacon.py`. For some reason i fails to set the advertising type as ADV_NONCONN_IND, but instead leaves it as ADV_IND. This happens when running any bluez advertising code and it seems that somehow my hci controller does not care whether the advertisement is registered with "broadcast" or "peripheral" as the advertisements are all the same.

Any ideas as to why this may happen? Below is some of the output from bmon when running the eddystone-url-beacon example. 

```
Bluetooth monitor ver 5.53
= Note: Linux version 5.4.0-1022-raspi (aarch64)                                             0.111749
= Note: Bluetooth subsystem version 2.22                                                     0.111762
= New Index: B8:27:EB:95:0F:6D (Primary,UART,hci0)                                    [hci0] 0.111766
= Open Index: B8:27:EB:95:0F:6D                                                       [hci0] 0.111770
= Index Info: B8:27:EB:95:0F:6D (Broadcom Corporation)                                [hci0] 0.111772
@ MGMT Open: bluetoothd (privileged) version 1.14                                   {0x0001} 0.111778
@ MGMT Open: btmon (privileged) version 1.14                                        {0x0002} 0.111858
@ MGMT Command: Add Advertising (0x003e) plen 34                             {0x0001} [hci0] 4.047704
        Instance: 1
        Flags: 0x00000000
        Duration: 0
        Timeout: 0
        Advertising data length: 23
        16-bit Service UUIDs (complete): 1 entry
          Google (0xfeaa)
        Service Data (UUID 0xfeaa): 10080367697468756200756b42617a
        Scan response length: 0
@ MGMT Event: Advertising Added (0x0023) plen 1                              {0x0002} [hci0] 4.047751
        Instance: 1
< HCI Command: LE Set Advertising Parameters (0x08|0x0006) plen 15                 #1 [hci0] 4.047813
        Min advertising interval: 1280.000 msec (0x0800)
        Max advertising interval: 1280.000 msec (0x0800)
        Type: Connectable undirected - ADV_IND (0x00)
        Own address type: Public (0x00)
        Direct address type: Public (0x00)
        Direct address: 00:00:00:00:00:00 (OUI 00-00-00)
        Channel map: 37, 38, 39 (0x07)
        Filter policy: Allow Scan Request from Any, Allow Connect Request from Any (0x00)
```

Best regards

ukBaz, 2020-11-03:
This seems like it might be a genuine bug and change in behavior. Might be worth posting to the BlueZ mailing list.