ukBaz/python-bluezero #272: Discovering devices quicker with nearby_discovery()

CLansdaleBEC, 2020-11-04:
Hello! 

I have an Ubuntu PC that I want to connect to an advertising microcontroller (nrf52840).

They connect just fine when the PC has already discovered the microcontroller using a command line tool like bluetoothctl. The issue is when the devices are interacting for the first time.

According to the Bluezero docs it seems like nearby_discovery() is just the function for the job. The issue is unless I set the timeout to be very long, like ~45 seconds, the Central cannot be created because (I think) it hasn't been discovered. 

This simple test demonstrates the behavior
```
from bluezero import adapter, central

dongles = adapter.list_adapters()
dongle = adapter.Adapter(dongles[0])

dongle.nearby_discovery(timeout=45)

myCentral = central.Central(adapter_addr=dongle.address, device_addr="AA:BB:CC:DD:EE:FF")
```
I also tried using start_discovery() and on_device_found() to register a callback to tell me when a device was connected like in the [async example](https://github.com/ukBaz/python-bluezero/blob/master/examples/async_discover_and_pair.py), but my callback was never triggered even when I know it discovered a device.

Is there a better way to discover/connect to new devices?

Thank you

ukBaz, 2020-11-04:
As I said in my [StackOverflow](https://stackoverflow.com/a/64676056/7721752) answer, I don't find people typically use `discovery` as normally it is simpler to provision the a new device the once with `bluetoothctl`. On subsequent connections, then because the device is in the known list, there is only the need to `connect` not `discover`.

Having said that, if there are examples in the library they should work. so I'll take a look.

As a side note, I am waiting for some new hardware to turn up so that I can automate the testing of scripts like these. Hopefully this will allow better testing of the library going forward.

As you are creating a central device, I would also suggest you take a look at the [pydbus](https://pypi.org/project/pydbus/) library. This would allow you to work directly with the BlueZ D-Bus API. Here is an example: https://raspberrypi.stackexchange.com/a/114205/121848

CLansdaleBEC, 2020-11-04:
Thanks, I will check out the pydbus library. 

For this case I want to test connecting to many new devices that are not in the known devices list. As you mentioned in the SO answer, it seems strange that there is a significant difference in discovery time with bluezero and bluetoothctl, because they make the same calls to bluetooth daemon. I will investigate as well and let you know if I find anything.

CLansdaleBEC, 2020-11-04:
Update: The reason the discovery was taking so long was because I had reset the CPU right before, so it took some time before it started advertising. Nothing wrong with the bluezero API. Thanks!

closed 2020-11-04 by CLansdaleBEC

ukBaz, 2020-11-04:
Appreciate the update. Thanks for letting me know.