ukBaz/python-bluezero #274: Extending ble_uart.py with additional callbacks

ukBaz, 2020-11-09:
I have tested it and it worked perfectly. It solved the issue.  
I have quick question can we use this module as asynchronously (non blocking code). 
Currently after uart.start() no code is executed after this line. 
have good experience c and c++.
Is it possible something like this? 
```
def read_callback(value):
     #do some  stuff like print value 
     print(val)

#register above callback in service or when some thing is received then the above function is called 

while True:
     if condition == True:
          uart.notify(value)

```

some thing like this.

Thanks.

_Originally posted by @arslan437 in https://github.com/ukBaz/python-bluezero/issues/266#issuecomment-723791205_

ukBaz, 2020-11-09:
I am not sure I understand the question. Does the [ble_uart.py](https://github.com/ukBaz/python-bluezero/blob/master/examples/ble_uart.py) example show you what you want?

Here is where it defines what callback to use when the client writes to the peripheral:
```python
        self.rx_uart.add_write_event(self.uart_print)
```

To send data the other way, instead of the `while True` loop, it uses the event loop and adds callback with `GLib.idle_add`. So your above code might look like:

```python
def my_callback():
     if condition == True:
          dev_loop.tx_uart.send_notify_event(value)

GLib.idle_add(my_callback)
```

arslan437, 2020-11-09:
I will test this and share the results.

ukBaz, 2021-01-01:
@arslan437 Did the above work for you? Can I close this issue?

arslan437, 2021-01-02:
I have used ESP32 instead of the raspberry pi. In the next week, I will check this.

arslan437, 2021-01-02:
But you can close the issue. If I encounter any problem I will let you know.

closed 2021-01-02 by ukBaz

y3nd, 2021-03-01:
Hello,

I'm trying to do something similar. Where should I put the GLib.idle_add line ?
Since publish() is locking the thread, how could I make my code asynchronous ?

Thanks

ukBaz, 2021-03-01:
@y3nd,

Is the `cpu_temperature.py` example similar to what you want to do? That uses the `add_timer` in the `notify_callback` which gets called on notify being enabled.