ukBaz/python-bluezero #276: RPi <> micro:bit workshop - async button notifications

johngouk, 2020-11-10:
I have tried the examples, they work fine. Just because it's interesting, I've tried using callbacks via Microbit.subscribe_button_a to detect button pushes. Strangely, the dBus code is calling my callback, not the Microbit callback. I've copied/modified the code in Microbit._decode_btn_a and then I'm able to pick up the button state value.

Any ideas?? I'm happy to do that, but it seems a bit mad when the Microbit methods are supposed to do that...

```
[pi@raspberrypi:~/Code $ python3 ubitTestCallback.py 
Starting...
ERROR:dbus.connection:Exception in handler for D-Bus signal:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 230, in maybe_handle_message
    self._handler(*args, **kwargs)
TypeError: buttonA() takes 1 positional argument but 3 were given
```

ukBaz, 2020-11-10:
OK, @johngouk you are going to have to give me a little bit more information.

I have just ran the following:
```python
from sys import argv, exit
from bluezero import microbit

def button_callback(status):
    print('Button A status: ', status)

if __name__ == '__main__':
    if len(argv) != 3:
        print(f'arg[0] <adapter_address> <device_address>')
        exit(2)
    adapter = argv[1]
    device = argv[2]
    print(f'Adapter: {adapter}\nDevice: {device}')

    ubit = microbit.Microbit(adapter_addr=adapter,
                         device_addr=device,
                         accelerometer_service=False,
                         button_service=True,
                         led_service=False,
                         magnetometer_service=False,
                         pin_service=False,
                         temperature_service=False)
    
    ubit.connect()
    ubit.subscribe_button_a(button_callback)
    print('About to launch')
    ubit.run_async()
    print('We have crashed and burned')
```
Which gave me the following output:
```none
pi@raspberrypi:~/development/python-bluezero $ python examples/async_example.py 00:02:5B:03:44:07 DC:DB:16:6B:8C:5F
Adapter: 00:02:5B:03:44:07
Device: DC:DB:16:6B:8C:5F
About to launch
Button A status:  1
Button A status:  0
Button A status:  1
Button A status:  2
Button A status:  0
```

What are we doing differently?

johngouk, 2020-11-10:
Good question! I checked my code against yours - the only difference was that yours set only `button_service=True`, all other False.  So I modified my code to set only `button_service=True`, and it didn't work. However... I then tried running your code, with the following result:

`[pi@raspberrypi:~/Code $ python3 ubitAsyncExample.py B8:27:EB:2F:E2:9F C0:39:5C:F7:68:5
Adapter: B8:27:EB:2F:E2:9F
Device: C0:39:5C:F7:68:5E
About to launch
ERROR:dbus.connection:Exception in handler for D-Bus signal:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 230, in maybe_handle_message
    self._handler(*args, **kwargs)
TypeError: button_callback() takes 1 positional argument but 3 were given
INFO:bluezero.GATT:Notifications enabled
`

I restarted my RPi, just in case, and tried again. Nope. Still doesn't work. 

What do you need to know? It could be something really obvious... I hope it is. Apologies.

ukBaz, 2020-11-10:
Is your system up to date?

These are the packages that I have:
```none
(venv) pi@raspberrypi:/tmp/test $ dpkg -l | grep bluez
ii  bluez                                5.50-1.2~deb10u1+rpt2               armhf        Bluetooth tools and daemons
ii  bluez-firmware                       1.2-4+rpt6                          all          Firmware for Bluetooth devices
```
I've tested with the version from PyPI:
```none
(venv) pi@raspberrypi:/tmp/test $ pip freeze | grep bluezero
bluezero==0.3.0
```
And a git clone:
```none
(venv) pi@raspberrypi:~/development/python-bluezero $ pip freeze | grep bluezero
-e git+https://github.com/ukBaz/python-bluezero.git@0df375b14f766ab4c2c7b0247d1297aed3887edf#egg=bluezero
```
The thing that is confusing me is that there is a mis-match in parameters. It seems like your call back is registered against the `on properties changed` D-Bus event that does have three parameters. Maybe try changing the callback to accept any number of parameters so we can see what parameters are being passed. As an example:
```python
def button_callback(*args):
    print('Button A status: ', args)

johngouk, 2020-11-10:
Yes, that's exactly what I did when it didn't work at first:

```python
def buttonA(*args):
        print('A callback')
        for count, thing in enumerate(args):
                print( '{0}. {1}'.format(count, thing))
```

giving

```
A callback
0. org.bluez.GattCharacteristic1
1. dbus.Dictionary({dbus.String('Value'): dbus.Array([dbus.Byte(1)], signature=dbus.Signature('y'), variant_level=1)}, signature=dbus.Signature('sv'))
2. dbus.Array([], signature=dbus.Signature('s'))
```
which is pretty much what the Microbit callback is expecting, since it's filtering on Value.

I don't get the same results using ```pip``` as you do - I don't have ```bluezero``` installed using pip. 
```
pi@raspberrypi:~/Code $ dpkg -l | grep bluez
ii  bluez                                 5.50-1.2~deb10u1+rpt2                  armhf        Bluetooth tools and daemons
ii  bluez-firmware                        1.2-4+rpt6                             all          Firmware for Bluetooth devices
pi@raspberrypi:~/Code $ pip freeze | grep bluezero
pi@raspberrypi:~/Code $ 
```

Should I? I just used ```sudo python3 setup.py install``` in the directory from the unzipped archive I obtained at [https://github.com/ukBaz/python-bluezero/archive/event_service.zip](url)

ukBaz, 2020-11-11:
`event_service` is a stale development branch so I can believe that doesn't work. Is there a particular reason you chose that one?

If your intention is only to have Bluezero as a dependency in another project then I would recommend using `pip install bluezero`. If you want to contribute to Bluezero then using `git clone` is the way to go. To install dependencies and setup the local repository then I use `pip install -e .[dev]`

To prove to myself that it would work from a zip file I did a test and below is my transcript for your reference.(I use Python virtual environments just to allow having several Bluezero versions installed. It is not a requirement to use Bluezero).

```none
 curl -L https://github.com/ukBaz/python-bluezero/archive/master.zip --output bluezero.zip
 unzip bluezero.zip 
 python3 -m venv --system-site-packages venv
 . venv/bin/activate
 cd python-bluezero-master/
 pip install .
 python ../async_example.py 00:02:5B:03:44:07 DC:DB:16:6B:8C:5F
```

It worked as expected.  Let me know if this fixes it for you and I can close this issue.

johngouk, 2020-11-11:
I used the event_service archive because I stumbled over a blog that referred to it whilst looking for RPi <> microbit comms - bad plan! it's not like I don't use GitHub or anything... read on...

Right. Basically, it worked using venv, and it didn't work without venv. Clearly an environment difference, and thanks for the clue.

I performed a correct install of bluezero 0.3.0 into the global python3 installation, and lo and behold, it works! Looking at the event_service version in GitHub, I can see that you previously just put the user callback into the dSub slot, so not surprisingly, that's what it called :-/.

Thanks and apologies for such a beginner error. I really appreciate your efforts - I'm hoping to use this setup with a Code Club to collect data from multiple microbits simultaneously. Should be interesting.

closed 2020-11-11 by johngouk

ukBaz, 2020-11-11:
Thanks for letting me know it is working and closing the issue.

Good luck with the Code Club experiments. Let me know if you have any feedback after the sessions.