ukBaz/python-bluezero #278: Add Feature to Advertisement (Local Name & Appearance)

drkingpo, 2020-11-23:
I'm trying to add the Advertisment of LocalName and Appearance, and show this in HeartRateSensor  - Example.

I commited already my changes and added a heartrate.py - Example

Advertisment of Local Name is working fine, but Appearance will run me into:
Failed to register advertisement: org.bluez.Error.Failed: Failed to register advertisement

Using bluez 5.54 and bluetoothctl I'm having no issues to do this advertisment.
I'm sure its again a simple code issue I'm overseeing.
Could someone help me out in here? Many thanks

Commits to advertisement.py
`    def __init__(self, advert_id, ad_type, name):
        # Setup D-Bus object paths and register service
        self.path = '/ukBaz/bluezero/advertisement{0:04d}'.format(advert_id)
        self.bus = dbus.SystemBus()
        self.eventloop = async_tools.EventLoop()
        self.interface = constants.LE_ADVERTISEMENT_IFACE
        dbus.service.Object.__init__(self, self.bus, self.path)
        self.props = {
            constants.LE_ADVERTISEMENT_IFACE: {
                'Type': ad_type,
                'ServiceUUIDs': None,
                'ManufacturerData': None,
                'SolicitUUIDs': None,
                'ServiceData': None,
                'IncludeTxPower': False,
                'Appearance': None,
                'LocalName': name
            }
        }`

`    @property
    def local_name(self):
        """Local name of the device included in Advertisement."""
        return self.Get(constants.LE_ADVERTISEMENT_IFACE,
                        'LocalName')

    @local_name.setter
    def local_name(self, name):
        self.Set(constants.LE_ADVERTISEMENT_IFACE,
                 'LocalName',
                 name)

    @property
    def appearance(self):
        """Appearance to be used in the advertising report."""
        return self.Get(constants.LE_ADVERTISEMENT_IFACE,
                        'Appearance')

    @appearance.setter
    def appearance(self, appearance):
        self.Set(constants.LE_ADVERTISEMENT_IFACE,
                 'Appearance',
                 appearance)
`

`        response['LocalName'] = self.props[interface_name]['LocalName']
        if self.props[interface_name]['Appearance'] is not None:
            response['Appearance'] = dbus.uint16(
                self.props[interface_name]['Appearance'],
                signature='q')`

ukBaz, 2020-11-29:
@drkingpo I've done a commit in a new branch at:  https://github.com/ukBaz/python-bluezero/commit/777d132e33f896ee517aa54427902609cc0466f7

This adds LocalName and Appearance. I've also updated the CPU Temperature Peripheral example to use it.
My only test so far has been to scan for it in nRF Connect which seems to be correct:
[image]

Can you try that branch out and let me know if it works for you. If it works for you, I'll do a few more manual tests and (all being well) I'll merge it into the main branch.

ukBaz, 2020-11-29:
This is a relatively small change and testing has gone OK for me so I'm merged my changes.
So with:
https://github.com/ukBaz/python-bluezero/commit/73681f9a73abac33b069966206892c824c6f0a99
I'm am closing this. Any problems then please open a new issue.

closed 2020-11-29 by ukBaz

drkingpo, 2020-11-29:
> Can you try that branch out and let me know if it works for you.

Yes its working fine! Thx