ukBaz/python-bluezero #279: Lockup calling connect() (RPi 3 Model A+ and Micro Bit v2)

alanbchristie, 2020-11-28:
I can connect to and control the Micro Bit (v2) from an IOS device and am now trying to connect to the MicroBit from a Raspberry Pi 3 Model A+.

It's a 'headless' device running **Raspberry PI OS Lite (32-bit)** (`Linux spynie-one 5.4.51-v7+ #1333 SMP Mon Aug 10 16:45:19 BST 2020 armv7l GNU/Linux`) and (I think) the necessary dbus drivers. When I follow your tutorial (https://ukbaz.github.io/howto/ubit_workshop.html) I can appear to pair, connect and disconnect from the MicroBit using the `bluetoothctl ` console: -

```bash
$ sudo bluetoothctl
Agent registered
[bluetooth]# devices
Device yy:yy:yy:yy:yy:yy BBC micro:bit
[bluetooth]# connect yy:yy:yy:yy:yy:yy
Attempting to connect to yy:yy:yy:yy:yy:yy
[CHG] Device yy:yy:yy:yy:yy:yy Connected: yes
Connection successful
[CHG] Device yy:yy:yy:yy:yy:yy Connected: no
[CHG] Deviceyy:yy:yy:yy:yy:yy Paired: yes
```

Neat, the MicroBit on-connect and on-disconnect methods run.

But when I try and run a basic connect from within Python3 on the RPi A+ the `connect()` call locks up (At the moment I'm running Python as root)...

```
$ sudo -i
# python3
Python 3.7.3 (default, Jul 25 2020, 13:03:44) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from bluezero import microbit
>>> ubit = microbit.Microbit(adapter_addr='B8:27:EB:16:77:37',
...                          device_addr='DB:EF:1A:65:1A:B2')
>>> ubit.connect()
```

- The `connect()` call appears to work (as the MicroBit's "on-connect) method runs.
- The `connect()` call locks up (does not return)
- The bluetooth connection eventually disconnects (the Microbit's "on-disconnect" method is triggered) but the `connect()` call is still locked up.

Is there some debug/log I can obtain to understand what's going on?

WayneKeenan, 2020-11-28:
The `d-feet` app can be used to debug the DBUS side.
The `hcidump` utility will log low-level bluetooth info. 

Or, if you can, just witch to `full fat` Rasbian, I never had much luck with `lite` with Python and BLE.

alanbchristie, 2020-11-28:
Switched to **Raspberry PI OS Full (32-bit)**.

The `bluetoothctl` console successfully pairs and connects to the MicroBit but the Python3 `connect()` call continues to lockup (but clearly does temporarily connect to the MicroBit as it's "on-connect" method was executed - shortly before it's "on-disconnect").

> Sadly I have run out of time today so I have not looked at `d-feet` or `hcidump` (having no experience with these tools).

It would be nice to have a timeout/log/error reported at the Python level in the `connect()`. A connect timeout is always a good idea I guess?

alanbchristie, 2020-11-28:
Incidentally ... running **Raspberry PI OS Full (32-bit)** on a RPi 4 `bluetoothctl` console successfully pairs and connects to the MicroBit as previously described. But on the RPi 4 I cannot get to the `connect()` method because when I try and create the Microbit object I'm presented with `TypeError: validate_object_path() argument 1 must be str, not None`...

```
# python3
Python 3.7.3 (default, Jul 25 2020, 13:03:44) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from bluezero import microbit
>>> ubit = microbit.Microbit(adapter_addr='B8:27:EB:16:77:37',
...                          device_addr='DB:EF:1A:65:1A:B2')
Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
  File "/usr/local/lib/python3.7/dist-packages/bluezero/microbit.py", line 80, in __init__
    device_addr=device_addr)
  File "/usr/local/lib/python3.7/dist-packages/bluezero/central.py", line 30, in __init__
    self.dongle = adapter.Adapter(adapter_addr)
  File "/usr/local/lib/python3.7/dist-packages/bluezero/adapter.py", line 85, in __init__
    self.path)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 241, in get_object
    follow_name_owner_changes=follow_name_owner_changes)
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 244, in __init__
    _dbus_bindings.validate_object_path(object_path)
TypeError: validate_object_path() argument 1 must be str, not None
```

(using `bluezero==0.3.0`)

ukBaz, 2020-11-28:
@alanbchristie the way that you are instantiating  `ubit = microbit.Microbit(adapter_addr='B8:27:EB:16:77:37', device_addr='DB:EF:1A:65:1A:B2')` might be the source of the issue. If you look at the [class definition](https://bluezero.readthedocs.io/en/stable/level1.html#bluezero.microbit.Microbit) you'll see that there is a list of services that the micro:bit will have running. If you have a service set to `True` but it is not running the micro:bit it will get an error which is what you are seeing in https://github.com/ukBaz/python-bluezero/issues/279#issuecomment-735227513

Back to the start of your issue with running RPi OS Lite. There is a better way than running with `sudo`. You can add the `pi` user to the `bluetooth` group:
```
pi@raspberrypi4b:~ $ sudo usermod -G bluetooth -a pi  
``` 

There have been some issues of late with BlueZ firmware on RPi. 
```
pi@raspberrypi:~ $ dpkg -l | grep bluez
ii  bluez                                5.50-1.2~deb10u1+rpt2               armhf        Bluetooth tools and daemons
ii  bluez-firmware                       1.2-4+rpt6                          all          Firmware for Bluetooth devices
```
If you don't have matching (or newer) versions of the above, it might worth doing a [full-upgrade](https://www.raspberrypi.org/documentation/raspbian/updating.md)

For debug information, the place to start is have `bluetoothctl` to start with and see what it prints out when you run your script. If that doesn't tell you anything then inspect the D-Bus information using `dbus-monitor --system`. For low-level information on what Bluetooth is doing use `sudo btmon`. Although sometimes the information can be found by looking at `service bluetooth status`.

alanbchristie, 2020-11-28:
A _full-upgrade_...

```
$ sudo apt update
$ sudo apt full-upgrade
$ sudo reboot
```

...followed by a reboot of the RPi seems to have fixed the `TypeError`.

> FYI - Be prepared for a long wait - the full-upgrade takes a while.

But I still get the lockup.

Running `hcidump` in a separate RPi terminal while trying to connect in another reveals the following info (obviously `yy:yy:yy:yy:yy:yy` is replaced by the real MicroBit device ID)...
```
$ hcidump -i yy:yy:yy:yy:yy:yy -x
HCI sniffer - Bluetooth packet analyzer ver 5.50
device: hci0 snap_len: 1500 filter: 0xffffffff
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x00
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
> HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x00 ncmd 1
> HCI Event: Command Status (0x0f) plen 4
    LE Read Remote Used Features (0x08|0x0016) status 0x00 ncmd 1
> HCI Event: Command Complete (0x0e) plen 14
    LE Read Remote Used Features (0x08|0x0016) ncmd 1
    00 00 00 00 00 00 00 00 00 00 00 
> HCI Event: Command Status (0x0f) plen 4
    LE Start Encryption (0x08|0x0019) status 0x00 ncmd 1
> HCI Event: Encrypt Change (0x08) plen 4
    status 0x00 handle 64 encrypt 0x01
> HCI Event: Command Complete (0x0e) plen 6
    Unknown (0x03|0x007c) ncmd 1
    00 40 00 
> HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 64 reason 0x08
    Reason: Connection Timeout
```

Does this explain why the connection is locking up?

alanbchristie, 2020-11-28:
> Thanks for the hints on adding pi to bluetooth group.

With regard to the  BlueZ firmware on the current **RPi 4** I see the following, which match your reference...

```
$ dpkg -l | grep bluez
ii  bluez                                 5.50-1.2~deb10u1+rpt2                  armhf        Bluetooth tools and daemons
ii  bluez-firmware                        1.2-4+rpt6                             all          Firmware for Bluetooth devices
ii  bluez-hcidump                         5.50-1.2~deb10u1+rpt2                  armhf        Analyses Bluetooth HCI packets
```

Im using **RPi 4** (and **RPi 3 A+**) connected to a **MicroBit v2** and I think my `Microbit()` constructor services match those on the MicroBit.

ukBaz, 2020-11-28:
As @WayneKeenan mentioned, RPi OS Lite can be a tricky to get working. However the full OS should be fairly reliable.

Let's remove Bluezero from the equation. Can you install the Python library [pydbus](https://pypi.org/project/pydbus/) and try the following:

```python
import pydbus
from time import sleep
from gi.repository import GObject

# Setup of device specific values
dev_id = 'DE:82:35:E7:43:BE'
btn_a_uuid = 'E95DDA90-251D-470A-A062-FA1922DFA9A8'

# DBus object paths
bluez_service = 'org.bluez'
adapter_path = '/org/bluez/hci0'
device_path = f"{adapter_path}/dev_{dev_id.replace(':', '_')}"

# setup dbus
bus = pydbus.SystemBus()
mngr = bus.get(bluez_service, '/')
adapter = bus.get(bluez_service, adapter_path) 
device = bus.get(bluez_service, device_path)

# Access adapter properties
print(adapter.Name)
print(adapter.Powered)
print(adapter.Address)

# Commands and properties from the DBus API
print(dir(adapter))
print(dir(device))

# Assume device has been paired already so can use connect
device.Connect()
sleep(10)
device.Disconnect()

ukBaz, 2020-11-28:
If the above works, then replace the `sleep(10)` with the following to read the value from button A.
```python
while not device.ServicesResolved:
    sleep(0.5)

def get_characteristic_path(dev_path, uuid):
    """Look up DBus path for characteristic UUID"""
    mng_objs = mngr.GetManagedObjects()
    for path in mng_objs:
        chr_uuid = mng_objs[path].get('org.bluez.GattCharacteristic1', {}).get('UUID')
        if path.startswith(dev_path) and chr_uuid == uuid.casefold():
           return path

def as_int(value):
    """Create integer from bytes"""
    return int.from_bytes(value, byteorder='little')

# Characteristic DBus information
btn_a_path = get_characteristic_path(device._path, btn_a_uuid)
btn_a = bus.get('org.bluez', btn_a_path)
# Read value of characteristics
print(btn_a.ReadValue({}))
# [0]
```

alanbchristie, 2020-11-28:
OK - the initial code works (i.e. just sleeping). With my `dev_id` and some connecting/disconnecting log the basic connect/disconnect seems to have worked...

```
raspberrypi
True
DC:A6:32:03:A2:C5
['ActiveInstances', 'Address', 'AddressType', 'Alias', 'Class', 'ConnectDevice', 'Discoverable', 'DiscoverableTimeout', 'Discovering', 'Get', 'GetAll', 'GetDiscoveryFilters', 'Introspect', 'Modalias', 'Name', 'Pairable', 'PairableTimeout', 'Powered', 'PropertiesChanged', 'Register', 'RegisterAdvertisement', 'RegisterApplication', 'RegisterEndpoint', 'RegisterPlayer', 'RemoveDevice', 'Set', 'SetDiscoveryFilter', 'StartDiscovery', 'StopDiscovery', 'SupportedIncludes', 'SupportedInstances', 'UUIDs', 'Unregister', 'UnregisterAdvertisement', 'UnregisterApplication', 'UnregisterEndpoint', 'UnregisterPlayer', '_Introspect', '__class__', '__delattr__', '__dict__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattribute__', '__getitem__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__le__', '__lt__', '__module__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', '__weakref__', '_bus', '_bus_name', '_object', '_path', 'onPropertiesChanged']
['Adapter', 'Address', 'AddressType', 'AdvertisingData', 'AdvertisingFlags', 'Alias', 'Appearance', 'Blocked', 'CancelPairing', 'Class', 'Connect', 'ConnectProfile', 'Connected', 'Disconnect', 'DisconnectProfile', 'Get', 'GetAll', 'Icon', 'Introspect', 'LegacyPairing', 'ManufacturerData', 'Modalias', 'Name', 'Pair', 'Paired', 'PropertiesChanged', 'RSSI', 'ServiceData', 'ServicesResolved', 'Set', 'Trusted', 'TxPower', 'UUIDs', '_Introspect', '__class__', '__delattr__', '__dict__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattribute__', '__getitem__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__le__', '__lt__', '__module__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', '__weakref__', '_bus', '_bus_name', '_object', '_path', 'onPropertiesChanged']
Connecting...
Connected
Disconnecting...
Disconnected
```

But when I replace the sleep with your button-reading block it locks up.

alanbchristie, 2020-11-28:
Noticed this in the **BLE connection timeout on Pi3 B+** issue (https://raspberrypi.stackexchange.com/questions/116651/ble-connection-timeout-on-pi3-b)...

...it mentions _On Pi 3B+/4 I had to downgrade to bluez version: 5.50-1.2~deb10u1_ (i.e. go from `5.50-1.2~deb10u1+rpt2` to `5.50-1.2~deb10u1`?)

alanbchristie, 2020-11-28:
If I just re-run the code again and again sometimes it gets through the sleep but whether the MicroBit Button-A is depressed or not (for the rare occasion it actually runs to completion) I see...

```
Connecting...
Connected
Resolving...
Resolved
[0]
Disconnecting...
Disconnected
```

The button value is always zero but I can't trust it because (about) 19 times out of 20 it fails for some other reason and so I suspect there is an unresolved timing issue or some other connect issue.

I don't have a MicroBit v1.x and just thought...

- Does all this actually work on a **MicroBit v2**?
- Is `btn_a_uuid` a universal constant? i.e. is it valid for every MicroBit (v1.x or v2)?

ukBaz, 2020-11-28:
My micro:bit V2 has just arrived. I'll run a test and report back here...

ukBaz, 2020-11-28:
Tested on micro:bit V2 without issue:
[image]

With the code on the RPi:
```python
from time import sleep
import pydbus
from gi.repository import GLib

DEVICE_ADDR = 'E9:06:4D:45:FC:8D'
BTN_A_STATE = 'E95DDA90-251D-470A-A062-FA1922DFA9A8'

# DBus object paths
BLUEZ_SERVICE = 'org.bluez'
ADAPTER_PATH = '/org/bluez/hci0'
device_path = f"{ADAPTER_PATH}/dev_{DEVICE_ADDR.replace(':', '_')}"

# setup dbus
bus = pydbus.SystemBus()
mngr = bus.get(BLUEZ_SERVICE, '/')
adapter = bus.get(BLUEZ_SERVICE, ADAPTER_PATH) 
device = bus.get(BLUEZ_SERVICE, device_path)

device.Connect()

while not device.ServicesResolved:
    sleep(0.5)

def get_characteristic_path(dev_path, uuid):
    """Look up DBus path for characteristic UUID"""
    mng_objs = mngr.GetManagedObjects()
    for path in mng_objs:
        chr_uuid = mng_objs[path].get('org.bluez.GattCharacteristic1', {}).get('UUID')
        if path.startswith(dev_path) and chr_uuid == uuid.casefold():
           return path
# Characteristic DBus information
btn_a_path = get_characteristic_path(device._path, BTN_A_STATE)
btn_a = bus.get(BLUEZ_SERVICE, btn_a_path)
# Read button without event loop
print(btn_a.ReadValue({}))

# Enable eventloop for notifications
def btn_handler(iface, prop_changed, prop_removed):
    """Notify event handler for button press"""
    if 'Value' in prop_changed:
        new_value = prop_changed['Value']
        print(f"Button A state: {new_value}")
        print(f'As byte: {bytes(new_value)}')
        print(f'As bytearray: {bytearray(new_value)}')
        print(f'As int: {int(new_value[0])}')
        print(f'As bool: {bool(new_value[0])}')

mainloop = GLib.MainLoop()
btn_a.onPropertiesChanged = btn_handler
btn_a.StartNotify()
try:
    mainloop.run()
except KeyboardInterrupt:
    mainloop.quit()
    btn_a.StopNotify()
    device.Disconnect()
```

This was run on:
[image]
```none
(venv) pi@raspberrypi:~/stack_overflow $ uname -a
Linux raspberrypi 5.4.72-v7+ #1356 SMP Thu Oct 22 13:56:54 BST 2020 armv7l GNU/Linux
(venv) pi@raspberrypi:~/stack_overflow $ dpkg -l | grep bluez
ii  bluez                                5.50-1.2~deb10u1+rpt2               armhf        Bluetooth tools and daemons
ii  bluez-firmware                       1.2-4+rpt6                          all          Firmware for Bluetooth devices
(venv) pi@raspberrypi:~/stack_overflow $ python btn_notify.py 
[0]
Button A state: [1]
As byte: b'\x01'
As bytearray: bytearray(b'\x01')
As int: 1
As bool: True
Button A state: [0]
As byte: b'\x00'
As bytearray: bytearray(b'\x00')
As int: 0
As bool: False
```

The only thing that didn't work was the long press on the button but that is a known issue with V2.

Not sure what else I can do to help given that I can't reproduce the issue. It certainly doesn't look to be a Bluezero specific issue

alanbchristie, 2020-11-28:
OK - thanks. I'll re-flash my MicroBit later and try with _clean_ distributions on an RPi 4 and RPi 3A+.

alanbchristie, 2020-11-28:
Luckily I also have a **RPi Model 3B V1.2** and (as you'd tested it on this model and not the one I'd mentioned) I thought I'd better just make sure that I can reproduce your success. And, yes, I can. The **RPi Model 3B V1.2** does work with MicroBit V2: -

- I flash the RPi SD card with **Raspberry Pi OS Full (32-bit)**
- I pair the device with the RPi using `bluetoothctl`
- `pip3 install pydbus`
- `sudo usermod -G bluetooth -a pi` (and logout and back in for it to take effect)
- Replicate your `btn_notify.py` (changing the device address)
- And run `python3  btn_notify.py`

...and it works

The OS and `bluez` package details are as follows...

```
$ uname -a
Linux raspberrypi 5.4.51-v7+ #1333 SMP Mon Aug 10 16:45:19 BST 2020 armv7l GNU/Linux
$ dpkg -l | grep bluez
ii  bluez                                 5.50-1.2~deb10u1+rpt2                  armhf        Bluetooth tools and daemons
ii  bluez-firmware                        1.2-4+rpt4                             all          Firmware for Bluetooth devices
```

alanbchristie, 2020-11-29:
On the **Pi Model 3B+ V1.3** - although pairing with the MicroBit is listed as successful from the `bluetoothctl` console the MicroBit displays a large red "cross" rather than a "tick" - indicating pairing failure. Execution of the `btn_notify.py` code simply results in: -

```
$ python3 btn_notify.py
Traceback (most recent call last):
  File "btn_notify.py", line 19, in <module>
    device.Connect()
  File "/home/pi/.local/lib/python3.7/site-packages/pydbus/proxy_method.py", line 75, in __call__
    0, timeout_to_glib(timeout), None).unpack()
gi.repository.GLib.Error: g-io-error-quark: GDBus.Error:org.bluez.Error.Failed: Software caused connection abort (36)
```

> This is hardly surprising as pairing on the **3B+ V1.3** doesn't actually work.

alanbchristie, 2020-11-29:
On the **Pi Model 4B  V1.1** pairing appears to be successful but the response from the `pair` command is significantly different from that on the **3B V1.2**: -

```
[bluetooth]# pair DB:EF:1A:65:1A:B2
Attempting to pair with DB:EF:1A:65:1A:B2
[CHG] Device DB:EF:1A:65:1A:B2 Connected: yes
[CHG] Device DB:EF:1A:65:1A:B2 UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Device DB:EF:1A:65:1A:B2 UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Device DB:EF:1A:65:1A:B2 UUIDs: 0000180a-0000-1000-8000-00805f9b34fb
[CHG] Device DB:EF:1A:65:1A:B2 UUIDs: 0000fe59-0000-1000-8000-00805f9b34fb
[CHG] Device DB:EF:1A:65:1A:B2 UUIDs: e95d93af-251d-470a-a062-fa1922dfa9a8
[CHG] Device DB:EF:1A:65:1A:B2 UUIDs: e97dd91d-251d-470a-a062-fa1922dfa9a8
[CHG] Device DB:EF:1A:65:1A:B2 ServicesResolved: yes
[CHG] Device DB:EF:1A:65:1A:B2 Paired: yes
[NEW] Primary Service
	/org/bluez/hci0/dev_DB_EF_1A_65_1A_B2/service000a
	00001801-0000-1000-8000-00805f9b34fb
	Generic Attribute Profile
[NEW] Characteristic
	/org/bluez/hci0/dev_DB_EF_1A_65_1A_B2/service000a/char000b
	00002a05-0000-1000-8000-00805f9b34fb
	Service Changed
[NEW] Descriptor
	/org/bluez/hci0/dev_DB_EF_1A_65_1A_B2/service000a/char000b/desc000d
	00002902-0000-1000-8000-00805f9b34fb
	Client Characteristic Configuration
[NEW] Primary Service
	/org/bluez/hci0/dev_DB_EF_1A_65_1A_B2/service000e
	0000fe59-0000-1000-8000-00805f9b34fb
	Nordic Semiconductor ASA
[NEW] Characteristic
	/org/bluez/hci0/dev_DB_EF_1A_65_1A_B2/service000e/char000f
	8ec90004-f315-4f60-9fb8-838830daea50
	Vendor specific
[NEW] Descriptor
	/org/bluez/hci0/dev_DB_EF_1A_65_1A_B2/service000e/char000f/desc0011
	00002902-0000-1000-8000-00805f9b34fb
	Client Characteristic Configuration
[NEW] Primary Service
	/org/bluez/hci0/dev_DB_EF_1A_65_1A_B2/service0012
	e97dd91d-251d-470a-a062-fa1922dfa9a8
	Vendor specific
[NEW] Characteristic
	/org/bluez/hci0/dev_DB_EF_1A_65_1A_B2/service0012/char0013
	e97d3b10-251d-470a-a062-fa1922dfa9a8
	Vendor specific
[NEW] Descriptor
	/org/bluez/hci0/dev_DB_EF_1A_65_1A_B2/service0012/char0013/desc0015
	00002902-0000-1000-8000-00805f9b34fb
	Client Characteristic Configuration
[NEW] Primary Service
	/org/bluez/hci0/dev_DB_EF_1A_65_1A_B2/service0016
	0000180a-0000-1000-8000-00805f9b34fb
	Device Information
[NEW] Characteristic
	/org/bluez/hci0/dev_DB_EF_1A_65_1A_B2/service0016/char0017
	00002a24-0000-1000-8000-00805f9b34fb
	Model Number String
[NEW] Characteristic
	/org/bluez/hci0/dev_DB_EF_1A_65_1A_B2/service0016/char0019
	00002a25-0000-1000-8000-00805f9b34fb
	Serial Number String
[NEW] Characteristic
	/org/bluez/hci0/dev_DB_EF_1A_65_1A_B2/service0016/char001b
	00002a26-0000-1000-8000-00805f9b34fb
	Firmware Revision String
[NEW] Primary Service
	/org/bluez/hci0/dev_DB_EF_1A_65_1A_B2/service001d
	e95d93af-251d-470a-a062-fa1922dfa9a8
	MicroBit Event Service
[NEW] Characteristic
	/org/bluez/hci0/dev_DB_EF_1A_65_1A_B2/service001d/char001e
	e95d9775-251d-470a-a062-fa1922dfa9a8
	MicroBit Event Data
[NEW] Descriptor
	/org/bluez/hci0/dev_DB_EF_1A_65_1A_B2/service001d/char001e/desc0020
	00002902-0000-1000-8000-00805f9b34fb
	Client Characteristic Configuration
[NEW] Characteristic
	/org/bluez/hci0/dev_DB_EF_1A_65_1A_B2/service001d/char0021
	e95d5404-251d-470a-a062-fa1922dfa9a8
	MicroBit Client Events
[NEW] Characteristic
	/org/bluez/hci0/dev_DB_EF_1A_65_1A_B2/service001d/char0023
	e95d23c4-251d-470a-a062-fa1922dfa9a8
	MicroBit Client Requirements
[NEW] Characteristic
	/org/bluez/hci0/dev_DB_EF_1A_65_1A_B2/service001d/char0025
	e95db84c-251d-470a-a062-fa1922dfa9a8
	MicroBit Requirements
[NEW] Descriptor
	/org/bluez/hci0/dev_DB_EF_1A_65_1A_B2/service001d/char0025/desc0027
	00002902-0000-1000-8000-00805f9b34fb
	Client Characteristic Configuration
Pairing successful
[CHG] Device DB:EF:1A:65:1A:B2 ServicesResolved: no
[CHG] Device DB:EF:1A:65:1A:B2 Connected: no
[bluetooth]# paired-devices
Device DB:EF:1A:65:1A:B2 BBC micro:bit [vapuv]
```

But, even though I do everything else in the same order as I did on the **3B V1.2** model `btm_notify.py` fails. i.e. I do...

- `pip3 install pydbus`
- `sudo usermod -G bluetooth -a pi` (and logout and back in for it to take effect)
- I recreate your `btn_notify.py` (changing the device address)
- And run `python3 btn_notify.py`

But the program fails with the same error we see on the **3B+**: -

```
$ python3 btn_notify.py
Traceback (most recent call last):
  File "btn_notify.py", line 19, in <module>
    device.Connect()
  File "/home/pi/.local/lib/python3.7/site-packages/pydbus/proxy_method.py", line 75, in __call__
    0, timeout_to_glib(timeout), None).unpack()
gi.repository.GLib.Error: g-io-error-quark: GDBus.Error:org.bluez.Error.Failed: Software caused connection abort (36)
```

alanbchristie, 2020-11-29:
Finally, back to the original **Pi Model 3A+ V1.0** the pairing works (i.e. it's successful and the MicroBit responds with a "tick") but the example `btn_notify.py` just lock up again, there is no error. The MicroBit responds with a "C" in the LED display and then, a few moments later disconnects displaying a "D".

### Summary

- **Pi Model 3B V1.2**
  [ok]
- **Pi Model 3B+ V1.3**
  [FAILED - GLib.Error]
- **Pi Model 4B V1.1**
  [FAILED - GLib.Error]
- **Pi Model 3A+ V1.0**
  [FAILED - locks up]

Obviously the hardware (SoC etc.) on each of these devices is significantly different and they all comply with different Bluetooth specs but I can only confirm that your test works on the earliest Model 3 - and fails on all the other models.

---

Regarding my original issue (i.e. the lockup on the **3A+**) -  do you have anything newer than the **Pi Model 3B V1.2**? i.e. can you confirm that the code works on anything other than the **Pi Model 3B V1.2**?

alanbchristie, 2020-11-29:
I am re-testing the failed devices, applying the following to each: -

```
$ sudo apt update
$ sudo apt full-upgrade
```

ukBaz, 2020-11-29:
I only have two RPi Model 3B V1.2 so can't test with other models.

My sense of this is that it is more likely to be a hardware issue along the lines of https://github.com/RPi-Distro/firmware-nonfree/ that is being discussed in issues 8

alanbchristie, 2020-11-29:
OK, thanks. I'll stop re-testing as it's clear that this only works on the **Pi Model 3B V1.2**. I'll adjust the issue subject and leave it here.

> Is it worth updating The **Status** text in your repo's `README` and your wonderful _ubit workshop_ tutorial with a little "verified on the following hardware"? Just in case anyone else tries connecting to MicroBits with anything other than a **Pi Model 3B V1.2**

Sadly I need to connect to MicroBits via an A+ but clearly that's not possible atm so I'll concentrate on testing my code with the **Pi Model 3B V1.2** in the hope that (at some point) this gets fixed for the newer devices.

alanbchristie, 2020-11-29:
Interestingly ... I've also had an opportunity to test the code against the **Pi Zero W V1.1** ... and this works!

That's nice (for me) as it's a viable alternative to the **A+**.

ukBaz, 2021-01-01:
Closing this because of inactivity. Please reopen if you think there is still something Bluezero related.

closed 2021-01-01 by ukBaz