ukBaz/python-bluezero #282: ImportError using bluezero on Raspberry Pi 4 Model B

vkb008, 2020-12-05:
Hello,

I am a beginner to bluetooth programming and am having issues using the utlitizing bluezero API on my Pi 4.

I am programming through Mu on my Pi 4.

I get this output on the console when attempting to flash my code:

```
Traceback (most recent call last):
  File "main.py", in <module>
ImportError: no module named 'bluezero'
MicroPython v1.13 on 2020-11-24; micro:bit v2.0.0-beta.1 with nRF52833
```

I initially followed the steps on this website to install bluezero and pair my micro:bit v2 with my Pi 4: https://ukbaz.github.io/howto/ubit_workshop.html

I ran on the Pi 4 terminal: `sudo pip3 install bluezero ` 

I also confirmed the micro:bit is connected to my Pi 4: 

```
[bluetooth]# devices
Device E8:9E:9F:A9:79:33 BBC micro:bit [gatog]
```

I just checked my bluetooth version on my Pi 4 and I have 5.50 installed (documentation on bluezero seems to suggest it is intended to work with version 5.43, so I'm not sure if this is source of the probem).

I then tried to input some of the commands listed here: https://bluezero.readthedocs.io/en/stable/install_bluez.html

```
sudo apt-get install bluetooth
sudo apt-get install bluez-tools
sudo apt-get install bluez-test-scripts
sudo apt-get install bluez-hcidump
sudo apt-get install python-bluez
sudo pip3 install Sphinx
sudo pip3 install sphinx_rtd_theme
sudo pip3 install pycodestyle
```

However, I'm a bit lost and still getting the original ImportError. Getting some help to diagnose what is happening would be really great.

Thanks!

WayneKeenan, 2020-12-05:
It may be the case your/the default Python is python2. Try re-running the script with ‘python3’ instead of ‘python’

WayneKeenan, 2020-12-05:
(Because ‘pip3’ installed it into Python 3 libs)

WayneKeenan, 2020-12-05:
You could also try just doing ‘pip install’ (omit the 3, but, this might help Mu but I’m not sure if BlueZero still supports Python 2)

Hope the pointers help, sorry it’s not a suggestion I can test to repeat the problem.

ukBaz, 2020-12-05:
I've only limited experience with Mu. It used to be that it had its own version of Python that it would use and not the system version. It also had the restriction that you could only use the installed packages and you could not add additional ones. Has that changed? If not, then adding Bluezero to the system package will not make it available in Mu.

vkb008, 2020-12-05:
> I've only limited experience with Mu. It used to be that it had its own version of Python that it would use and not the system version. It also had the restriction that you could only use the [...]

@ukBaz I see, in that case would do you have another IDE you'd recommmend? I have Thonny Python IDE also installed.

ukBaz, 2020-12-05:
If you search for "third party packages" in the [Mu Issues](https://github.com/mu-editor/mu/issues?q=is%3Aissue+%22third+party+packages%22+) there is lots of activity around this topic.

There seems to be some instructions on how to add packages to Mu at:
https://automatetheboringstuff.com/2e/appendixa/

vkb008, 2020-12-05:
@ukBaz so looking at the microbit support site link (https://support.microbit.org/support/solutions/articles/19000083637-using-the-micro-bit-wirelessly-)

`It looks like BLE is not supported in MicroPython`

I think this is root issue haha.

That being said, is there an IDE/program you'd recommend for me to use/implement bluezero? Preferrably something that is compatible with Linux/Ubuntu.

ukBaz, 2020-12-06:
Apologies, I had missed you were trying to run micropython on the micro:bit. You had referenced the Bluezero workshop doesn't use micropython as so I had assumed you were using Mu in Python mode. Anyway, you are correct that currently there is no BLE with micropython on the micro:bit. However with V2 I am expecting that to change.

Bluezero works with regular Python on the RPi and is not geared towards a particular IDE or editor. You should be able to use any that you feel comfortable with.

I'll go ahead and close this issue but please open another one if you have questions about using Bluezero.

closed 2020-12-06 by ukBaz

vkb008, 2020-12-07:
@ukBaz Thanks for confirming my suspicion. I am trying PyCharm and will see if I can this IDE to work.

vkb008, 2020-12-07:
@ukBaz I have installed PyCharm and installed the bluezero Python Interpreter in PyCharm, in addition to running the `sudo pip3 install bluezero` command in my Pi 4 Terminal.

In order to flash my python program (I am starting with the first demo program (**sending a message to the micro:bit**), I am uploading the code to the [micro bit python editor](python.microbit.org/v/2), downloading the .hex file, then dragging it into the microbit drive.

However, when I do this, I am still getting a `ImportError: no module named 'bluezero'`

Any suggestions?

ukBaz, 2020-12-07:
@vkb008 , In the [workshop ](https://ukbaz.github.io/howto/ubit_workshop.html) you said you are following, there are instructions for setting up the micro:bit using  https://makecode.microbit.org/. In that workshop there is no python on the micro:bit. Python with BLE will not work with the micro:bit.  The Bluezero library is for use on Linux machines like the Raspberry Pi. What you are trying to do is not designed to work and will not work.

vkb008, 2020-12-08:
@ukBaz ohh I see. I think I misunderstood your workshop. I assumed the following programs were intended to be flashed onto the micro:bit. But it sounds like they are just to be run on the Pi itself. Is that correct?