ukBaz/python-bluezero #283: AttributeError: module 'gobject' has no attribute 'MainLoop'

vkb008, 2020-12-08:
I'm programming through PyCharm, before running into this error, I've needed to do the following:

1. Follow installation of bluezero per [workshop](https://ukbaz.github.io/howto/ubit_workshop.html) guidance
2. Install Python Interpreter packages
- bluezero
- dbus-python
- gi
- gobject
3. Fix "runtime issues" on __init__.py:
- Changed print syntax on lines 39 and 41 to include parantheses around the values to be printed

After doing these 3 things, I've run into this error:
[image]

Any suggestions? Please let me know if additional background information is needed to better help diagnose.

ukBaz, 2020-12-08:
This is a very tried and tested workshop so I would be surprised you are having to do steps 2 & 3 that you describe above.

For 3, what `init.py` file are you referring to?

For 2, I suspect that when you created the virtual environment in PyCharm you didn't select the "Inherit global site-packages" checkbox as shown in the PyCharm Help:
https://www.jetbrains.com/help/pycharm/creating-virtual-environment.html#python_create_virtual_env
[image]

The other option is to tell PyCharm to run with the "System Interpreter" 

The packages you refer to are already installed on the RPi OS in Python3

vkb008, 2020-12-08:
@ukBaz ah thank you for that suggestion. I'm sure this code is tried and proven, I'm just a rookie learning how to properly use IDEs haha.

__init__.py is the file located in this location on my pi: /home/pi/PycharmProjets/MicrobitTestBluetooth/venv/lib/python3.7/site-packages/gi

However, I created a new projected and made sure to select "Inherit global site-packages" and this seemed to fix issues related to packages.

I'm now getting this runtime error:

```
/home/pi/PycharmProjects/TestMicrobitBle2/venv/bin/python/home/pi/PycharmProjects/TestMicrobitBle2/TestMicrobitSendText.py
Traceback (most recent call last):
  File "/home/pi/PycharmProjects/TestMicrobitBle2/TestMicrobitSendText.py", line 12, in <module>
    ubit.connect()
  File "/usr/local/lib/python3.7/dist-packages/bluezero/microbit.py", line 148, in connect
    self.ubit.connect()
  File "/usr/local/lib/python3.7/dist-packages/bluezero/central.py", line 79, in connect
    self.rmt_device.connect()
  File "/usr/local/lib/python3.7/dist-packages/bluezero/device.py", line 266, in connect
    self.remote_device_methods.Connect()
  File "/home/pi/.local/lib/python3.7/site-packages/dbus/proxies.py", line 147, in __call__
    **keywords)
  File "/home/pi/.local/lib/python3.7/site-packages/dbus/connection.py", line 653, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.

Process finished with exit code 1
```

ukBaz, 2020-12-08:
Have you followed the "Pairing with the micro:bit" section in the workshop?

vkb008, 2020-12-09:
@ukBaz yes and I confirmed the micro:bit has been paired with the Pi:

[image]

ukBaz, 2020-12-16:
Can you do `info E8:9E:9F:A9:79:33` and show the output of that please.

ukBaz, 2021-01-01:
Closing this because of inactivity. If it is still an issue then please reopen.

closed 2021-01-01 by ukBaz