ukBaz/python-bluezero #284: Multiple Beacon Advertisements

nickdataskill, 2020-12-10:
Hi
Great code - does all the complex stuff for me!
However... I'm trying to setup an Eddystone beacon with multiple adverts attached, but seems the sample code isn't ideally suited to do this and I'm mystified about the best way to go about modifying. Can you give me some pointers please?

ukBaz, 2020-12-10:
This seems to have been caused by the hard coded `1` in the broadcaster.py file:
https://github.com/ukBaz/python-bluezero/blob/0ac465ac5ca8c790a3f4d066eaf25126e44e0b17/bluezero/broadcaster.py#L27

Which is probably not ideal but it was done to simplify the UI.

There is access to the details and indeed there is an example in advertisement.py 
https://github.com/ukBaz/python-bluezero/blob/0ac465ac5ca8c790a3f4d066eaf25126e44e0b17/bluezero/advertisement.py#L58-L70

It requires the `1` to be a unique number for each advert.

Not sure it's the best example, but here is something I put together that will advertise two URL's:

```python
from bluezero import adapter
from bluezero import advertisement 
from bluezero import async_tools
from bluezero.tools import url_to_advert

dongle = adapter.Adapter()
dongle.discoverable = True
beacon1 = advertisement.Advertisement(1, 'broadcast')
beacon2 = advertisement.Advertisement(2, 'broadcast')
beacon1.service_UUIDs = ['FEAA']
beacon2.service_UUIDs = ['FEAA']
beacon1.service_data =  {'FEAA': url_to_advert(url='https://www.bluetooth.com/', frame_type=0x10, tx_power=0x08)}
beacon2.service_data =  {'FEAA': url_to_advert(url='https://microbit.org/', frame_type=0x10, tx_power=0x08)}
ad_manager = advertisement.AdvertisingManager()
ad_manager.register_advertisement(beacon1, {}) 
ad_manager.register_advertisement(beacon2, {}) 
mainloop = async_tools.EventLoop()
mainloop.run()
```

nickdataskill, 2020-12-10:
Fantastic, many thanks will give that a try
  

  
Cheers
  

  
Nick

ukBaz, 2021-01-01:
Closing this as this is already being tracked with #35

closed 2021-01-01 by ukBaz