ukBaz/python-bluezero #290: service: "Device Information" will be created automatically with the PnP_ID characteristic Bluez version above 5.50

inonoob, 2020-12-16:
Hey, 

I' currently working on a project where I use the GATT Server and the GATT advertiser. 

**Problem**:  
- If Bluez is compiled from source following this post link](https://scribles.net/updating-bluez-on-raspberry-pi-from-5-43-to-5-50/). (Please replace the Bluez version with the actual one [Bluez5.5](https://mirrors.edge.kernel.org/pub/linux/bluetooth/bluez-5.55.tar.gz) )
- Then if the Python script is used with bluez version 5.50 or 5.54  in order to create the Gatt Server then the service: "Device Information" will be created automatically with the PnP_ID characteristic. 
- If in the Gatt Server python script the service: "Device Information" is also defined, then it will result in having 2 "Device Information"  Services which are not combined. (Checkable via nRF app) 
- In my case for example, my scripts checks if "Device Information" is available and want to check the software version string characteristic. But as it already sees the "Device Information" service with only having the PnP_ID characteristic, my script fails. 
I have no idea for the moment how to prevent Bluez version above 5.50 to create "Device Information" service with the PnP_ID characteristic. 

**Short Term solution**: 
- Do not upgrade Bluez higher as 5.50 if the full control over the Service: Device Information wants be maintained. 

**Tested on/with**:
- Raspberry pi 1 (Gatt Server)
  - Bluetooth dongle 4.1 
  - Raspberry Pi OS Lite 02.12.2020
  - Bluez 5.44 ==> Service: Device Information automatically created with PnP_ID characteristic
  - Bluez 5.55 ==> Service: Device Information automatically created with PnP_ID characteristic
  - Bluez 5.50 ==> No Service Device Information automatically created
- Raspberry pi zero W (Gatt Server)
  - Bluetooth on board 4.1
  - Raspberry Pi OS Lite 02.12.2020
  - Bluez 5.44 ==> Service: Device Information automatically created with PnP_ID characteristic
  - Bluez 5.55 ==> Service: Device Information automatically created with PnP_ID characteristic
  - Bluez 5.50 ==> No Service Device Information automatically created
 - Android: (BLE client)
   - Version 6.0 
   - Version 8.0
   - Version 10.0
   - Checked with nRF app

**Request**: 
- Has someone already a longterm solution for this problem. Should I change something during the compilation ? or can I control via the python GATT server script and DBUS that Bluez shouldn't create the "Device Information" service on it's own.

ukBaz, 2020-12-16:
I see you have also asked the same question over at: https://github.com/getsenic/gatt-python/issues/60

From a Bluezero point of view, I don't see this as a library issue. 

After reading the [GATT Specification Supplement](https://www.bluetooth.org/DocMan/handlers/DownloadDoc.ashx?doc_id=502132&vId=542989) I am also of the opinion that the PnP ID characteristic is not something the target audience of Bluezero would be concerned about.

I would suggest you contact the BlueZ project with your question. Their contact details are at: http://www.bluez.org/contact/

closed 2020-12-16 by ukBaz

inonoob, 2020-12-16:
Hey,

Thanks for the feedback and also the contact information. I will write them.

Best Regards

inonoob, 2020-12-16:
Hey, 

For the archive, I found the patch in the Bluez git: [Link](https://git.kernel.org/pub/scm/bluetooth/bluez.git/commit/?id=d5e07945c4aa36a83addc3c269f55c720c28afdb) 

If this part is removed from source and re-compiled it fixes the issue but it isn't a long-term solution. I could find docs how to set in the /etc/bluetooth/main.conf the correct flag so it doesn't get picked up 

Best regards

ukBaz, 2020-12-16:
It looks like this is the function that reads the `main.conf` for these values:
https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/src/main.c?id=d5e07945c4aa36a83addc3c269f55c720c28afdb#n182

so I think this is the important line in the conf file:
https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/src/main.conf?id=d5e07945c4aa36a83addc3c269f55c720c28afdb#n30

Although I haven't tested this....

inonoob, 2020-12-16:
Thing is, If it is commented out in the main.conf as it is by default, shouldn't it be inactive ?

ukBaz, 2020-12-16:
Does the following put defaults if it is not specified in the `main.conf` file?
https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/src/main.c?id=d5e07945c4aa36a83addc3c269f55c720c28afdb#n677

dhruvelectro, 2023-05-23:
I have found same issue for our ble peripheral code. I have done following:
`sudo nano /etc/bluetooth/main.conf`

Added following line (Enabled the DeviceID and set) followings
`DeviceID = false`

restarted the blutooth deamon
```
sudo systemctl daemon-reload
sudo systemctl restart bluetooth
sudo systemctl status bluetooth
```

then if I re-run my BLE Peripheral python code