ukBaz/python-bluezero #291: Disconnected with IOS phone

Star152, 2020-12-17:
When i running the examples/cpu_temperature.py in RPi,there do not have any errors in terminal.But use 'nRF Connect' to connect the service there are the errors:

> [Callback]centralManager(central,didDisconnectPeripheral:peripheral,error:Failed to encrypt the connection,the connection has timed out unexpectedly.)

Platform is a RPi3 running Raspbian GNU/Linux 10
Kernel Linux raspberrypi 4.19.97-v7l+
bluez package version 5.50-1.2~deb10u1+rpt1
Python 3.7.3

Thanks

ukBaz, 2020-12-17:
You seem to be a little behind on your versions. I would recommend you do an update.

There were some issues with earlier versions having issues.

If you still have issues can you report the output from `dpkg -l | grep bluez`

Star152, 2020-12-17:
Packages that I have on my pi：

> pi@ezb-test:~/python-bluezero/examples $ dpkg -l | grep bluez
ii  bluez                                5.50-1.2~deb10u1+rpt2               armhf        Bluetooth tools and daemons
ii  bluez-firmware                       1.2-4+rpt3                          all          Firmware for Bluetooth devices

Star152, 2020-12-17:
Packages that I have on my pi：

`pi@ezb-test:~/python-bluezero/examples $ dpkg -l | grep bluez
ii  bluez                                5.50-1.2~deb10u1+rpt2               armhf        Bluetooth tools and daemons
ii  bluez-firmware                       1.2-4+rpt3                          all          Firmware for Bluetooth devices`

ukBaz, 2020-12-17:
I don't have an iOS device so I can't test that piece. However, it has worked with nRF Connect on an Android device.  

Other people that have had iOS issues have seen it improve if they set the RPi to BLE only. You might want to have `sudo btmon` running in another terminal to see if that gives any useful information.

Star152, 2020-12-17:
Thank you for your answer. I have founded the reason

> HCI Event: Encryption Change (0x08) plen 4                                     #527 [hci0] 579.931876
        Status: Connection Timeout (0x08)
        Handle: 65
        Encryption: Disabled (0x00)
< HCI Command: Disconnect (0x01|0x0006) plen 3                                   #528 [hci0] 579.931967
        Handle: 65
        Reason: Authentication Failure (0x05)
> HCI Event: Disconnect Complete (0x05) plen 4                                   #529 [hci0] 579.932504
        Status: Success (0x00)
        Handle: 65
        Reason: Connection Timeout (0x08)
@ MGMT Event: Device Disconnected (0x000c) plen 8                            {0x0002} [hci0] 579.932543
        LE Address: 88:64:40:54:F6:EA (OUI 88-64-40)
        Reason: Connection timeout (0x01)
@ MGMT Event: Device Disconnected (0x000c) plen 8                            {0x0001} [hci0] 579.932543
        LE Address: 88:64:40:54:F6:EA (OUI 88-64-40)
        Reason: Connection timeout (0x01)
> HCI Event: Command Status (0x0f) plen 4                                        #530 [hci0] 579.934977
      Disconnect (0x01|0x0006) ncmd 1
        Status: Invalid HCI Command Parameters (0x12)

Star152, 2020-12-17:
Do you have a solution for the "Authentication Failure ", I look forward to your reply!

ukBaz, 2020-12-17:
My guess is that something wasn't right with the pairing.

Was the cpu_temperature example running when you did the pairing? Could be that the service isn't authorized if other things work

Star152, 2020-12-17:
The example works normally when pairing. Android devices do not need pairing when connecting. IOS devices will have a pairing pop-up window when they are connected. Disconnect when click pairing

ukBaz, 2020-12-17:
So are you saying it is fixed?  You just need to pair if you are on iOS?

Star152, 2020-12-17:
no， iOS need to pair but pair failed

ukBaz, 2020-12-17:
How were you doing the pairing? the cpu example does not handle the pairing.  I recommend that people do the pairing with `bluetoothctl` before running the example.

ukBaz, 2022-06-17:
This is related to #336 so will close this here

closed 2022-06-17 by ukBaz