ukBaz/python-bluezero #292: Sytem seting can find ble device,But use app cannot find device

kscorpio, 2020-12-17:
I use run demo code ble_uart.py , and use many mobiles (include android and iOS) to scan , cannot find the devices
but in mobile system setiing  - bluetooth can find this device

ukBaz, 2020-12-17:
You are going to have to give me more information if you want me to help. If just run the example again now and it has worked with [nRF Connect App ](https://www.nordicsemi.com/Software-and-tools/Development-Tools/nRF-Connect-for-mobile)on Android

kscorpio, 2020-12-18:
could  ble service uuid definition affect it？ and the PC as peripheral bluetooth version is 4.2

kscorpio, 2020-12-18:
Maybe I misunderstood   knowledge of Bluetooth， I use nRF connect in iphone ，nRF can find device(ble_uart) ， but  lightblue app can not find

ukBaz, 2020-12-18:
Changing the UUID in that way shouldn't be necessary. Sounds like LightBlue isn't living up to their claim of "With LightBlue®, you can scan, connect to and browse any nearby BLE device."

closed 2020-12-18 by ukBaz

SXKJ-luochao, 2024-03-15:
I met the same problem on my raspberry pi，can't found from the nRF Connect App (Android).

[image]

from the bluetoothctl, can see the info below:
[image]

device: raspberry pi 3B+
system: 
[image]

bluetoothctl版本：5.64

how can i solve this problem?  thanks

ukBaz, 2024-03-15:
@SXKJ-luochao, It would be better if you started a new issue with this rather than adding to a closed one. Especially as I don't see this as the same issue.

Can you look at the https://bluezero.readthedocs.io/en/stable/system_setup.html#notes-for-getting-debug-information section and open a new ticket if you would still like help.