ukBaz/python-bluezero #307: Additional install steps when installing via 'pip install'

WayneKeenan, 2021-01-08:
On a pi4 & pi-zero-w,  latest Raspbian, Bluetoothd 5.50, python3 (virtual env) these are the additional steps I'm performing to be able to use `bluezero` in a Central role when the module is installed via `pip  install`:

```
sudo apt install libglib2.0-dev libgirepository1.0-dev  libcairo2-dev
python3 -m venv .venv && . ./.venv/bin/activate
pip install bluezero dbus-python gi PyGObject
```

I was getting an Python2/3 incompatibility related error from the `gi` module, although I can't replicate it now.

But for prosperity/reference, it was: 

Error:
```
  File "/home/pi/projects/sonicbit/.venv/lib/python3.7/site-packages/gi/__init__.py", line 39
    print url
            ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print(url)?
```

The monkey-patch "fix" in the venv:
```
pip install 2to3
2to3 -w .venv/lib/python3.7/site-packages/gi/__init__.py
```

A PR may be forthcoming for documentation and `setup.py` / `requirements.txt`.

ukBaz, 2021-01-08:
If I am using a Python virtual environment with Bluezero I tend to go with the following:

```none
python3 -m venv --system-site-packages venv
. venv/bin/activate
pip install bluezero
```

There hasn't been a need to `apt install` anything for a while. Do we know why that has changed?

Also, I couldn't find `gi` on PyPI. Are you sure that is correct?

WayneKeenan, 2021-01-08:
Thanks, adding `--system-site-packages`  is what was missing.

`gi` came from `https://www.piwheels.org/simple/gi/gi-1.2-py3-none-any.whl`

WayneKeenan, 2021-01-08:
I should add that I didn't test on a clean OS, the apt-packages were already installed, so need to redo it.

WayneKeenan, 2021-01-08:
I've just tested on a fresh OS install using your steps and all is well, thanks.

I think if if nots there already (sorry if I missed it) then adding a note in the docs mentioning the use of `--system-site-packages` when creating a virtual is a plus.

Not sure what to say about the state of the `gi` module across Pypi and piwheels, slightly worrying in general... 
Although piwheels is unlikely to go away, its not the authoritative source of `gi` [status page](https://www.piwheels.org/project/gi/) but it sure seems gone from PyPi, any potential transient errors aside...

ukBaz, 2021-01-09:
Resolved with 06480a3430e55325a10d3854db509efbfc412ba4

closed 2021-01-09 by ukBaz

ukBaz, 2023-02-11:
I know this is a closed ticket but I had some issues around this today and I wanted to record the steps somewhere. This seems as good a place as anywhere....

Getting Bluezero to work in a Virtual Environment I did the installation of the two packages into the venv:
```none
$ pip install dbus-python
$ pip install vext.gi
```
However, it did not work and it took me a while to work out why. On the command line you can list which external vext packages that are installed with:
```none
$ vext -l
Running in virtualenv [disabled]
gi.vext
```
What I had failed to notice was the `disabled` status.
This can be checked on the command line with:
```none
$ vext -c gi.vext
import gi.repository: [failed]
```
And also in Python:
```none
$ python3 -c "from bluezero import adapter"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/machine/python-bluezero/bluezero/adapter.py", line 10, in <module>
    from bluezero import async_tools
  File "/home/machine/python-bluezero/bluezero/async_tools.py", line 7, in <module>
    from gi.repository import GLib
ModuleNotFoundError: No module named 'gi'
```
The package needs to be enabled with `-e`. e.g.:
```none
$ vext -e
$ vext -l
Running in virtualenv [enabled]
gi.vext
$ vext -c gi.vext
import gi.repository: [success]
```