ukBaz/python-bluezero #309: How do I scan for all services and their characteristics?

Yatekii, 2021-01-13:
Basically the title :) Docs are kinda lacking here.

Creating a Service object requires an id. All I get from listing services on a device is a UUID tho.

Any help is very much appreciated :)

Best,
Noah

ukBaz, 2021-01-14:
I am not confident that I have understood your question. What I have understood it to be is that you are creating a `central` role which has connected to a device. After connecting to that device you want to get a list of all the services available on that device. Is that correct?

The way to iterate through all of the information with BlueZ/D-Bus is with `get_managed_objects`. Specifying which interface you are interested in will target the information you want.

This is an example of getting all of the GATT services BlueZ knows about:

```python
more bluezero_find_services.py
from bluezero import dbus_tools
from bluezero import constants
mngd_objs = dbus_tools.get_managed_objects()
for path, obj_info in mngd_objs.items():
    srv = obj_info.get(constants.GATT_SERVICE_IFACE)
    if srv:
        print(srv.get('UUID') )
```
Example output:
```none
e95dd91d-251d-470a-a062-fa1922dfa9a8
e95d9882-251d-470a-a062-fa1922dfa9a8
e95d93af-251d-470a-a062-fa1922dfa9a8
0000180a-0000-1000-8000-00805f9b34fb
e97dd91d-251d-470a-a062-fa1922dfa9a8
```
If you wanted to get the characteristics and then walk back up the tree and get the service and device associated with it then you would do something like:
```python
for path, obj_info in mngd_objs.items():
    chrc = obj_info.get(constants.GATT_CHRC_IFACE)
    if chrc:
            chrc_uuid = chrc.get('UUID')
            srv_path = chrc.get('Service')
            srv = mngd_objs[srv_path].get(constants.GATT_SERVICE_IFACE)
            srv_uuid = srv.get('UUID')
            dvc_path = srv.get('Device')
            dvc = mngd_objs[dvc_path].get(constants.DEVICE_INTERFACE)
            dvc_addr = dvc.get('Address')
            print(f'Device [{dvc_addr}] has Service [{srv_uuid}] with Characteristic [{chrc_uuid}]')
```
Example output:
```none
Device [E5:10:5E:37:11:2D] has Service [e95dd91d-251d-470a-a062-fa1922dfa9a8] with Characteristic [e95d0d2d-251d-470a-a062-fa1922dfa9a8]
Device [E5:10:5E:37:11:2D] has Service [e95dd91d-251d-470a-a062-fa1922dfa9a8] with Characteristic [e95d93ee-251d-470a-a062-fa1922dfa9a8]
Device [E5:10:5E:37:11:2D] has Service [e95dd91d-251d-470a-a062-fa1922dfa9a8] with Characteristic [e95d7b77-251d-470a-a062-fa1922dfa9a8]
Device [E5:10:5E:37:11:2D] has Service [e95d9882-251d-470a-a062-fa1922dfa9a8] with Characteristic [e95dda91-251d-470a-a062-fa1922dfa9a8]
Device [E5:10:5E:37:11:2D] has Service [e95d9882-251d-470a-a062-fa1922dfa9a8] with Characteristic [e95dda90-251d-470a-a062-fa1922dfa9a8]
Device [E5:10:5E:37:11:2D] has Service [e95d93af-251d-470a-a062-fa1922dfa9a8] with Characteristic [e95db84c-251d-470a-a062-fa1922dfa9a8]
Device [E5:10:5E:37:11:2D] has Service [e95d93af-251d-470a-a062-fa1922dfa9a8] with Characteristic [e95d23c4-251d-470a-a062-fa1922dfa9a8]
Device [E5:10:5E:37:11:2D] has Service [e95d93af-251d-470a-a062-fa1922dfa9a8] with Characteristic [e95d5404-251d-470a-a062-fa1922dfa9a8]
Device [E5:10:5E:37:11:2D] has Service [e95d93af-251d-470a-a062-fa1922dfa9a8] with Characteristic [e95d9775-251d-470a-a062-fa1922dfa9a8]
Device [E5:10:5E:37:11:2D] has Service [0000180a-0000-1000-8000-00805f9b34fb] with Characteristic [00002a26-0000-1000-8000-00805f9b34fb]
Device [E5:10:5E:37:11:2D] has Service [0000180a-0000-1000-8000-00805f9b34fb] with Characteristic [00002a25-0000-1000-8000-00805f9b34fb]
Device [E5:10:5E:37:11:2D] has Service [0000180a-0000-1000-8000-00805f9b34fb] with Characteristic [00002a24-0000-1000-8000-00805f9b34fb]
Device [E5:10:5E:37:11:2D] has Service [e97dd91d-251d-470a-a062-fa1922dfa9a8] with Characteristic [e97d3b10-251d-470a-a062-fa1922dfa9a8]
Device [E5:10:5E:37:11:2D] has Service [e95d93b0-251d-470a-a062-fa1922dfa9a8] with Characteristic [e95d93b1-251d-470a-a062-fa1922dfa9a8]
Device [E5:10:5E:37:11:2D] has Service [00001801-0000-1000-8000-00805f9b34fb] with Characteristic [00002a05-0000-1000-8000-00805f9b34fb]
```
I would be interested to hear why you need something like this with Bluezero. The library has been designed to allow people to connect to a BLE device with the Central API documented at: https://bluezero.readthedocs.io/en/stable/level10.html#module-bluezero.central.
If you can explain the scenario then I can better understand if there is something missing from that API. My experience previously is that people know the device address, Service UUID, and Characteristic UUID on the device they are interested in interacting with. For an example of a central I guide people to the [microbit.py](https://github.com/ukBaz/python-bluezero/blob/master/bluezero/microbit.py#L90) file as an example.

Yatekii, 2021-01-15:
Thanks for your extensive and elaborate explanation!

Your second example is exactly what I wanted :)

Discovery is absolutely vital if you are debugging something which is about 99% of my usecases ;)
I am developing my own driver just like the microbit one as in this case I actually know the UUIDs :) Thanks a lot!

There is some other trouble I am experiencing but I am not sure this is really bluezero related as it could also be bluez.
One real problem is: When I start discovery it only ever lists new devices that broadcast for the first time. Even if I say that it should not filter duplicates. This is kinda always undesirable as you don't want to list all the devices in range and not just the ones that are seen for the first time. And if you list all the devices the Central knows that's bad too because then you get no indicator which ones are actually present. I tried looking at RSSI, but it sometimes fails to look RSSI up, even tho the device is in range (and my Phone scanner picks it up as well). So I feel like bluez is kinda unfinished here :)

Thanks again!

ukBaz, 2021-01-15:
OK, sounds like I've shown you how to access the information you want.

There wasn't anything in your reply that made me think it was obvious that there is something that should be added to Bluezero. If I have that wrong then let me know.

The handling of duplicates is not ideal with BlueZ. From reading discussions on the BlueZ user email list it seems like the problem is made worse because the problem involves the Linux kernel and the Linux user space. 

Bluezero offers easier access to the filter:

https://github.com/ukBaz/python-bluezero/blob/06480a3430e55325a10d3854db509efbfc412ba4/bluezero/adapter.py#L230-L237

But it still doesn't work great. For the beacon scanning Bluezero has implemented, it deletes the device once it has read the device data. This means there are never duplicates but you lose your pairing information with that strategy if it is a device.

I prefer doing the one-off scanning and pairing via `bluetoothctl` and then connect and handle the exception in the code if it can't connect to the device.

Anyway, sounds like we are done on this issue so I'll close it. Please re-open if it is not done.

closed 2021-01-15 by ukBaz