ukBaz/python-bluezero #313: No need to register a named application on the System bus

andysan, 2021-01-23:
The GATT library currently registers a named destination (ukBaz.bluezero) on the system bus. This is actually not needed and using an anonymous destination would remove the need to modify the system policy.

Simply registering the Application object using `dbus.service.Object.__init__(self, self.bus, self.path)` seems to do the trick and works for me without additional configuration.
`

ukBaz, 2021-01-23:
If the need to update the D-Bus policy could be removed that would definitely help getting people setup on the library so there is benefit if this does work.

This is something I've wrestled with in the past and there was some reason why we went with the approach we did. It was all a long time ago now and lost in the mists of time.

I don't have time to look at this now but I'll leave it open for anyone (or my future self) to take a look and do a PR.

ukBaz, 2021-01-23:
While I'm looking at this, it is the following lines that are being referred to:
https://github.com/ukBaz/python-bluezero/blob/06a566eb81c6111182c74181e084233ae4e316a8/bluezero/localGATT.py#L47-L49

With the proposal being to remove lines 47-48 and changing 49 to use `self.bus` instead of `self.bus_name`

ukBaz, 2021-01-23:
And the removal of the following section from the documentation
https://github.com/ukBaz/python-bluezero/blob/master/docs/system_setup.rst#change-dbus-permissions-for-bluezero

And the removal of the file from the examples:
https://github.com/ukBaz/python-bluezero/blob/master/examples/ukBaz.bluezero.conf

andysan, 2021-01-23:
Yes, that's exactly what I did when I played around with the GATT example. Let's just say I was to lazy to update the DBus configuration and went debugging instead. 😉 (I have some experience hacking BLE mesh using Bluez in Python, so I knew it was possible)

ukBaz, 2021-01-23:
Thanks for coming back with more details. Sounds like a low risk/high benefit change then. :-)

I have just done an overhaul of the Peripheral code so this would be a good thing to add before I do a release. I'll hopefully find time soon.

Cheers.

ukBaz, 2021-01-28:
Fixed with 7e4bf7459c98ad592f20a34bffe96bd599e1d618 and new release put on PyPI

Thanks @andysan for the suggestion

closed 2021-01-28 by ukBaz