ukBaz/python-bluezero #317: Not working in venv

ukBaz, 2021-02-01:
Doesn't seem to work without this. Is there another way to get it working?

On Raspberry Pi running `python script.py` from venv doesn't work. I'm running as `pi` user and I've added `pi` user to `bluetooth` group.

_Originally posted by @MaikuMori in https://github.com/ukBaz/python-bluezero/issues/315#issuecomment-770622944_

MaikuMori, 2021-02-01:
Thanks, for opening this. I didn't want to open an issue in case it was something simple.

ukBaz, 2021-02-01:
I think this might be a different issue. There is a dependency in Bluezero on using Python D-Bus which on the RPi is in the main system python install. I have typically solved this by using the option `--system-site-packages`. Although the down side of that is that you lose much of the benefit of having a venv.

You might want to have a look at pip installing https://pypi.org/project/dbus-python/ in your venv as that might get you around the issue. You might also need https://pypi.org/project/vext.gi/

MaikuMori, 2021-02-01:
I believe I have it already installed.

These are my dependencies:
```toml
[tool.poetry.dependencies]
python = "^3.9"
bluezero = "^0.5.0"
dbus-python = "^1.2.16"
PyGObject = "^3.38.0"

[tool.poetry.dev-dependencies]
pytest = "^5.2"
flake8 = "^3.8.4"
black = {version = "^20.8b1", allow-prereleases = true}
```

With a modified version of the old dbus config most stuff works. I have a peripheral which I can connect and interact with my phone. I'm currently having some trouble with central, but I don't think it's related. If I remove the dbus config nothing works.

ukBaz, 2021-02-01:
Here is the transcript of me creating a venv and installing Bluezero into it. I have used the `cpu_temperature.py` from the example in Bluezero. I have run without things being installed in the venv to show the error messages. As you will see at the end, the example does run and I can successfully get the notifications on my phone.

```bash
pi@SensePi:/tmp/test_bluezero $ ll /etc/dbus-1/system.d/uk*
ls: cannot access '/etc/dbus-1/system.d/uk*': No such file or directory
pi@SensePi:/tmp/test_bluezero $ python3 -m venv venv
pi@SensePi:/tmp/test_bluezero $ . venv/bin/activate
(venv) pi@SensePi:/tmp/test_bluezero $ pip install bluezero
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting bluezero
  Using cached https://files.pythonhosted.org/packages/4d/8e/3ccfa30096214c23c7b84843ba4f6b8f7ea476f9b57b57047c98a6ff6d57/bluezero-0.5.0-py2.py3-none-any.whl
Installing collected packages: bluezero
Successfully installed bluezero-0.5.0
(venv) pi@SensePi:/tmp/test_bluezero $ python cpu_temperature.py 
Traceback (most recent call last):
  File "cpu_temperature.py", line 7, in <module>
    from bluezero import async_tools
  File "/tmp/test_bluezero/venv/lib/python3.7/site-packages/bluezero/async_tools.py", line 5, in <module>
    import dbus
ModuleNotFoundError: No module named 'dbus'
(venv) pi@SensePi:/tmp/test_bluezero $ pip install dbus-python
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting dbus-python
  Using cached https://www.piwheels.org/simple/dbus-python/dbus_python-1.2.16-cp37-cp37m-linux_armv7l.whl
Installing collected packages: dbus-python
Successfully installed dbus-python-1.2.16
(venv) pi@SensePi:/tmp/test_bluezero $ python cpu_temperature.py 
Traceback (most recent call last):
  File "cpu_temperature.py", line 7, in <module>
    from bluezero import async_tools
  File "/tmp/test_bluezero/venv/lib/python3.7/site-packages/bluezero/async_tools.py", line 7, in <module>
    from gi.repository import GLib
ModuleNotFoundError: No module named 'gi'
(venv) pi@SensePi:/tmp/test_bluezero $ pip install vext.gi
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting vext.gi
  Using cached https://files.pythonhosted.org/packages/6f/51/c83d41bda3b2766c4dc792f328078fdc1e1fa5742b6880301bf56db63878/vext.gi-0.7.4.tar.gz
Collecting vext>=0.7.4 (from vext.gi)
  Using cached https://www.piwheels.org/simple/vext/vext-0.7.4-py3-none-any.whl
Collecting ruamel.yaml>=0.11.10 (from vext>=0.7.4->vext.gi)
  Using cached https://files.pythonhosted.org/packages/7e/39/186f14f3836ac5d2a6a042c8de69988770e8b9abb537610edc429e4914aa/ruamel.yaml-0.16.12-py2.py3-none-any.whl
Collecting ruamel.yaml.clib>=0.1.2; platform_python_implementation == "CPython" and python_version < "3.9" (from ruamel.yaml>=0.11.10->vext>=0.7.4->vext.gi)
  Using cached https://www.piwheels.org/simple/ruamel-yaml-clib/ruamel.yaml.clib-0.2.2-cp37-cp37m-linux_armv7l.whl
Building wheels for collected packages: vext.gi
  Running setup.py bdist_wheel for vext.gi ... error
  Complete output from command /tmp/test_bluezero/venv/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-qwj8st23/vext.gi/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-q34k1581 --python-tag cp37:
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help
  
  error: invalid command 'bdist_wheel'
  
  ----------------------------------------
  Failed building wheel for vext.gi
  Running setup.py clean for vext.gi
Failed to build vext.gi
Installing collected packages: ruamel.yaml.clib, ruamel.yaml, vext, vext.gi
  Running setup.py install for vext.gi ... done
Successfully installed ruamel.yaml-0.16.12 ruamel.yaml.clib-0.2.2 vext-0.7.4 vext.gi-0.7.4
(venv) pi@SensePi:/tmp/test_bluezero $ pip install vext.gi
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: vext.gi in ./venv/lib/python3.7/site-packages (0.7.4)
Requirement already satisfied: vext>=0.7.4 in ./venv/lib/python3.7/site-packages (from vext.gi) (0.7.4)
Requirement already satisfied: ruamel.yaml>=0.11.10 in ./venv/lib/python3.7/site-packages (from vext>=0.7.4->vext.gi) (0.16.12)
Requirement already satisfied: ruamel.yaml.clib>=0.1.2; platform_python_implementation == "CPython" and python_version < "3.9" in ./venv/lib/python3.7/site-packages (from ruamel.yaml>=0.11.10->vext>=0.7.4->vext.gi) (0.2.2)
(venv) pi@SensePi:/tmp/test_bluezero $ python cpu_temperature.py 
CPU temperature is 38.9°C
Advertisement registered
```

Are you able to run the same experiment on you system to see if it works?

WayneKeenan, 2021-02-01:
There’s also this: https://github.com/ukBaz/python-bluezero/issues/307

MaikuMori, 2021-02-01:
I'll try to reproduce using your steps. Probably tomorrow. It's getting very late here.

ukBaz, 2022-06-17:
This looks to be resolved so closing

closed 2022-06-17 by ukBaz