ukBaz/python-bluezero #324: Callback on Initial Connection

a-krawciw, 2021-02-15:
Hi I am trying to connect the nRF Connect app on my phone to a Raspberry Pi 4 using bluezero based on the UART example. The app shows it connecting and then it says connection closed by the pi. 

I have used the nRF app with other peripheral devices so I'm pretty sure my phone is working fine with BLE.

For figuring out the lifecycle it would be nice to have a separate callback when the device establishes a connection and when that connection closes. 

Alternatively is there some way to find the python logging to see if the pi thinks it is connected?

ukBaz, 2021-02-16:
To get something working quickly, you can use the `dongle` property which has `on_connect` and `on_disconnect`. So for example:
```python
ble_uart = peripheral.Peripheral(adapter_address, local_name='BLE UART')
ble_uart.dongle.on_connect = my_connect_callback
ble_uart.dongle.on_disconnect = my_disconnect_callback
```

What would be even better would be to do a PR that adds this to peripheral.py as a property:
```python
    @property
    def on_connect(self):
        """
        Callback for when a device connects to the peripheral. 
        Callback should take one parameter which is a Bluezero device object
        """
        return self.dongle.on_connect

    @on_connect.setter
    def on_connect(self, callback):
        self.dongle.on_connect = callback

    @property
    def on_disconnect(self):
        """
        Callback for when a device disconnects from the peripheral. 
        Callback takes no parameters.
        """
        return self.dongle.on_disconnect

    @on_disconnect.setter
    def on_disconnect(self, callback):
        self.dongle.on_disconnect = callback
```
Using this "facade" design pattern would then allow your code to read better and be easier to document in Bluezero:
```python
ble_uart = peripheral.Peripheral(adapter_address, local_name='BLE UART')
ble_uart.on_connect = my_connect_callback
ble_uart.on_disconnect = my_disconnect_callback
```

ukBaz, 2021-02-21:
Closed with #325

closed 2021-02-21 by ukBaz