ukBaz/python-bluezero #328: Characteristic Length Error

owenber123, 2021-03-31:
I have just resized all of my characteristics to take 1 - 16 bytes on my embedded peripheral bluetooth device and when I try to write a 16 byte value to one of the characteristics I get a bluezero error of "...threw an exception: Expected a length-1 bytes but found 16 bytes".  I have confirmed with bluetoothctl that writing a 16 byte value works just fine.  I have removed all bluetooth caching from my central device and can confirm that the other characteristic reads are working by receiving 16 byte values and validating them.  I am convinced that at no point does the central configure the characteristic lengths and its all done by the peripheral.  Am I missing something?  Thanks in advance.

ukBaz, 2021-03-31:
How have you resized your characteristics?

owenber123, 2021-03-31:
I am using the Nordic nrf52840.  When configuring characteristics I have set the max length to 16.

ukBaz, 2021-03-31:
I will need some more information to help you debug this.

When running your script have separate terminals open with the following running to get more debug information:

- bluetootctl
- journalctl -f -u bluetooth
- sudo busctl monitor org.bluez
- sudo btmon

Hopefully there will be something in there that narrows down where the issue is.

owenber123, 2021-04-01:
Here are the results:
Thanks for the help!

bluetootctl displays the characteristics uuid properly: 
[NEW] Characteristic (Handle 0xfb5b)
	/org/bluez/hci0/dev_DA_AE_22_8E_37_C6/service000e/char002b
	2a191b22-9dc6-4b0b-bb11-9eae903da5c2
	Vendor specific

jounalctl -f -u bluetooth:
bluetoothd[2297892]: src/device.c:load_gatt_db() No cache for DA_AE_22_8E_37_C6

sudo busctl monitor org.bluez:
                  DICT_ENTRY "oa{sa{sv}}" {
                          OBJECT_PATH "/org/bluez/hci0/dev_DA_AE_22_8E_37_C6/service000e/char002b";
                          ARRAY "{sa{sv}}" {
                                  DICT_ENTRY "sa{sv}" {
                                          STRING "org.freedesktop.DBus.Introspectable";
                                          ARRAY "{sv}" {
                                          };
                                  };
                                  DICT_ENTRY "sa{sv}" {
                                          STRING "org.bluez.GattCharacteristic1";
                                          ARRAY "{sv}" {
                                                  DICT_ENTRY "sv" {
                                                          STRING "UUID";
                                                          VARIANT "s" {
                                                                  STRING "2a191b22-9dc6-4b0b-bb11-9eae903da5c2";
                                                          };
                                                  };
                                                  DICT_ENTRY "sv" {
                                                          STRING "Service";
                                                          VARIANT "o" {
                                                                  OBJECT_PATH "/org/bluez/hci0/dev_DA_AE_22_8E_37_C6/service000e";
                                                          };
                                                  };
                                                  DICT_ENTRY "sv" {
                                                          STRING "Value";
                                                          VARIANT "ay" {
                                                                  ARRAY "y" {
                                                                  };
                                                          };
                                                  };
                                                  DICT_ENTRY "sv" {
                                                          STRING "Flags";
                                                          VARIANT "as" {
                                                                  ARRAY "s" {
                                                                          STRING "read";
                                                                          STRING "write";
                                                                  };
                                                          };
                                                  };
                                          };
                                  };
                                  DICT_ENTRY "sa{sv}" {
                                          STRING "org.freedesktop.DBus.Properties";
                                          ARRAY "{sv}" {
                                          };
                                  };
                          };
                  };

Here is something that also stands out from busctl monitor org.bluez
‣ Type=method_call  Endian=l  Flags=0  Version=1 Cookie=3663
  Sender=:1.2645  Destination=:1.2643  Path=/org/bluez/hci0/dev_DA_AE_22_8E_37_C6/service000e/char002b  Interface=org.freedesktop.DBus.Introspectable  Member=Introspect
  UniqueName=:1.2645
  MESSAGE "" {
  };

‣ Type=method_return  Endian=l  Flags=1  Version=1 Cookie=3008  ReplyCookie=3663
[... 75 lines cut ...]
< ACL Data TX: Han.. flags 0x00 dlen 10  #1643 [hci0] 119.821272
      LE L2CAP: Connection Parameter Update Response (0x13) ident 3 len 2
        Result: Connection Parameters accepted (0x0000)
< HCI Command:... (0x08|0x0013) plen 14  #1644 [hci0] 119.821278
        Handle: 16
        Min connection interval: 15.00 msec (0x000c)
        Max connection interval: 30.00 msec (0x0018)
        Connection latency: 0 (0x0000)
        Supervision timeout: 4000 msec (0x0190)
        Min connection length: 0.000 msec (0x0000)
        Max connection length: 0.000 msec (0x0000)
> HCI Event: Command St.. (0x0f) plen 4  #1645 [hci0] 119.822205
      LE Connection Update (0x08|0x0013) ncmd 2
        Status: Success (0x00)
> HCI Event: Number of... (0x13) plen 5  #1646 [hci0] 119.911211
        Num handles: 1
        Handle: 16
        Count: 1
> HCI Event: LE Meta E.. (0x3e) plen 10  #1647 [hci0] 120.316213
      LE Connection Update Complete (0x03)
        Status: Success (0x00)
        Handle: 16
        Connection interval: 30.00 msec (0x0018)
        Connection latency: 0 (0x0000)
        Supervision timeout: 4000 msec (0x0190)

owenber123, 2021-04-01:
I just needed to convert my bytes to a list of integers when writing in my setter, seems to be working fine now.  Thanks for the assistance.

closed 2021-04-01 by owenber123