ukBaz/python-bluezero #329: dbus dependency missing / not documented

PhilippSchindler, 2021-04-14:
Hi, 

I installed this library using pip in a virtual environment. Unfortunately, my first test `from bluezero import adapter` failed at line 4 for `adapter.py` with the `import dbus` statement. There are quite a few dbus packages available on Pypi. Could you please tell me which dbus package or other dependencies are required? Maybe you also want to added them to a requirements.txt?

All the best,
Philipp

ukBaz, 2021-04-14:
This library is primarily aimed at use on the Raspberry Pi where it is common not to use venv. There is some information on installing with venv's at:

https://github.com/ukBaz/python-bluezero/blob/main/docs/developer_docs.rst#developer-install

This has been [discussed previously ](https://github.com/ukBaz/python-bluezero/issues/111) but there is not a good solution when there is system dependencies.

If venv's are important to you, then there is another Bluetooth library (that doesn't depend on dbus-python, the reference implementation of the D-Bus protocol) and so is totally installed within the venv. That is available at: https://github.com/ukBaz/BLE_GATT

closed 2021-04-14 by ukBaz

PhilippSchindler, 2021-04-14:
Thank you very much, that explanation was very helpful. Sorry for not spotting this developer docs. I will have a look at BLE_GATT.