ukBaz/python-bluezero #330: CPU example bot showing the number

SmartTransp, 2021-04-25:
Hello, 
I have install and run the CPU Example 
I can see in the terminal he is showing the temp
`
 python3 BLE_Temperture.py
CPU temperature is 47.7�°C
Advertisement registered
`

but when I scan BLE with my phone I only see the name "CPU Monitor" and I don't get the value 
even if I connect to the device 

what did I miss?

Thanks,

ukBaz, 2021-04-25:
1. do I have to connect to the PI (BLE) in order to see the temperature?
Yes. This is a BLE peripheral and that [role](https://bluezero.readthedocs.io/en/stable/overview.html#bluetooth) requires you to connect to view the characteristic value
2. how can I see the MAC of the connected device?
There are a number of ways of doing this. To get a message when the device connects you could do the following in the `main` function:
```python
    cpu_monitor.dongle.on_connect = device_connected_cb
```
And define the callback that it calls. e.g.
```python
def device_connected_cb(dev_obj):
    print(f'Device connected: {dev_obj.address}')
```
3. where do I need to add this (eddystone_beacon.EddystoneURL('https://www.mysite.com/'))  line
The `eddystone_beacon` and `peripheral` modules have not been written in a way that makes them easy to combine in this style. There is the issue of the advertisement is stopped when a connection is made. There is also limited space for advertising so you are likely to need to create multiple advertisements. You would likely need to use the lower level APIs of `peripheral` and `advertisement` to achieve what you wanted.

ukBaz, 2021-06-01:
Closing because of lack of activity. Please reopen if you still have questions.

closed 2021-06-01 by ukBaz