ukBaz/python-bluezero #335: RPi 4 Buster Sending Rapid Pairing Requests to iOS Device

garretthagen21, 2021-06-11:
This is a problem I encountered and I wanted to make note of it incase anyone else ran into the same issue. I am using a Raspberry Pi 4 on Buster and tried this on multiple iOS devices using both the nRFConnect and LightBlue BLE connection apps. 

There is an issue with bluez that causes the Raspberry Pi to continually send pairing requests, even if the user accepts it on their device. 

For anyone else facing this issue, the resolution can be found here under thombles comment:  https://developer.apple.com/forums/thread/678925

Below is the step-by-step post from the forum incase something happens to the link in the future.

--------------------------------------------------------------------------------------------------------------------------
One cause of this is that Linux tries to read the battery state from the iPhone, which triggers the pairing request. You can configure bluez to prevent this. One of my colleagues at Ditto wrote some docs to fix it, which the forum won't let me link directly, but I will copy-paste the main instructions here.

To disable auto Battery reading

1. Open the bluetooth service file /usr/lib/systemd/system/bluetooth.service, or /etc/systemd/system/bluetooth.target.wants/bluetooth.service in a text editor. You may need sudo permission to write to this file.

2. Add -P battery to the end of the ExecStart line to disable the Battery feature in bluetoothd. Your ExecStart should look something like ExecStart=/usr/lib/bluetooth/bluetoothd -P battery now.
Save the file.

3. Run systemctl daemon-reload and systemctl restart bluetooth to apply the changes to the Bluetooth service

garretthagen21, 2021-06-11:
This issue is resolved. Wanted to add it to help anyone else also facing it.

ukBaz, 2021-06-12:
Thanks @garretthagen21 for noting this clearly. It was documented in #331 but this is cleaner. I'll open a issue to update the documentation for the example as that is the most timely place to present the information.

I'll also note here that `/lib/systemd/system/bluetooth.service` should be the correct file to edit. `etc/systemd/system/bluetooth.target.wants/bluetooth.service` should be a a symlink created by the system (usually as part of `systemctl enable`)

closed 2021-06-12 by ukBaz