ukBaz/python-bluezero #338: Is there anyway that I can modify Generic Access Profile?

denny61302, 2021-06-16:
I am trying to create a customer service with a notification called back.

Currently, I have to use my phone to enable the notification callback.

I also built another BLE device using a Cypress chip. In this chip, I can set up the Peripheral Preferred Connection Parameters [R] (0x2A04) in the Generic Access (0x1800) service.

The problem I faced is that the Cypress BLE will call the notification automatically once the connection is built, but the same service built on the RPI 3B+ is not. I have to enable the notification through a mobile app (NRFconnect)

I am not sure what casues this issue, so I record log files and saw the only difference is the setting of the Peripheral Preferred Connection Parameters [R] (0x2A04) in the Generic Access (0x1800) service.

The followings are the log files from NRFconnect

The correct one from the Cypress BLE chip

nRF Connect, 2021-06-15
Austofix ezyaim 2.0 (00:A0:50:00:00:01)
V	15:31:33.280	Connecting to 00:A0:50:00:00:01...
D	15:31:33.281	gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)
D	15:31:33.353	[Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
D	15:31:33.377	[Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	15:31:33.377	Connected to 00:A0:50:00:00:01
V	15:31:33.396	Discovering services...
D	15:31:33.396	gatt.discoverServices()
I	15:31:33.843	Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
D	15:31:34.141	[Callback] Services discovered with status: 0
I	15:31:34.141	Services discovered
V	15:31:34.156	Generic Access (0x1800)
- Device Name [R] (0x2A00)
- Appearance [R] (0x2A01)
**- Peripheral Preferred Connection Parameters [R] (0x2A04)**
Generic Attribute (0x1801)
- Service Changed [I] (0x2A05)
   Client Characteristic Configuration (0x2902)
Unknown Service (0000fff0-0000-1000-8000-00805f9b34fb)
- Unknown Characteristic [R W] (0000fff1-0000-1000-8000-00805f9b34fb)
   Unknown Descriptor (0ec2697a-62cb-48b4-952f-bcce97f2ee36)
- Unknown Characteristic [N R] (0000fff2-0000-1000-8000-00805f9b34fb)
   Client Characteristic Configuration (0x2902)
   Characteristic User Description (0x2901)
- Unknown Characteristic [W] (0000fff3-0000-1000-8000-00805f9b34fb)
   Unknown Descriptor (b506fac5-2f9b-4048-8910-46fb11029627)
- Unknown Characteristic [R] (0000fff4-0000-1000-8000-00805f9b34fb)
   Unknown Descriptor (f8319b11-b07f-45c9-8756-c7ceb388d0f2)
- Unknown Characteristic [R] (0000fff5-0000-1000-8000-00805f9b34fb)
   Unknown Descriptor (b1754db0-ce7d-489e-aea7-f30576d5c110)
Battery Service (0x180F)
- Battery Level [N R] (0x2A19)
   Characteristic Presentation Format (0x2904)
   Client Characteristic Configuration (0x2902)
D	15:31:34.156	gatt.setCharacteristicNotification(00002a05-0000-1000-8000-00805f9b34fb, true)
D	15:31:34.159	gatt.setCharacteristicNotification(0000fff2-0000-1000-8000-00805f9b34fb, true)
D	15:31:34.161	gatt.setCharacteristicNotification(00002a19-0000-1000-8000-00805f9b34fb, true)
I	15:31:34.204	Notification received from 0000fff2-0000-1000-8000-00805f9b34fb, value: (0x) FF-45-00-01
A	15:31:34.204	"(0x) FF-45-00-01" received
I	15:31:34.256	Notification received from 0000fff2-0000-1000-8000-00805f9b34fb, value: (0x) FF-46-00-01
A	15:31:34.256	"(0x) FF-46-00-01" received

The wrong one from the RPI 3B+

nRF Connect, 2021-06-15
Austofix ezyaim 2.0 (B8:27:EB:84:55:D3)
V	15:46:39.293	Connecting to B8:27:EB:84:55:D3...
D	15:46:39.293	gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)
D	15:46:39.966	[Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	15:46:39.966	Connected to B8:27:EB:84:55:D3
V	15:46:40.007	Discovering services...
D	15:46:40.007	gatt.discoverServices()
D	15:46:40.009	[Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
I	15:46:40.727	Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
D	15:46:40.947	[Callback] Services discovered with status: 0
I	15:46:40.947	Services discovered
V	15:46:40.966	Generic Access (0x1800)
- Device Name [R] (0x2A00)
- Appearance [R] (0x2A01)
Generic Attribute (0x1801)
- Service Changed [I] (0x2A05)
   Client Characteristic Configuration (0x2902)
Unknown Service (0000fff0-0000-1000-8000-00805f9b34fb)
- Unknown Characteristic [R W] (0000fff1-0000-1000-8000-00805f9b34fb)
   Unknown Descriptor (0ec2697a-62cb-48b4-952f-bcce97f2ee36)
- Unknown Characteristic [N R] (0000fff2-0000-1000-8000-00805f9b34fb)
   Client Characteristic Configuration (0x2902)
- Unknown Characteristic [W] (0000fff3-0000-1000-8000-00805f9b34fb)
   Unknown Descriptor (b506fac5-2f9b-4048-8910-46fb11029627)
- Unknown Characteristic [R] (0000fff4-0000-1000-8000-00805f9b34fb)
   Unknown Descriptor (f8319b11-b07f-45c9-8756-c7ceb388d0f2)
- Unknown Characteristic [R] (0000fff5-0000-1000-8000-00805f9b34fb)
   Unknown Descriptor (b1754db0-ce7d-489e-aea7-f30576d5c110)
Battery Service (0x180F)
- Battery Level [N R] (0x2A19)
   Client Characteristic Configuration (0x2902)
D	15:46:40.966	gatt.setCharacteristicNotification(00002a05-0000-1000-8000-00805f9b34fb, true)
D	15:46:40.968	gatt.setCharacteristicNotification(0000fff2-0000-1000-8000-00805f9b34fb, true)
D	15:46:40.969	gatt.setCharacteristicNotification(00002a19-0000-1000-8000-00805f9b34fb, true)
**I	15:46:41.039	Connection parameters updated (interval: 45.0ms, latency: 0, timeout: 5000ms)
V	15:46:49.228	Enabling notifications for 0000fff2-0000-1000-8000-00805f9b34fb**
D	15:46:49.228	gatt.setCharacteristicNotification(0000fff2-0000-1000-8000-00805f9b34fb, true)
D	15:46:49.231	gatt.writeDescriptor(00002902-0000-1000-8000-00805f9b34fb, value=0x0100)
I	15:46:49.321	Data written to descr. 00002902-0000-1000-8000-00805f9b34fb, value: (0x) 01-00
A	15:46:49.321	"Notifications enabled" sent
I	15:46:49.457	Notification received from 0000fff2-0000-1000-8000-00805f9b34fb, value: (0x) FF-35-00-00
A	15:46:49.457	"(0x) FF-35-00-00" received

Is there anyone faces the same issuse?

ukBaz, 2021-06-16:
This is an issue I've been looking at for a while in #321

I have asked the question on the BlueZ slack channel but not received a actionable answer. Your data is interesting because it is not something I was aware of previously.
Do you have a reference with more information about Peripheral Preferred Connection Parameters [R] (0x2A04)?

I've looked on bluetooth.com and found the value in the list of [16-bit UUIDs](https://btprodspecificationrefs.blob.core.windows.net/assigned-values/16-bit%20UUID%20Numbers%20Document.pdf) but I can't find mention of it in the [GATT Specification Supplement](https://www.bluetooth.org/DocMan/handlers/DownloadDoc.ashx?doc_id=514633) 

Looking in the BlueZ source code there seems to be some hardcoded numbers associated with it:
https://github.com/bluez/bluez/blob/7c3ca2a6b940d36c553fabe38066fabc66530dc9/unit/test-gatt.c#L1685-L1687

So in summary, I think the issue might be an underlying one with BlueZ but if you can share more information about the characteristic we might be able to work something out.

denny61302, 2021-06-16:
Thanks for the reply.

I did find the GAP and Peripheral Preferred Connection Parameters UUIDs are defined in the BlueZ source code.

https://github.com/bluez/bluez/blob/7c3ca2a6b940d36c553fabe38066fabc66530dc9/lib/uuid.h#L82

This snapshot is the Peripheral Preferred Connection Parameters setting of my Cypress chip.

[image]

I have double-checked my log file, showing that it contains 8 bytes of data.

Hope this will help.

ukBaz, 2021-06-16:
Thanks for sharing that. There is nothing in that screenshot that suggests it is controlling the enablement of a notification callback. Could getting Peripheral Preferred Connection Parameters to appear to be a separate issue getting a custom characteristics to start with notifications enabled by default?

denny61302, 2021-06-17:
Agreed. Just like you said there is no enablement of a notification callback in Peripheral Preferred Connection Parameters. 

BTW, my application is based on the cpu_temperature example. 

I also tried to set notification to True when I created the characteristic, but there is no effect so far.

Here is the snippet of my code. Did I miss something in here?

ble.add_characteristic(srv_id=1, chr_id=2, uuid='0000FFF2-0000-1000-8000-00805F9B34FB',
                                   value=[], **notifying=True**,
                                   flags=['read','notify'],
                                   read_callback=None,
                                   write_callback=None,
                                   notify_callback=notify_callback
                                   )

ukBaz, 2021-06-17:
> I also tried to set notification to True when I created the characteristic, but there is no effect so far.
> Here is the snippet of my code. Did I miss something in here?

Nope!  That is exactly the same problem as we are working on. I'm going to close this issue as it sounds like a duplication of #321 because this is about activating notifications by default. If I have misunderstood then let me know and I'll reopen.

closed 2021-06-17 by ukBaz