ukBaz/python-bluezero #340: Redundant and incorrect notification

dsjove, 2021-07-08:
Maybe I am using the library wrong. I am seeing an extraneous and incorrect value sent to the client. 

The client writes a value [3, 4].
write_callback parses it, transforms it to [75, 100] and calls characteristic.set_vaue
The client receives the [75, 100], but then it also receives the [3, 4] immediately after!

I am 100% certain I am never calling set_value with [3, 4] from the blue zero client.

This is a single characteristic. I am pretty certain two characteristics works around this but that complicates the code. flags are ['read', 'notify', 'write-without-response']. 

Any help appreciated.

ukBaz, 2021-07-08:
Having a characteristic `write` cause a `notify` event on the same characteristic wasn't a scenario that I had considered  when creating the library.

Indeed, even if you didn't have the `write_callback`, I'm questioning if a `notify` event is the correct thing to happen on a `write`. Looking around I've found the BBC micro:bit [Magnetometer Calibration ](https://lancaster-university.github.io/microbit-docs/resources/bluetooth/bluetooth_profile.html) characteristic that has `write` and `notify`. It seems to have that the notify on write and then a notify on result of the calibration being run. Although it would be good to confirm that from another example also

The documentation would also need to be updated to reflect that there is now the option
https://github.com/ukBaz/python-bluezero/blob/392a115ace80a6d7257b76708ec24b99ec9e2e5d/bluezero/peripheral.py#L79-L88

After thinking about this some more, I wonder if the better way of doing this is to put the `Set` before the `write_callback` and then the `write_callback` is responsible for doing the notification on property change. i.e.
```python
    def WriteValue(self, value, options):  # pylint: disable=invalid-name
        """
        DBus method for setting the characteristic value
        :return: value
        """
        self.Set(constants.GATT_CHRC_IFACE, 'Value', value)
        if self.write_callback:
            self.write_callback(dbus_tools.dbus_to_python(value),
                                dbus_tools.dbus_to_python(options))
```

This probably depends if a `write` on its own should cause a notification to be sent.

dsjove, 2021-07-08:
I started out with the UART implementation. But my use case is simpler. Two characteristics does work around the issue but complicates code for a simple use case.

Client says volume should be 11.
Device changes volume from 5 to 10.
Volume is now broadcasted out as 10.

The invalid 11 should never be broadcasted from the device. There are a hundred ways to solve this. My suggestion was not ideal but it was non breaking. :-)

When a client attempts to set a (possible invalid) value on a characteristic, the device should have an opportunity to drop it or transform it. 

Another non breaking way to fix this is to add another optional callback that returns none for drop or same/transformed value to be set.

BTW, thank you for the quick response.

dsjove, 2021-07-08:
https://bitbucket.org/drjove/thejoveexpress/src/master/pi/btProperty.py

This is my use case.

dsjove, 2021-07-14:
I was wondering if you had any more thoughts on this?

ukBaz, 2021-07-15:
I think I am in agreement that there is something that could probably be improved here.

The current design methodology was that whatever got sent to the characteristic value got sent to the write_callback and got stored which caused a notification to be sent.

My understanding of what you've posted above is that you want the write_callback to check that the received value is inside acceptable bounds. If the value is not inside those bounds then the value would be changed before it is stored (and notified).

The key issue with the current implementation is that if the write_callback wants to change the values then the notifications happen in the wrong order. My preference is to switch around the Set and write_callback statements. The problem with that for you is (to use your example) there would be an 11 and a 10 notification.
I've been trying to look around to see if there is some common practice on what is done. As of now I haven't been able to find an implementation that I'm happy to accept as a good reference to follow.

Would both 11 and 10 being broadcast, in that order, be acceptable to you?

Then it is *just* a question of someone implementing and testing it....

dsjove, 2021-07-20:
I have asked the question on how to become a contributor in another thread (undo pip install and pip install from something with my changes). I am new to this space.

I have an idea that would maintain the order, backward compatibility, and prove a way to to stop incoming garbage from being blindly broadcasted.

If you can give me instructions on how to do the reset of Bluezero on my Pi, I can get started.

ukBaz, 2021-08-30:
@dsjove I put the instructions in the other ticket. Did they help?

closed 2021-09-10 by ukBaz