ukBaz/python-bluezero #341: No clean shutdown method on Peripheral

dsjove, 2021-07-14:
Please a a stop method on Peripheral. The only way to shut down the self.mainloop.run() is via a KeyboardInterrupt exception.

RulerKun, 2021-07-15:
https://github.com/ukBaz/python-bluezero/issues/332#issuecomment-846437271

ukBaz, 2021-07-15:
I would suggest that the ctrl-c does close the peripheral down cleanly. If you mean that you want to do it programmatically, then @RulerKun has linked to information on how this can be done.

As a reminder, the goal of this library was to help users get started with Bluetooth on Linux (Raspberry Pi). In the docs it talks about the different [levels of API](https://bluezero.readthedocs.io/en/latest/overview.html#bluezero-api-complexity) that people can use to access finer control. People have the option to use the BlueZ API's directly to get the control they require.

With the Peripheral I had the scenario in mind that people would be creating embedded devices. I envisioned that people would have a headless device that would launch the Peripheral at boot and run until the machine was closed down.
I am not against adding some command that shuts the peripheral down. It *just* needs to be implemented, tested, and supported.

dsjove, 2021-07-20:
I really like the Peripheral class. :-) It does nearly everything I have needed so far. I think I can get to the clean shutdown via the adaptor.

I am building an over-engineered model train. It can be issued a shutdown command remotely for the stop everything. I am doing a kill-9 as last step from the python code.

dsjove, 2021-07-20:
I would very much like to create pull requests for these changes. I can make them. But I am a bit of a python noob and did not know how to undo the pip install of BlueZero, and fork it onto the raspberry device in an "installed" way.

ukBaz, 2021-07-21:
Part of the challenge here for me is that "clean shutdown" and "stop everything" are very imprecise terms. More helpful would be terms like "stop advertisement" or "make peripheral non-connectable" as these describe the functionality that the peripheral code controls.

`kill -9` is more of an emergency ejector seat rather than a nice controlled exit. I would be wary of this being a good solution.

As to how to create a pull-request, there should be plenty of information around on how to do this. On Bluezero we are using a [fork and pull](https://docs.github.com/en/github/collaborating-with-pull-requests/getting-started/about-collaborative-development-models) model.
This is common workflow for open source projects on GitHub so there are plenty of [tutorials](https://docs.github.com/en/get-started/quickstart/fork-a-repo) around. 
The challenge can be that different tutorials can do it slightly differently because they use alternative tools or slight variations on the flow. The good news is that you can't mess up the Bluezero main repository so you can jump in and experiment.

`pip uninstall` should be what you need to remove the PyPI Bluezero version.

ukBaz, 2021-08-30:
@dsjove Should I close the ticket or is there more to come on this?

closed 2021-09-10 by ukBaz