ukBaz/python-bluezero #343: Colouring logs

stsdc, 2021-08-15:
In private projects I'm using something like this.

```python
import logging
import coloredlogs
import os

os.environ['COLOREDLOGS_LOG_FORMAT'] ='%(asctime)s.%(msecs)03d %(name)s - %(message)s'
os.environ['COLOREDLOGS_DATE_FORMAT'] ='%H:%M:%S'

coloredlogs.install(level='DEBUG')

def logname(name='awesome module'):
    return logging.getLogger(name)
```

ukBaz, 2021-08-20:
@stsdc I am not sure what action you want me to take as a result of this information? Can you give me a little more background so I can know how to move forward with this.

Thanks.

stsdc, 2021-08-20:
Please consider that as hint to coloring log levels. When I tried to use `coloredlogs`, there was one line that was colored properly and also it's duplicate that was in a default color. I ended up disabling coloring because of that :/
Would be cool to see color logs, since they are more readable. Thanks :relieved: 

https://github.com/ukBaz/python-bluezero/blob/392a115ace80a6d7257b76708ec24b99ec9e2e5d/bluezero/tools.py#L170-L199

ukBaz, 2021-08-20:
I am reluctant to add the dependency so that everyone using Bluezero has to install `coloredlogs`. Bluezero is a library rather than an end application so I prefer it to have as few dependencies as possible. I know `coloredlogs` is only one dependency but then that has other dependencies... and so it goes on  :-(

However, that shouldn't stop you being about to use `coloredlogs`. The reason the logging doesn't work as expected is often that the wrong name has been passed to `getLogger()`. Each Bluezero file has its own logger. E.g. `localGATT`.

Let me know if you have an example that doesn't work and I can review.

stsdc, 2021-08-20:
> I am reluctant to add the dependency so that everyone using Bluezero has to install coloredlogs. Bluezero is a library rather than an end application so I prefer it to have as few dependencies as [...]

^ this totally understandable

Basically added these lines to https://github.com/synergia/synermycha-ble-simulator/blob/bluezero/main.py
```python
import coloredlogs
coloredlogs.install(level='DEBUG')
```
[image]

ukBaz, 2021-08-20:
The problem with your example above is you don't seem to have set the output for the `bluezero.localGATT` to use `coloredlogs`.  I think you need:
```python
    logger = logging.getLogger('bluezero.localGATT')
    coloredlogs.install(level='DEBUG', logger=logger)
```
I have done an experiment with the eddystone_scanner.py example in Bluezero. I've added some logging to the example and output from the `observer` 
[image]

Which gives the following output
[image]

This seems to be working as expected. Let me know if this works for you also and I'll close the issue.

stsdc, 2021-08-21:
You're 100% right. Thank You for Your help and time!

closed 2021-08-21 by stsdc