ukBaz/python-bluezero #352: Creating peripheral: Failed to create entry in database

cryptax, 2021-11-08:
Hello,
I mimicked code 'ble_uart.py` to create my own peripheral. But your code works fine, while mine complains "Failed to create entry in database" and "Does not exist".
**I changed the values for service and characteristic UUIDs. Should they be registered somehow?**

```
2021-11-08 10:10:16,864 - bluezero.GATT - WARNING - Failed to register application: org.bluez.Error.Failed: Failed to create entry in database
Failed to register advertisement: org.bluez.Error.Failed: Failed to parse advertisement.
^CTraceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/bluezero-0.6.0-py3.8.egg/bluezero/peripheral.py", line 145, in publish
  File "/usr/local/lib/python3.8/dist-packages/bluezero-0.6.0-py3.8.egg/bluezero/async_tools.py", line 46, in run
  File "/usr/lib/python3/dist-packages/gi/overrides/GLib.py", line 497, in run
    super(MainLoop, self).run()
  File "/usr/lib/python3.8/contextlib.py", line 120, in __exit__
    next(self.gen)
  File "/usr/lib/python3/dist-packages/gi/_ossighelper.py", line 251, in register_sigint_fallback
    signal.default_int_handler(signal.SIGINT, None)
KeyboardInterrupt

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "fakerope.py", line 63, in <module>
    main(list(adapter.Adapter.available())[0].address)
  File "fakerope.py", line 59, in main
    ble_rope.publish()
  File "/usr/local/lib/python3.8/dist-packages/bluezero-0.6.0-py3.8.egg/bluezero/peripheral.py", line 148, in publish
  File "/usr/local/lib/python3.8/dist-packages/bluezero-0.6.0-py3.8.egg/bluezero/advertisement.py", line 325, in unregister_advertisement
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 141, in __call__
    return self._connection.call_blocking(self._named_service,
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 652, in call_blocking
    reply_message = self.send_message_with_reply_and_block(
dbus.exceptions.DBusException: org.bluez.Error.DoesNotExist: Does Not Exist
```

ukBaz, 2021-11-08:
It is difficult to tell from the information you have shared exactly what the problem is likely to be. It is complaining that it hasn't been able to parse the advertisement data which is made up of service UUIDs, (and if set) local name, and appearance.

If I was to speculate then it is likely to be those values aren't formatted correctly or you have too much information for the advertisement.

Having `sudo busctl monitor org.bluez` and `sudo btmon` in other terminals may give you more clues as to where the error is.

ukBaz, 2022-06-17:
This has been sat here a while without an update so I'm closing. Please re-open with more information on how to reproduce if it is still an issue

closed 2022-06-17 by ukBaz