ukBaz/python-bluezero #353: resolve_gatt not implemented for bluezero.GATT.Descriptor

JanuszSamborski, 2021-11-09:
There is call for `self.resolve_gatt()` inside `bluezero.GATT.Descriptor`'s constructor, however it's not implemented.

JanuszSamborski, 2021-11-09:
Actually, on the same note, `resolve_gatt()` is not called inside `bluezero.GATT.Characteristic`'s constructor. Behavior should be probably unified.

JanuszSamborski, 2021-11-09:
I also found that (at least for my device) the first call to `resolve_gatt()` can fail, and second one succeeds, so it might be viable to call get_methods at least twice with some backoff, but this would need further testing.

ukBaz, 2021-11-09:
Hmm? That does look like it is all a bit of a mess.

I suspect that the resolve_gatt in the Service and Descriptor init's are never called because they are protected by that `if` statement. The services is never going to be resolved when you create the Service or Descriptor class instance.
 
https://github.com/ukBaz/python-bluezero/blob/6f46b6cb7102831c1427f12e65b727fc6b54774a/bluezero/GATT.py#L306-L307

I suspect the way it is done in the `Central` is the correct way:
https://github.com/ukBaz/python-bluezero/blob/e6b4e96342de6c66571a6660d711c018f8c6b470/bluezero/central.py#L92-L94

There being no `resolve_gatt` for the descriptor is an oversight. Although there are only about 16 descriptors assigned by the Bluetooth SIG with fairly niche behavior so I would curious to know which one you are using.