ukBaz/python-bluezero #355: Cannot advertise multiple primary services

ArchTonIO, 2021-11-29:
So, I'm working on a Raspberry Pi 4, I'm building a simple BLE device to get some led turned on and read some buttons value from an Android app. I started from the cpu_temperature.py example and modified it as follows: I have two services: one is the led service(write), the other is the button service(read), I setted both services to primary=True to advertise them.
Now, if I have only one service setted to primary=True it works just fine, but if I set both at True i get the error: "Failed to register advertisement: org.bluez.Error.Failed: Failed to parse advertisement.".
What causes the problem? How can I solve it?

ukBaz, 2021-11-29:
This could be a duplicate of #354 

I suspect the issue is that advertising two services with custom (128-bit) UUIDs is filling the space for the advert. 
You could have the LED and Button as two characteristics on one service.

ArchTonIO, 2021-11-29:
Can I use the short UUID to solve this ?

ukBaz, 2021-11-29:
The challenge with that is that the 16-bit UUIDs are [reserved ](https://www.novelbits.io/uuid-for-custom-services-and-characteristics/) by the Bluetooth SIG

ArchTonIO, 2021-11-29:
I found something online and yes... they are reserved, so I will simply use one service with more characteristics, thanks you for explaining me the issue