ukBaz/python-bluezero #356: Error when disconnecting from BLE Service

skast96, 2021-12-03:
I get this error below when I disconnect my client device from the BLE CPU_temp example. Is this an error that is happening on the library's side or the core bluetooth modules? 

```
ERROR:dbus.connection:Exception in handler for D-Bus signal:
Traceback (most recent call last):
  File "/home/arch/.work/gateway/venv/lib/python3.9/site-packages/dbus/connection.py", line 232, in maybe_handle_message
    self._handler(*args, **kwargs)
  File "/home/arch/.work/gateway/venv/lib/python3.9/site-packages/bluezero/adapter.py", line 276, in _properties_changed
    adapter_addr = dbus_tools.get_adapter_address_from_dbus_path(path)
  File "/home/arch/.work/gateway/venv/lib/python3.9/site-packages/bluezero/dbus_tools.py", line 122, in get_adapter_address_from_dbus_path
    mngd_objs = get_managed_objects()
  File "/home/arch/.work/gateway/venv/lib/python3.9/site-packages/bluezero/dbus_tools.py", line 93, in get_managed_objects
    return manager.GetManagedObjects()
  File "/home/arch/.work/gateway/venv/lib/python3.9/site-packages/dbus/proxies.py", line 72, in __call__
    return self._proxy_method(*args, **keywords)
  File "/home/arch/.work/gateway/venv/lib/python3.9/site-packages/dbus/proxies.py", line 141, in __call__
    return self._connection.call_blocking(self._named_service,
  File "/home/arch/.work/gateway/venv/lib/python3.9/site-packages/dbus/connection.py", line 652, in call_blocking
    reply_message = self.send_message_with_reply_and_block(
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Message recipient disconnected from message bus without replying
ERROR:dbus.connection:Exception in handler for D-Bus signal:
Traceback (most recent call last):
  File "/home/arch/.work/gateway/venv/lib/python3.9/site-packages/dbus/bus.py", line 177, in activate_name_owner
    return self.get_name_owner(bus_name)
  File "/home/arch/.work/gateway/venv/lib/python3.9/site-packages/dbus/bus.py", line 361, in get_name_owner
    return self.call_blocking(BUS_DAEMON_NAME, BUS_DAEMON_PATH,
  File "/home/arch/.work/gateway/venv/lib/python3.9/site-packages/dbus/connection.py", line 652, in call_blocking
    reply_message = self.send_message_with_reply_and_block(
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NameHasNoOwner: Could not get owner of name 'org.bluez': no such name

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/arch/.work/gateway/venv/lib/python3.9/site-packages/dbus/connection.py", line 232, in maybe_handle_message
    self._handler(*args, **kwargs)
  File "/home/arch/.work/gateway/venv/lib/python3.9/site-packages/bluezero/adapter.py", line 276, in _properties_changed
    adapter_addr = dbus_tools.get_adapter_address_from_dbus_path(path)
  File "/home/arch/.work/gateway/venv/lib/python3.9/site-packages/bluezero/dbus_tools.py", line 122, in get_adapter_address_from_dbus_path
    mngd_objs = get_managed_objects()
  File "/home/arch/.work/gateway/venv/lib/python3.9/site-packages/bluezero/dbus_tools.py", line 90, in get_managed_objects
    manager = dbus.Interface(bus.get_object(
  File "/home/arch/.work/gateway/venv/lib/python3.9/site-packages/dbus/bus.py", line 241, in get_object
    return self.ProxyObjectClass(self, bus_name, object_path,
  File "/home/arch/.work/gateway/venv/lib/python3.9/site-packages/dbus/proxies.py", line 250, in __init__
    self._named_service = conn.activate_name_owner(bus_name)
  File "/home/arch/.work/gateway/venv/lib/python3.9/site-packages/dbus/bus.py", line 182, in activate_name_owner
    self.start_service_by_name(bus_name)
  File "/home/arch/.work/gateway/venv/lib/python3.9/site-packages/dbus/bus.py", line 277, in start_service_by_name
    return (True, self.call_blocking(BUS_DAEMON_NAME, BUS_DAEMON_PATH,
  File "/home/arch/.work/gateway/venv/lib/python3.9/site-packages/dbus/connection.py", line 652, in call_blocking
    reply_message = self.send_message_with_reply_and_block(
dbus.exceptions.DBusException: org.freedesktop.systemd1.NoSuchUnit: Unit dbus-org.bluez.service not found.
ERROR:dbus.connection:Exception in handler for D-Bus signal:
Traceback (most recent call last):
  File "/home/arch/.work/gateway/venv/lib/python3.9/site-packages/dbus/bus.py", line 177, in activate_name_owner
    return self.get_name_owner(bus_name)
  File "/home/arch/.work/gateway/venv/lib/python3.9/site-packages/dbus/bus.py", line 361, in get_name_owner
    return self.call_blocking(BUS_DAEMON_NAME, BUS_DAEMON_PATH,
  File "/home/arch/.work/gateway/venv/lib/python3.9/site-packages/dbus/connection.py", line 652, in call_blocking
    reply_message = self.send_message_with_reply_and_block(
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NameHasNoOwner: Could not get owner of name 'org.bluez': no such name

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/arch/.work/gateway/venv/lib/python3.9/site-packages/dbus/connection.py", line 232, in maybe_handle_message
    self._handler(*args, **kwargs)
  File "/home/arch/.work/gateway/venv/lib/python3.9/site-packages/bluezero/adapter.py", line 276, in _properties_changed
    adapter_addr = dbus_tools.get_adapter_address_from_dbus_path(path)
  File "/home/arch/.work/gateway/venv/lib/python3.9/site-packages/bluezero/dbus_tools.py", line 122, in get_adapter_address_from_dbus_path
    mngd_objs = get_managed_objects()
  File "/home/arch/.work/gateway/venv/lib/python3.9/site-packages/bluezero/dbus_tools.py", line 90, in get_managed_objects
    manager = dbus.Interface(bus.get_object(
  File "/home/arch/.work/gateway/venv/lib/python3.9/site-packages/dbus/bus.py", line 241, in get_object
    return self.ProxyObjectClass(self, bus_name, object_path,
  File "/home/arch/.work/gateway/venv/lib/python3.9/site-packages/dbus/proxies.py", line 250, in __init__
    self._named_service = conn.activate_name_owner(bus_name)
  File "/home/arch/.work/gateway/venv/lib/python3.9/site-packages/dbus/bus.py", line 182, in activate_name_owner
    self.start_service_by_name(bus_name)
  File "/home/arch/.work/gateway/venv/lib/python3.9/site-packages/dbus/bus.py", line 277, in start_service_by_name
    return (True, self.call_blocking(BUS_DAEMON_NAME, BUS_DAEMON_PATH,
  File "/home/arch/.work/gateway/venv/lib/python3.9/site-packages/dbus/connection.py", line 652, in call_blocking
    reply_message = self.send_message_with_reply_and_block(
dbus.exceptions.DBusException: org.freedesktop.systemd1.NoSuchUnit: Unit dbus-org.bluez.service not found.
```

ukBaz, 2021-12-03:
I can not reproduce your error.
I have installed Bluezero v0.6.0  on to a Raspberry Pi and connect using `nRF Connect` on an Android phone. When I press disconnect inside `nRF Connect` I see no error messages on the Raspberry Pi.
Is my test the same as yours?

skast96, 2021-12-07:
Is this library only working with a raspberry pi? I tested it with my laptop, which runs Arch Linux.

ukBaz, 2021-12-07:
The library is tested on "Linux". It has the dependency of BlueZ and D-Bus being available which should be most Linux systems.

Generally speaking, if you can run `bluetoothctl` on your system then I would expect Bluezero to work.

The hardware I have available is a few Raspberry Pi's and an old Ubuntu laptop so that is what I test with.

ukBaz, 2022-06-17:
This has been sat here a while without an update so closing

closed 2022-06-17 by ukBaz