ukBaz/python-bluezero #358: on_connect being called incorrectly

kdhaka, 2021-12-31:
https://github.com/ukBaz/python-bluezero/blob/main/bluezero/adapter.py#L322
AND
https://github.com/ukBaz/python-bluezero/blob/main/bluezero/adapter.py#L281
handles `on_connect()` differently. which cause this exception to be raised.

Inside `_interfaces_added()` call the function according to number of arguments

Note:same applies for `on_disconnect()`

```
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/dbus/connection.py", line 232, in maybe_handle_message
    self._handler(*args, **kwargs)
  File "/usr/lib/python3.10/site-packages/bluezero/adapter.py", line 322, in _interfaces_added
    self.on_connect(new_dev)
TypeError: on_connect() missing 1 required positional argument: 'device_address'
```

ukBaz, 2021-12-31:
The number of arguments for the `on_connect` callback has changed as the library has evolved. The currently supported methodology is to take one argument which is the `Device` value. This is demonstrated in the example at: https://github.com/ukBaz/python-bluezero/blob/392a115ace80a6d7257b76708ec24b99ec9e2e5d/examples/ble_uart.py#L18

Your second link is to `if-elif`  logic that was put in to stop it being a breaking change at the time. But anything new put in would have used the "one parameter" methodology.

I have been looking at this functionality over the last couple of days as there are a few issues with the functionality and the documentation. This is being highlighted as I try to move Bluezero from python-dbus to [Gio.DBusProxy](https://pygobject.readthedocs.io/en/latest/). This change is proving slow going as I struggle to disentangle what is there, understand the new library, and cause as little disturbance to existing users as possible.

As a result I don't see me changing anything in the immediate short term. If you use a single `Device` parameter in your callback, does that cause you a problem?

kdhaka, 2022-01-03:
When I use a single Device parameter in callback, it do not cause any problem. Thanks.

ukBaz, 2022-01-03:
Thanks for letting me know. Sounds like you are off and working again so I'll close this. If I have misunderstood then please re-open.

closed 2022-01-03 by ukBaz