ukBaz/python-bluezero #359: Characterstic.write_value sends flags as array instead of dictionary

FormidableArne, 2022-01-31:
Characteristic.write_value sends the flags as a dictionary. but the [gatt-api](https://github.com/bluez/bluez/blob/master/doc/gatt-api.txt) demands a dictionary.

```
ERROR:dbus.connection:Unable to set arguments (b'\x00\x00\x00\x01\x00\x01\x00\x00\xa0', dbus.Array(['type'], signature=None)) according to signature 'aya{sv}': <class 'TypeError'>: list indices must be integers or slices, not str
Traceback (most recent call last):
  File "dfu.py", line 244, in <module>
    asyncio.run(main(deviceAddress, firmwareFilePath))
  File "/usr/lib/python3.7/asyncio/runners.py", line 43, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.7/asyncio/base_events.py", line 584, in run_until_complete
    return future.result()
  File "dfu.py", line 216, in main
    images = await listImages(c)
  File "dfu.py", line 129, in listImages
    smpCharacteristic.write_value(payload, flags={ 'type': 'command' }) # type 'command' is write without response.
  File "/usr/local/lib/python3.7/dist-packages/bluezero/GATT.py", line 211, in write_value
    self.characteristic_methods.WriteValue(value, dbus.Array(flags))
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 641, in call_blocking
    message.append(signature=signature, *args)
```

ukBaz, 2022-01-31:
Yep, that looks like you've found something that hasn't been implemented correctly.
Do you want to give me some context around your usage of Bluezero so I can get a sense of how to prioritize this?

I'm in the process of moving Bluezero away from the `python-dbus` library to `PyGObject` library for D-Bus so how much of effort is focused on that at the moment.

I noticed that you using `asyncio` as your event loop.  I would have thought [Bleak](https://bleak.readthedocs.io/en/latest/index.html) would have been a better library for you?

FormidableArne, 2022-01-31:
I'm using Bluezero in a docker container on raspberry pi to manage a handful of IoT devices. 
I'm using [asyncio-glib](https://github.com/jhenstridge/asyncio-glib) to make asyncio work, but yes I will probably move to Bleak in the near future.

ukBaz, 2022-01-31:
> I will probably move to Bleak in the near future.

OK, thanks for the update. I'll leave this issue open so it gets fixed but I won't rush a fix out. It has been like that for 6 years so I'll assume most people don't need it.

If it is an issue for anyone else then please comment so the issue gets bumped up the priority