ukBaz/python-bluezero #360: Change scan interval and duration

ebuss81, 2022-02-08:
Hey,
thank you for the great implementation. At the moment I am working with STM32WB MCUs, which are used in a test scenario as broadcaster, a raspberrypi as observer. I can read the custom advertisement packets with included sensor values but I would like to do this repetitively in a high frequency. Can this be done with bluezero? Any hints?

Best regards
Eddy

ukBaz, 2022-02-08:
What have you done so far?
Did you use the https://github.com/ukBaz/python-bluezero/blob/main/bluezero/observer.py example?

What frequency are you looking for?
Have you looked at the output from `sudo btmon` to see the frequency of advertisements in there?
Is your D-Bus/Python implementation slower than that in `btmon`?

ebuss81, 2022-02-08:
Have added or modified the following 3 functions in observer.py
`    

    def on_device_found(cls, bz_device_obj):
        rssi = bz_device_obj.RSSI
        service_data = bz_device_obj.service_data
        
        manufacturer_data = bz_device_obj.manufacturer_data

        if bz_device_obj.name and bz_device_obj.name == 'B1':
            cls.process_stm(bz_device_obj)

        cls.clean_beacons()

    @classmethod
    def process_stm(cls, bz_device_obj):
        manufacturer_data = bz_device_obj.manufacturer_data
        test = 0
        test = int.from_bytes(bytearray(manufacturer_data[48][0:1]), 'big', signed=False)
        if cls.on_stm:
            cls.on_stm(test)
            cls.stop_scan()

    def scan_stm(on_data=None):
        try:
            Scanner.start_beacon_scan(on_stm = on_data)
        except KeyboardInterrupt:
            Scanner.stop_scan()

`
and call them in a second python file
`
      import logging
      from MyObserver import *
       
      def print_data(data):
          print(data)
  
      
      
      def main():
          while(True):
                  
              scan_stm(on_data=print_data)
      
      
      if __name__ == '__main__':
          
          logger.setLevel(logging.INFO)
          main()
`

Almost the same what you have done. 
To print updated values I just start and stop the discovery procedure recurrently. Currently not sure how to get the current and updated values after receiving a new package. Only when starting the code the value is displayed once. 
I suspect that later it could be frequencies between 1Hz- 100Hz. 
When compared with btmon, no great delay can be seen. However, the packets are sent with a frequency of about 10hz from the MCU.

ukBaz, 2022-02-08:
BlueZ does some filtering in the kernel and in the user space. The deleting of discovered device in the above code is to get around some of that filtering. I am not aware of much you can do in BlueZ to get more messages than is appearing in `btmon`.

There is a slack channel to ask the BlueZ developers if you want to ask for more detail.
http://www.bluez.org/contact/

ebuss81, 2022-02-09:
many thanks!

ukBaz, 2022-06-17:
This hasn't had an update for a while so I'm closing.

closed 2022-06-17 by ukBaz