ukBaz/python-bluezero #361: Outdated docs about Peripheral config

vovagorodok, 2022-02-23:
Should be removed following notes? :
at: [Peripheral](https://github.com/ukBaz/python-bluezero#gatt-server-peripheral-role)
```
You will need to have modified the dbus configuration file to open the permissions for 'ukBaz.bluezero'. This is covered in the System Setup section of the documentation
```
at: https://github.com/ukBaz/python-bluezero/blob/main/docs/install_bluez.rst#creating-a-bluezero-peripheral
```
A peripheral application will be registered on the DBus using the bus name of ukBaz.bluezero. An example dbus configuration file is provided and will need to be copied to the correct location:
```

vovagorodok, 2022-02-23:
With new method examples should always runs with sudo unfortunately

ukBaz, 2022-02-23:
No, you should not need to use `sudo`.  As the `ukBaz.bluezero` name is not being claimed then it can be done without `sudo`

vovagorodok, 2022-02-23:
In case without sudo I have AccessDenied:
```
pi@home:~/python-bluezero/examples $ python3 ble_uart.py 
ERROR:dbus.proxies:Introspect error on :1.5:/: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.AccessDenied: Rejected send message, 2 matched rules; type="method_call", sender=":1.19" (uid=1000 pid=618 comm="python3 ble_uart.py ") interface="org.freedesktop.DBus.Introspectable" member="Introspect" error name="(unset)" requested_reply="0" destination=":1.5" (uid=0 pid=424 comm="/usr/lib/bluetooth/bluetoothd ")
Traceback (most recent call last):
  File "ble_uart.py", line 69, in <module>
    main(list(adapter.Adapter.available())[0].address)
  File "/home/pi/.local/lib/python3.7/site-packages/bluezero/adapter.py", line 49, in available
    mng_objs = dbus_tools.get_managed_objects()
  File "/home/pi/.local/lib/python3.7/site-packages/bluezero/dbus_tools.py", line 93, in get_managed_objects
    return manager.GetManagedObjects()
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 70, in __call__
    return self._proxy_method(*args, **keywords)
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.AccessDenied: Rejected send message, 2 matched rules; type="method_call", sender=":1.19" (uid=1000 pid=618 comm="python3 ble_uart.py ") interface="org.freedesktop.DBus.ObjectManager" member="GetManagedObjects" error name="(unset)" requested_reply="0" destination=":1.5" (uid=0 pid=424 comm="/usr/lib/bluetooth/bluetoothd ")
```

ukBaz, 2022-02-23:
When you run `bluetoothctl` do you have to run that with `sudo` also?

vovagorodok, 2022-02-23:
No, `bluetoothctl` doesn't require sudo

ukBaz, 2022-02-23:
OK, let me run some tests

ukBaz, 2022-02-23:
I've done a fresh install of Bluezero and checked that there is no `ukBaz.bluezero` is `/etc/dbus-1/system.d/`.

I think your issue is coming from else were.

```
(venv) [bash]python-bluezero $ python examples/ble_uart.py 
Advertisement registered
Connected to 58:19:A8:D6:77:DB
raw bytes: [116, 101, 115, 116]
With options: {'device': '/org/bluez/hci0/dev_58_19_A8_D6_77_DB', 'link': 'LE', 'mtu': 517}
Text value: test
Sending
raw bytes: [98, 108, 117, 101]
With options: {'device': '/org/bluez/hci0/dev_58_19_A8_D6_77_DB', 'link': 'LE', 'mtu': 517}
Text value: blue
Sending
^C(venv) [bash]python-bluezero $ ll /etc/dbus-1/system.d/uk*
ls: cannot access '/etc/dbus-1/system.d/uk*': No such file or directory
```

vovagorodok, 2022-02-23:
Yes RPiOS Lite on RPi Zero 2.
Works, but I don't know exact solution. What I'm done:
```
sudo apt install pulseaudio-module-bluetooth
sudo reboot
sudo apt remove pulseaudio-module-bluetooth
```

ukBaz, 2022-02-23:
I think it is adding user `pi` to group `bluetooth` that is the key bit.
```
pi@raspberrypi4b:~ $ sudo usermod -G bluetooth -a pi  
```

vovagorodok, 2022-02-23:
Yes. I forgot. This one was done by me to

ukBaz, 2022-02-23:
OK. I think this sounds like a general RPi setup rather than an issue with Bluezero. Can I go ahead and close this issue?